arendst/Tasmota #9389: ESP-xxx wifi after wifi off

ilvalle, 2020-09-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After running `wifi off` the device starts to publish a new wifi `ESP-xxx`, the network configuration (as gateway and netmask) is the same as before (when connected to the wifi) but the ip which is not set.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff basic
- [x] Tasmota binary firmware version number used: 8.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: rpi
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:31:22 CMD: Backlog Template; Module; GPIO 255
08:31:23 MQT: tasmota/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:31:23 MQT: tasmota/stat/RESULT = {"Module":{"1":"Sonoff Basic"}}
08:31:23 MQT: tasmota/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
08:31:52 MQT: tasmota/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":56,"Free":455,"Rules":"on Time#Minute=1230 do Backlog power off; wifi off endon"}
08:31:52 MQT: tasmota/stat/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":35,"Free":476,"Rules":"on Time#Minute=510 do wifi on endon"}
08:31:52 MQT: tasmota/stat/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":89,"Free":422,"Rules":"on power1#boot do Backlog time 1600025700; time 0 endon on Button1#State do wifi on endon"}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:32:17 MQT: tasmota/stat/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:32:17 MQT: tasmota/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:17","StartupUTC":"2020-09-26T06:30:00","Sleep":50,"CfgHolder":4617,"BootCount":47,"BCResetTime":"2020-09-18T07:49:32","SaveCount":127,"SaveAddress":"F5000"}}
08:32:17 MQT: tasmota/stat/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-09T11:41:02","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"356/699"}}
08:32:17 MQT: tasmota/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Pibu2","Pibu2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00800200","00006000","00000000"]}}
08:32:17 MQT: tasmota/stat/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:32:17 MQT: tasmota/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3250","IPAddress":"192.168.100.245","Gateway":"192.168.100.250","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.250","Mac":"C8:2B:96:02:EC:B2","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
08:32:17 MQT: tasmota/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_02ECB2","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:32:17 MQT: tasmota/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-09-26T06:32:17","Local":"2020-09-26T08:32:17","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:43","Sunset":"19:38"}}
08:32:17 MQT: tasmota/stat/STATUS10 = {"StatusSNS":{"Time":"2020-09-26T08:32:17"}}
08:32:17 MQT: tasmota/stat/STATUS11 = {"StatusSTS":{"Time":"2020-09-26T08:32:17","Uptime":"0T00:02:17","UptimeSec":137,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Pibu2","BSSId":"F8:1A:67:09:9E:97","Channel":6,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
### TO REPRODUCE
_Steps to reproduce the behavior:_
Run `wifi off` while connected to the AP

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The wifi to be turned off completely  

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Please add the `wifi on` and `wifi off` commands in the docs. 
Related issues:
https://github.com/arendst/Tasmota/issues/5979

**(Please, remember to close the issue when the problem has been addressed)**
```
08:33:45 CMD: wificonfig
08:33:45 MQT: tasmota/stat/RESULT = {"WifiConfig":{"5":"Wait"}}
```

Jason2866, 2020-09-26:
Yes, you shot the ESP8266 in the knee. The command is only for the ESP32
Entry from the Tasmota changelog
```Add basic support for ESP32 ethernet adding commands Wifi 0/1 and Ethernet 0/1 both default ON``` 
You are invited to add the info to the docs. The documentation is Open Source too.

ilvalle, 2020-09-26:
is there any plan to support ESP8266 too? Is it an hw limitation?

Jason2866, 2020-09-26:
Many functions in Tasmota does not work without a connection to a NTP time server.
There are no plans for. This feature was only introduced for the ESP32 (if there is a wired LAN
interface active) to switch off the wifi interface. It is not for switching off (W)LAN connectivity in general.

ilvalle, 2020-09-26:
Hi Jason, I understand your point regarding NTP, as you can see in my basic rules, the device can still work, the single item necessary is to set, after a reboot, the time few minutes later.
In my case, I switch off the wifi off at 8:30pm and after the reboot the time device is set to 8:35pm, at 8:30 am the wifi is switched on again.
As requested in other issues there are few use cases when switching off wifi is useful.

ilvalle, 2020-10-13:
hi @Jason2866 is there any plan to implement such feature for ESP8266 too?

Jason2866, 2020-10-13:
Afaik, no.

sfromis, 2020-10-13:
Turing off wifi is mostly relevant for ESP32 as this chip also allows the use of wired ethernet. As Tasmota is designed for having network access, there would have to be a pretty good use case for *needing* to turn off wifi on ESP8266.

ilvalle, 2020-10-16:
hi @sfromis there are few use cases where is useful to turn off wifi. Is it really so complex to implement it?

sfromis, 2020-10-16:
I was not trying to argue cost/benefit, just explaining. It may not be "terribly complex", just well outside of the raison d'être for a MQTT based project on a Wifi chip. Not sure that convincing someone to do the changes (for a probably very minimal benefit)  is easier than "just" doing it.

ilvalle, 2020-10-17:
@sfromis I see you point. Let me know if i can help in some way when you will decide to do something in that direction

FoxtrottPL, 2020-11-10:
I can confirm it .
When router lost connection on WAN - WiFi still present my all SP111 goes to AP mode (lost SSID and password) I must connect to Tasmota-1234 (5 sockets)and write my WiFi and password , and again , again ,again - what can I do to prevent this?

Jason2866, 2020-11-10:
@FoxtrottPL  take a look in the docs and search for `wificonfig`
This is not related to the feature request

FoxtrottPL, 2020-11-10:
> @FoxtrottPL take a look in the docs and search for `wificonfig`
> This is not related to the feature request
Thank You for the info , I am new in Tasmota enviroment - 
I must type WifiConfig 5 in console? Or something else?

Jason2866, 2020-11-10:
Yes, thats all

FoxtrottPL, 2020-11-10:
Thank You Good Man !

ascillato2, 2020-11-16:
This command was working for esp8266 too.
For latest Tasmota is not working anymore.

Changing label to BUG.

ascillato, 2020-11-16:
Now, if you do `wifi 0` on the console, it starts populating an **ESP-xxx** wifi spot without webserver. If then, you send by serial `wifi 1`, your device restarts and start working normally. When this command was introduced in june'20, was working fine.

arendst, 2020-11-16:
As it was never the intent to have it used on ESP8266 I will make it's use unavailable.

Solves issue here.

closed 2020-11-16 by ascillato2