arendst/Tasmota #9394: Air pressure offset zigbee

Oblivionevil, 2020-09-26:
**Have you looked for this feature in other issues and in the docs?**  

yes but i found only solution for i2c

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Sea level offset for zigbee devices 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Offset for pressure

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._
Hello,

today i paired my Aqara Temperature Humidity Sensor.
[image]

but it is not possible to calibrate the pressure level.
Is there a option to set an offset for the device ?

Thanks for the help

oblivionevil

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-09-26:
Mount the ausensensor ten meter higher...

Oblivionevil, 2020-09-26:
Sorry misunderstanding the correct value is at the moment 990 hpa

Jason2866, 2020-09-26:
What makes you think 990hpa is the correct value?
Two of your sensors show nearly identic values

Oblivionevil, 2020-09-26:
3 Weather Stations show this value and online service

Jason2866, 2020-09-26:
Most pressure sensors are very precise. In my experience better than temperature and humidity...
[image]
[image]

Oblivionevil, 2020-09-26:
Altitude has no effect on zigbee

Jason2866, 2020-09-26:
Okay, you want the feature sea pressure for zigbee! This is NOT a offset
Different thing. Sea pressure is calculated in Tasmota

s-hadinger, 2020-09-27:
I didn't know about Sea Pressure calculation and can add it to Mqtt messages

PR #9407 (2020-09-27): Zigbee add SeaPressure
## Description:

Add `SeaPressure` attribute calculate using the `Altitude` setting.

Ex: (with `Altitude 130`)
```
[...]"Pressure":990,"SeaPressure":1006[...]
```

The WebUI now reports SeaPressure instead of Pressure.

**Related issue (if applicable):** fixes #9394 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x] The code change is tested and works on core ESP32 V.1.12 [... cut]
closed 2020-09-27 by s-hadinger