arendst/Tasmota #9395: Translation of LWT payload

S474N, 2020-09-26:
### PROBLEM DESCRIPTION
Don't know, if this problem is in all languages, but in Czech version of Tasmota v8.5.0.1 is problem with translation of LWT payload.

It shows `Aktivní/Neaktivní` instead of `Online/Offline`.

This can make problems with another softwares/plugins, which expect standard english words.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff RFR3, Sonoff Mini
- [x] Tasmota binary firmware version number used: 8.5.0.1
  - [x] Pre-compiled: precompiled CZECH version
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Check payload of `/tele/device-id/LWT` and power cycle

### EXPECTED BEHAVIOUR
Use standard english strings, not translated.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-26:
I do not see a translation for LWT payload in the cz laguage file. There is no other file for language specific translations. It looks like you have redefined "somewhere"

S474N, 2020-09-26:
Please see link, which is mentioned.

I used standard tasmota-cz.bin, no changes. When I tested now with standard tasmota.bin, everything is OK.

arendst, 2020-09-26:
There are indeed translatioms in the cz.h language file.

A change is needed in the language file.

arendst, 2020-09-26:
Trouble is some other languages did indeed translate online/offline in their language....

What is common sense here....

ascillato, 2020-09-26:
May be we need to move the `online` key out of translation files due to it is not related to the webUI. What about move it to my_user_config.h? It makes more sense there?

Jason2866, 2020-09-26:
I vote for moving out of language file too. I didnt even recognized that this was LWT...
Imho there is no need for this to be a user configurable anywhere

closed 2020-09-27 by ascillato2