arendst/Tasmota #9398: Tasmota-Minimal.bin does not start the Webserver.

kugelkopf123, 2020-09-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When flashing a Shelly2.5 with TasmotaMinimal FW. This device doesnt start the Webserver. 
So it isnt possible to upload the new FW. 
The Device is Software Bricked then.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly2.5
- [x] Tasmota binary firmware version number used: 8.5.0.1 latest Dev.
  - [x] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO on VSCode
- [ ] Flashing tools used: PIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
{$␀lܟ|␀�$�|␂␄␄␌�␄l�␄c|ǃ␃�␛�r�c�␄c��no�dog���␌#␜p��dsd;lp�'�␐␂␌␄�␌d␄��␄␌␄b␌g�|␃섏��c��og�␀$��d`␂�␓␛og␄$`␃␎␃orǛ�g␄␌ss�`␃␎{��o␌␄{r�`␃�␓{�␌d���l ␃��g�␃00:00:00 Projekt tasmota  Version 8.5.0.1(minimal)-STAGE
00:00:00 ACHTUNG: Diese Version unterstützt keine persistenten Einstellungen
00:00:00 WIF: Prüfe Verbindung...
00:00:00 WIF: Verbindungsversuch...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Prüfe Verbindung...
00:00:01 WIF: Verbindungsversuch...
00:00:02 WIF: Prüfe Verbindung...
00:00:02 WIF: Verbindungsversuch...
00:00:03 WIF: Prüfe Verbindung...
00:00:03 WIF: Verbindungsversuch...
00:00:04 WIF: Network 0, AP-, SSId ESP-05A8E9, Channel 1, BSSId 8E:AA:B5:05:A8:E9, RSSI -75, Encryption 0
00:00:04 WIF: Network 1, AP-, SSId FB, Channel 1, BSSId E0:28:6D:22:3F:BD, RSSI -57, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId FRITZ!Box 7590 DB, Channel 11, BSSId 44:4E:6D:A5:1F:71, RSSI -78, Encryption 1
00:00:04 WIF: Prüfe Verbindung...
00:00:04 WIF: Verbindungsversuch...
00:00:05 WIF: Prüfe Verbindung...
00:00:05 WIF: Verbindungsversuch...
00:00:06 WIF: Prüfe Verbindung...
00:00:06 WIF: Verbindungsversuch...
00:00:07 WIF: Prüfe Verbindung...
00:00:07 WIF: Verbindungsversuch...
00:00:08 WIF: Prüfe Verbindung...
00:00:08 WIF: Verbindungsversuch...
00:00:08 APP: Anzahl Startvorgänge 0
00:00:09 WIF: Prüfe Verbindung...
00:00:09 WIF: Verbindungsversuch...
00:00:10 WIF: Prüfe Verbindung...
00:00:10 WIF: Verbindungsversuch...
00:00:11 WIF: Prüfe Verbindung...
00:00:11 WIF: Verbindung fehlgeschlagen, da der AP nicht antwortet (timeout)
00:00:12 WIF: Prüfe Verbindung...
00:00:12 WIF: Verbindungsversuch...
00:00:13 WIF: Prüfe Verbindung...
00:00:13 WIF: Verbindungsversuch...
00:00:15 WIF: Prüfe Verbindung...
00:00:15 WIF: Verbindungsversuch...
00:00:16 WIF: Prüfe Verbindung...
00:00:16 WIF: Verbindungsversuch...
00:00:17 WIF: Prüfe Verbindung...
00:00:17 WIF: Verbindungsversuch...
00:00:18 WIF: Prüfe Verbindung...
00:00:18 WIF: Verbindungsversuch...
00:00:19 WIF: Prüfe Verbindung...
00:00:19 WIF: Verbindungsversuch...
00:00:20 WIF: Prüfe Verbindung...
00:00:20 WIF: Verbindungsversuch...
00:00:21 WIF: Prüfe Verbindung...
00:00:21 WIF: Verbindungsversuch...
00:00:22 WIF: Prüfe Verbindung...
00:00:22 WIF: Verbindungsversuch...
00:00:23 WIF: Prüfe Verbindung...
00:00:23 WIF: Verbindung fehlgeschlagen, da der AP nicht antwortet (timeout)
00:03:35 APP: starte neu

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xcc
csum 0xcc
va3281fe2
~ld
00:00:00 SRC: Restart
00:00:00 Projekt tasmota  Version 8.5.0.1(minimal)-STAGE
00:00:00 ACHTUNG: Diese Version unterstützt keine persistenten Einstellungen
00:00:00 WIF: Prüfe Verbindung...
00:00:00 WIF: Verbindungsversuch...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Prüfe Verbindung...
00:00:01 WIF: Verbindungsversuch...
00:00:02 WIF: Prüfe Verbindung...
00:00:02 WIF: Verbindungsversuch...
00:00:03 WIF: Prüfe Verbindung...
00:00:03 WIF: Verbindungsversuch...
00:00:04 WIF: Network 0, AP-, SSId ESP-05A8E9, Channel 1, BSSId 8E:AA:B5:05:A8:E9, RSSI -76, Encryption 0
00:00:04 WIF: Network 1, AP-, SSId Kol-Fre, Channel 1, BSSId 2C:91:AB:B0:2F:83, RSSI -92, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId FB, Channel 1, BSSId E0:28:6D:22:3F:BD, RSSI -57, Encryption 1
00:00:04 WIF: Network 3, AP-, SSId FRITZ!Box Gastzugang, Channel 1, BSSId 2E:91:AB:B0:2F:83, RSSI -90, Encryption 1
00:00:04 WIF: Network 4, AP-, SSId FRITZ!Box 7590 DB, Channel 11, BSSId 44:4E:6D:A5:1F:71, RSSI -82, Encryption 1
00:00:04 WIF: Prüfe Verbindung...
00:00:04 WIF: Verbindungsversuch...
00:00:05 WIF: Prüfe Verbindung...
00:00:05 WIF: Verbindungsversuch...
00:00:06 WIF: Prüfe Verbindung...
00:00:06 WIF: Verbindungsversuch...
00:00:07 WIF: Prüfe Verbindung...
00:00:07 WIF: Verbindungsversuch...
00:00:08 WIF: Prüfe Verbindung...
00:00:08 WIF: Verbindungsversuch...
00:00:09 APP: Anzahl Startvorgänge 0
00:00:09 WIF: Prüfe Verbindung...
00:00:09 WIF: Verbindungsversuch...
00:00:10 WIF: Prüfe Verbindung...
00:00:10 WIF: Verbindungsversuch...
00:00:11 WIF: Prüfe Verbindung...
00:00:11 WIF: Verbindung fehlgeschlagen, da der AP nicht antwortet (timeout)
00:00:12 WIF: Prüfe Verbindung...
00:00:12 WIF: Verbindungsversuch...
00:00:14 WIF: Prüfe Verbindung...
00:00:14 WIF: Verbindungsversuch...
00:00:15 WIF: Prüfe Verbindung...
00:00:15 WIF: Verbindungsversuch...
00:00:16 WIF: Prüfe Verbindung...
00:00:16 WIF: Verbindungsversuch...
00:00:17 WIF: Prüfe Verbindung...
00:00:17 WIF: Verbindungsversuch...
00:00:18 WIF: Prüfe Verbindung...
00:00:18 WIF: Verbindungsversuch...
00:00:19 WIF: Prüfe Verbindung...
00:00:19 WIF: Verbindungsversuch...
00:00:20 WIF: Prüfe Verbindung...
00:00:20 WIF: Verbindungsversuch...
00:00:21 WIF: Prüfe Verbindung...
00:00:21 WIF: Verbindungsversuch...
00:00:22 WIF: Prüfe Verbindung...
00:00:22 WIF: Verbindungsversuch...
00:00:23 WIF: Prüfe Verbindung...
00:00:23 WIF: Verbindung fehlgeschlagen, da der AP nicht antwortet (timeout)
00:03:35 APP: starte neu

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Self compile Tasmota Minimal without something in the user_config_override or Download precompiled from Github.
1. Option:
Erase Flash via PIO,
Flash Tasmota Minimal
You can find the AP but if you log into it, and try to connect to 192.168.4.1, the Menu is unavailable. 
Try to config something via Serialconsole isnt possible. Get Error:

```
00:00:03 CFG: Text overflow by 3461 char(s)
00:00:03 CFG: Text overflow by 3461 char(s)
00:00:03 CFG: Text overflow by 3461 char(s)
00:00:03 RSL: /SERIALRECEIVED = {"SerialReceived":"6173646661736466"}

```
while tickin something.

2. Option.
Have a running Tasmota on it.
Flash Tasmota-Minimal

By both options you get the same result.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota opens an AP and it is possible to connect on 192.168.4.1 to upload the new FW.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-26:
You use core stage. `Projekt tasmota  Version 8.5.0.1(minimal)-STAGE`
For stage version there is no support.

kugelkopf123, 2020-09-26:
@Jason2866 
As I already mentioned in my Post, Ive tried it with both. Self Compiled and preCompiled from Github. Or is this not the right file? https://github.com/arendst/Tasmota/releases/download/v8.5.0/tasmota-minimal.bin

ascillato, 2020-09-27:
If you have flashed minimal as the first time from stock firmware, that is not going to work. Minimal is not Tasmota, it is just a firmware to make updates from an old Tasmota version to another.
If this is your issue, please erase all flash and flash again by serial but with Tasmota.bin

ascillato2, 2020-09-27:
The technical side of this, is that minimal does not write any config to flash. It uses the config from the Tasmota version that was BEFORE using the minimal version.
Minimal does not have features. It is only to free flash space while updating by OTA from a full Tasmota version to another.

closed 2020-09-27 by ascillato2

kugelkopf123, 2020-09-27:
Ok. So if i get it right. When you have a installed version of Tasmota, with WIFI connected to your Router and then OTA update  with Minimal, this Minimal FW should connect to your Router so that you could upload a new FW. Right?
Great. 
I ve found the Problem. 
If the FW thats on the Device uses another Linkerfile. eg. 4m2m 2m1m, as you need when working with big scripts, then the minimal version doesn't find any wifi data after upload. 

So as the minimal Version is designed for upload from old to new FW with 1MB i would say, yes it works as designed.

ascillato, 2020-09-27:
Yes, scrambling different memory maps don't work. Your configuration, the SDK wifi stack and so on will be in different memory locations and won't work. Tasmota is designed for being used in 1MB memory map. You don't need to change to another custom type as all the functions will be there also in the 1MB.

kugelkopf123, 2020-09-27:
> Yes, scrambling different memory maps don't work. Your configuration, the SDK wifi stack and so on will be in different memory locations and won't work. Tasmota is designed for being used in 1MB [...]

@ascillato https://tasmota.github.io/docs/Scripting-Language
Section: Script buffer Size.

Jason2866, 2020-09-27:
@kugelkopf123 This is not a contradiction. You have to build the minimal version with the same linker script. 
Thats why we dropped 4M precompiled builds. If you use not the matching versions you have
to reflash via serial.
Using and providing only builds with 1M generates no traps.

arendst, 2020-09-27:
Tasmota can be used with scripting (instead of rules) and works with the default 1MB memory map. 

Any need for a change other than 1MB memory map is not supported by the Tasmota although the Scripter programmer may report otherwise.

ascillato, 2020-09-27:
Yes, but that is only for that use case. For most use cases, the standard Tasmota version is enough. 

Let me explain. 

The esp8266 chip can address just 1MB for a program in flash. That is the maximum and it is a hardware limitation. The rest of the memory flash can be used for OTA (that needs an extra 1MB) and the remaining can be used ONLY for data. Not program, not OTA.
When using a 4MB memory map, you can forget about the minimal version. It is not needed nor useful anymore. In a 4MB memory map, your Esp8266 can use the maximum for OTA (that is 1MB firmware) so, OTA updates can be done at once without any intermediate firmware. In this example case of 4MB flash, your memory map should be 1MB program + 1MB OTA + 2MB data.
Very few commercial end-user devices came with 4MB. Most came with 1MB like most sonoffs and others with 2MB like shellys. Nodemcu came with 4MB.

In the other hand, now Tasmota supports compressed firmwares (Tasmota.bin.gz) meaning that minimal is not needed for most updates. Also Tasmota has OTAMagic when doing OTA updates by OTAurl. So you just point Tasmota to the firmware URL you want to upgrade to and Tasmota (if not using a compressed firmware) will automatically upgrade to a minimal version and then to the full version without the need of any user's intervention.

In your case (self compile with a custom memory map of 4MB with your custom selection of features) you don't need minimal at all.

Also minimal is designed ONLY for in-between OTA upgrades for standard 1MB Tasmota's firmwares. It won't work as a first flash.

Hope this explanation helps.

kugelkopf123, 2020-09-27:
@ascillato many thanks for the explanation! I had thought you could possibly use it to reformat it, so to say.

ascillato, 2020-09-27:
You are welcome.

If you want to dig more, you can search in the platform.io files and you can find the memory map for each case. Inside you can see that in every map, the address of flash used for each part (program, data, ota, sdk settings, etc) is different. So, an online reformat of flash seems to be possible but that seems to be a hard work and it is more arduino core related. The hardest part is the SDK that is closed source, so, it adds a big "if" to a reformat.
Anyway, as seems to be a big job and as the normal use of any firmware is to use the same memory map, I don't think it is going to be worked on.

So, the advice here is to use 1MB memory map for all your devices, so you can simplify your home setup, and use the 4MB as a last resource if your script is so big that can't fit in the normal flash. Remember that now Tasmota will compress your script, so you should have even more free space.