arendst/Tasmota #9399: Rotary dial on Xiaomi desk lamp is unreliable

poofyteddy, 2020-09-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
after the rework on the rotary dial (and the closure of #8263), i now have tried to update my Xiaomi desk lamp.
Sadly the rotary dial isn't reliable anymore.
It will not register, or register in the wrong rotation, not change rotation when i do...
i'm thinking about the new debounce and timeout but i am not sure.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Xiaomi desk lamp
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VSCODE + platformio (linux)
- [X] Flashing tools used: first flash with esptool, then OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:37:04 MQT: RoomGBedside2/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:37:04 MQT: RoomGBedside2/stat/RESULT = {"Module":{"66":"Mi Desk Lamp"}}
21:37:04 MQT: RoomGBedside2/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"17":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"150":"Rotary1a"},"GPIO13":{"151":"Rotary1b"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:38:35 MQT: RoomGBedside2/stat/STATUS = {"Status":{"Module":66,"DeviceName":"RoomGBedside2","FriendlyName":["RoomGBedside2"],"Topic":"RoomGBedside2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:38:35 MQT: RoomGBedside2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:43","StartupUTC":"2020-09-26T20:29:52","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-04-26T15:02:09","SaveCount":70,"SaveAddress":"FB000"}}
21:38:35 MQT: RoomGBedside2/stat/STATUS2 = {"StatusFWR":{"Version":"8.5.0.1(tasmota)","BuildDateTime":"2020-09-26T22:25:48","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"425/699"}}
21:38:35 MQT: RoomGBedside2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["REDACTED",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00206000","00000000"]}}
21:38:35 MQT: RoomGBedside2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":631,"Free":372,"Heap":18,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE7D7","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:38:35 MQT: RoomGBedside2/stat/STATUS5 = {"StatusNET":{"Hostname":"RoomGBedside2-7201","IPAddress":"REDACTED","Gateway":"REDACTED","Subnetmask":"REDACTED","DNSServer":"1.1.1.1","Mac":"REDACTED","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:38:35 MQT: RoomGBedside2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"REDACTED","MqttPort":REDACTED,"MqttClientMask":"REDACTED","MqttClient":"REDACTED","MqttUser":"REDACTED","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:38:35 MQT: RoomGBedside2/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-09-26T20:38:35","Local":"2020-09-26T21:38:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:43","Sunset":"18:38"}}
21:38:35 MQT: RoomGBedside2/stat/STATUS10 = {"StatusSNS":{"Time":"2020-09-26T21:38:35"}}
21:38:35 MQT: RoomGBedside2/stat/STATUS11 = {"StatusSTS":{"Time":"2020-09-26T21:38:35","Uptime":"0T00:08:43","UptimeSec":523,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":50,"Color":"0,128","White":50,"CT":500,"Channel":[0,50],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"REDACTED","Channel":1,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
This is on, 5 +, 5 -, off
21:45:38 APP: Button1 multi-press 1
21:45:38 MQT: RoomGBedside2/stat/BUTTON1T = {"TRIG":"SINGLE"}
21:45:38 SRC: Button
21:45:38 MQT: RoomGBedside2/tele/STATE = {"Time":"2020-09-26T21:45:38","Uptime":"0T00:15:46","UptimeSec":946,"Heap":18,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":50,"Color":"0,128","White":50,"CT":500,"Channel":[0,50],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"REDACTED","Channel":1,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:45:38 MQT: RoomGBedside2/stat/RESULT = {"POWER":"ON"}
21:45:38 MQT: RoomGBedside2/stat/POWER = ON
21:45:39 CFG: Saved to flash at F9, Count 72, Bytes 4096
21:45:39 MQT: RoomGBedside2/tele/STATE = {"Time":"2020-09-26T21:45:39","Uptime":"0T00:15:47","UptimeSec":947,"Heap":18,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":37,"MqttCount":1,"POWER":"ON","Dimmer":60,"Color":"0,153","White":60,"CT":500,"Channel":[0,60],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"REDACTED","Channel":1,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:45:40 MQT: RoomGBedside2/stat/RESULT = {"POWER":"ON","Dimmer":60,"Color":"0,153","White":60,"CT":500,"Channel":[0,60]}
21:45:41 MQT: RoomGBedside2/tele/STATE = {"Time":"2020-09-26T21:45:41","Uptime":"0T00:15:49","UptimeSec":949,"Heap":18,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":44,"MqttCount":1,"POWER":"ON","Dimmer":70,"Color":"0,179","White":70,"CT":500,"Channel":[0,70],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"REDACTED","Channel":1,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:45:42 MQT: RoomGBedside2/stat/RESULT = {"POWER":"ON","Dimmer":70,"Color":"0,179","White":70,"CT":500,"Channel":[0,70]}
21:45:43 MQT: RoomGBedside2/tele/STATE = {"Time":"2020-09-26T21:45:43","Uptime":"0T00:15:51","UptimeSec":951,"Heap":18,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":52,"MqttCount":1,"POWER":"ON","Dimmer":80,"Color":"0,204","White":80,"CT":500,"Channel":[0,80],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"REDACTED","Channel":1,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:45:44 MQT: RoomGBedside2/tele/STATE = {"Time":"2020-09-26T21:45:44","Uptime":"0T00:15:52","UptimeSec":952,"Heap":18,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":57,"MqttCount":1,"POWER":"ON","Dimmer":90,"Color":"0,230","White":90,"CT":500,"Channel":[0,90],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"REDACTED","Channel":1,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:45:45 MQT: RoomGBedside2/stat/RESULT = {"POWER":"ON","Dimmer":90,"Color":"0,230","White":90,"CT":500,"Channel":[0,90]}
21:45:46 CFG: Saved to flash at F8, Count 73, Bytes 4096
21:45:47 MQT: RoomGBedside2/tele/STATE = {"Time":"2020-09-26T21:45:47","Uptime":"0T00:15:55","UptimeSec":955,"Heap":18,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":73,"MqttCount":1,"POWER":"ON","Dimmer":80,"Color":"0,204","White":80,"CT":500,"Channel":[0,80],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"RED [... cut ...]
_Steps to reproduce the behavior:_
Compile with rotary enable, play with the dial

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
each step should have an effect, and in the right direction

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
See video [HERE](https://youtu.be/_dJPiMfIIL0)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-09-28:
I did some testing too and see no big issues.

It's good to know that the dimmer only changes after TWO clicks of the dial simply because only after two clicks a rotation cycle is completed. This is a change from the initial implemtation and on purpose to fix possible direction and mis-interpretations. This also allows more stable rotation for other uses.

My log below:
```
11:46:45 APP: Button1 multi-press 1
11:46:46 SRC: Button
11:46:46 MQT: stat/midesk/RESULT = {"POWER":"ON"}
11:46:46 MQT: stat/midesk/POWER = ON
11:46:46 CFG: Saved to flash at F9, Count 511, Bytes 4096
11:46:49 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":90,"Color":"17CF","White":90,"CT":466,"Channel":[9,81]}
11:46:50 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":100,"Color":"19E6","White":100,"CT":466,"Channel":[10,90]}
11:46:50 WIF: Checking connection...
11:46:52 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":100,"Color":"19E6","White":100,"CT":466,"Channel":[10,90]}
11:46:54 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":90,"Color":"17CF","White":90,"CT":466,"Channel":[9,81]}
11:46:54 CFG: Saved to flash at F8, Count 512, Bytes 4096
11:46:55 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":80,"Color":"14B8","White":80,"CT":466,"Channel":[8,72]}
11:46:56 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":70,"Color":"12A1","White":70,"CT":466,"Channel":[7,63]}
11:46:57 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":60,"Color":"0F8A","White":60,"CT":466,"Channel":[6,54]}
11:46:59 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":50,"Color":"0D73","White":50,"CT":466,"Channel":[5,45]}
11:46:59 CFG: Saved to flash at F7, Count 513, Bytes 4096
11:47:00 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":40,"Color":"0A5C","White":40,"CT":466,"Channel":[4,36]}
11:47:02 CFG: Saved to flash at F6, Count 514, Bytes 4096
11:47:04 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":50,"Color":"0D73","White":50,"CT":466,"Channel":[5,45]}
11:47:05 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":60,"Color":"0F8A","White":60,"CT":466,"Channel":[6,54]}
11:47:06 MQT: stat/midesk/RESULT = {"POWER":"ON","Dimmer":70,"Color":"12A1","White":70,"CT":466,"Channel":[7,63]}
11:47:06 APP: Button1 multi-press 1
11:47:07 SRC: Button
11:47:07 MQT: stat/midesk/RESULT = {"POWER":"OFF"}
11:47:07 MQT: stat/midesk/POWER = OFF
11:47:07 CFG: Saved to flash at F5, Count 515, Bytes 4096
```
follows exactly what I did with the dial where every two clicks increase/decrease dimmer by 10.

poofyteddy, 2020-09-28:
Thank for trying.
> It's good to know that the dimmer only changes after TWO clicks of the dial simply because only after two clicks a rotation cycle is completed.

I wouldn't mind this if i couldn't stop the rotation in the middle of those 2 click. I rarely set it by counting the number of step, i am more likely to turn it 10% to 30% without looking and going down the other way around if i overshoot.

But the main issue is that i have issue when i come to direction. As you might be able to see in the video, going up some time bring the brightness down and vice versa.

Also when i turn the dial quickly like 50-60% at a time, it will sometime not do anything, or light up then down...

could you send me your bin.gz as a sanity check ?
There shouldn't be anything but i could have screw something up ?

also i didn't find doc about the rule trigger with the SetOption98, would you have those on hand so i can make rule that debug print what the dial detected in the console ?

arendst, 2020-09-28:
My binary was the current dev version as available from the dev download site. See readme.

poofyteddy, 2020-09-28:
sorry i had no idea enabling the rotary flag at build time weren't required anymore
i have made 2 new video to show you the issue

https://youtu.be/TfHPvFWXLkk 
In this video i only turn down the dial. i start at 100% brightness, for the first 4 second nothing seam to react, then it go down, but at 6 second the light goes brighter for a bit before dimming again.

https://youtu.be/zv6-Q0GWUwU 
In this video i only turn up the dial, i start at 0% brightness, at the begining the light turn on and off a couple of time, then it got brighter for a bit, and dimmer at 4 second. then bright again until 8, where it get dimmer once again

This is running the last github release with only module 66 loaded, no other change

tastopesato, 2020-10-13:
> sorry i had no idea enabling the rotary flag at build time weren't required anymore
> i have made 2 new video to show you the issue
>
> https://youtu.be/TfHPvFWXLkk
> In this video i only turn down the [...]

Same Problem here.

arendst, 2020-10-13:
Try latest dev.

If your experience continues try to play with:
```
const uint8_t ROTARY_TIMEOUT = 5;              // 5 * RotaryHandler() call which is usually 5 * 0.05 seconds
const uint8_t ROTARY_DEBOUNCE = 10;            // Debounce time in milliseconds
```
in file ``support_rotary.ino``

tastopesato, 2020-10-13:
> Try latest dev.
> 
> If your experience continues try to play with:
> 
> ```
> const uint8_t ROTARY_TIMEOUT = 5;              // 5 * RotaryHandler() call which is usually 5 * 0.05 seconds
> const uint8_t ROTARY_DEBOUNCE = 10;            // Debounce time in milliseconds
> ```
> 
> in file `support_rotary.ino`

Tried latest dev version;
later recompile with ROTARY_DEBOUNCE = 5 and ROTARY_DEBOUNCE = 50 with no luck

tastopesato, 2020-10-13:
> Sorry, Can't help. Mine functions as it should.
> 
> Remember, it's not that you can turn the dial quickly left and quickly right. It just cannot detect the pulses that fast as it has a hysterisis for correct direction detection.
> 
> Anyway, if I do it "normally" I don't see any dimming issue.

So, i found better performance with these values:

const uint8_t ROTARY_TIMEOUT = 18;              // 5 * RotaryHandler() call which is usually 5 * 0.05 seconds
const uint8_t ROTARY_DEBOUNCE = 10;            // Debounce time in milliseconds

poofyteddy, 2020-10-13:
Thank's for trying, i have started to try to put back the old code who worked way better in my memory, but i don't have the energy to put in this after work for now :(

> So, i found better performance with these values

By performance you mean the higher accepted rotation speed ? or a lower number of error ?

tastopesato, 2020-10-13:
> Thank's for trying, i have started to try to put back the old code who worked way better in my memory, but i don't have the energy to put in this after work for now :(
>
>
> By performance you mean the [...]

Lower number of error.
Can you share your old code?

poofyteddy, 2020-10-15:
I didn't do much at all, just started removing some stuff and understanding how to put the old stuff back
This is the last commit before the big change if i am not wrong

arendst, 2020-10-19:
Gents. the latest dev version contains different rotary code.

The current active code (ROTARY_OPTION1) resembles closely the original MiDesk rotary driver. It is less useful on other rotary devices as it's steps are to coarse.

Pls give it a try on your MiDesks and report your experiences.

tastopesato, 2020-10-19:
> Gents. the latest dev version contains different rotary code.
> 
> The current active code (ROTARY_OPTION1) resembles closely the original MiDesk rotary driver. It is less useful on other rotary devices as it's steps are to coarse.
> 
> Pls give it a try on your MiDesks and report your experiences.

Just tried. Seems to work better than previous version, but sometimes, if i rotate clockwise the encoder after reaching full intensity, it seems to fade out and in ramdoly.

poofyteddy, 2020-10-19:
This new mode is really smooth and work pretty well !
I have the same issue as @tastopesato but very rarely, so it doesn't bother me.
Getting it perfect won't be worth your time and if it's only tuning from there, anyone can do it :)
I couldn't compile my own build because of mqtt issue (#define USE_MQTT_TLS) that is another story and maybe stuff have changed since.

arendst, 2020-10-20:
Thx. That's useful information.

poofyteddy, 2020-10-21:
something odd have happened yesterday.
maybe the planet where aligned just right ?
i flashed today build and found that most rotary direction where wrong, like 1 out of 3 ... 4 max.
So i checkout 1b25777 commit, compiled and flashed, and same...

I am sorry i cannot explain this change in behavior.
what about you @tastopesato ?

tastopesato, 2020-10-21:
> something odd have happened yesterday.
> maybe the planet where aligned just right ?
> i flashed today build and found that most rotary direction where wrong, like 1 out of 3 ... 4 max.
> So i checkout [...]

Same issue here. It appears randomly

arendst, 2020-10-22:
And another version to try...

poofyteddy, 2020-10-22:
For now I have 0 issue with the last one, but let's wait for the planet to move again ok ;)

arendst, 2020-10-23:
As the planet is not yet under my control I suggest you to close this issue.

poofyteddy, 2020-10-23:
Sorry Arendst, it was supposed to be humorous but apparently didn't come out well.
i flashed and tried the build a 11pm, just before bed. Last time i didn't found issue it's was in the same context, so i wanted to wait for a day and make sure it was working well.
This morning i tried again and didn't had any issue again.

@tastopesato since you've helped and have the device, i feel that it's fair for me to ask if you're ok with me closing this issue ?

arendst, 2020-10-23:
No offence taken. My request was also a bit of humorous.

tastopesato, 2020-10-23:
> Sorry Arendst, it was supposed to be humorous but apparently didn't come out well.
> i flashed and tried the build a 11pm, just before bed. Last time i didn't found issue it's was in the same context, [...]

Hi @poofyteddy PoofyTeddy, in my opinion the issues encountered are so minimum that you can close the issue.
Thanks @arendst arendst for your work!

closed 2020-10-24 by poofyteddy