arendst/Tasmota #9404: Update status of all connected Zigbee devices

hkrob, 2020-09-27:
**Have you looked for this feature in other issues and in the docs?**  
Yes - extensively

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

Unknown status of Zigbee devices

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

It would be great if we could send an MQTT command and have Tasmota respond with the current status of all connected devices. 
This would be useful because contact sensors for example do not periodically report their status, and restarting Home Assistant results in unknown status. Using Retain is an option, but not a great one.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Retain - works, but not ideal.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-09-27:
It's already the case for lights. But it's technically not possible due to the nature of the Zigbee protocol. Most device are in sleep mode and wake up only to send their data. There is no way to remotely wake them up. 

You need to be patient until they send data again.

hkrob, 2020-09-27:
Thanks - good to know..

I guess this needs another case, but what about topics?
It would be very handy for each Zigbee device to have it's own topic, including state .. Ideally we should be able to set a state_topic in HA for each Zigbee device.. Is that feasible?

s-hadinger, 2020-09-27:
It's already available via `SetOption112 1`. Documentation may not be up to date.  

See #9350

hkrob, 2020-09-27:
I will need to take another look, but I was trying earlier today and I didn't see a separate "clean" state topic, e.g. a tasmota switch looks like this:
office-ac/stat/POWER with value ON/OFF

But with the zigbee devices there was no simple "state" to be found, rather it was the long JSON string which isn't all that friendly to deal with in HA

s-hadinger, 2020-09-27:
There is no plan to do so. Z2T reports the data coming from zigbee devices with very little transformation. You need to parse the json in the Mqtt gateway or via rules. 

Did you explore using rules to publish exactly the Mqtt you wanted?

hkrob, 2020-09-27:
Thanks - no, I didn't look at rules.. Where would I implement those?

hkrob, 2020-09-27:
Following this.. what about connecting tasmota to zigbee2mqtt as that project already implements tonnes of rules...

I.e. Tasmota -> zigbee2mqtt -> HA (mosquitto etc..)

Jason2866, 2020-09-27:
@hkrob thats a feature request for zigbee2mqtt -> supporting the ZbBridge

ascillato2, 2020-09-27:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-27 by ascillato2

hkrob, 2020-09-29:
Doesn't look like Zigbee2MQTT is interested: https://github.com/Koenkk/zigbee2mqtt/issues/4479

This is a problem because the outputs from Tasmota -> MQTT are not very useful in their current state for Home Assistant..
Is there a "rules" project ongoing which could be inserted in between Tasmota and an MQTT server?

e.g. Tasmota --> Rules Engine --> Mosquitto

Jason2866, 2020-09-29:
You can use the ZHA implementation from HA. https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

hkrob, 2020-09-29:
Tried that, it's not very good.
Doesn't detect many features, lack of access to the settings, and worst of all, it randomly stopped working for me the other day and I had to re-pair everything... (This was the point where I went back to zigbee2mqtt with my cc2531...)