arendst/Tasmota #9409: ESP32 Touch pins does not work when ESP32 is connected to battery

mamaretti, 2020-09-27:
### PROBLEM DESCRIPTION
I've teste some ESP32 boards (eg DOIT Dev Kit) 
Touch-Pins working great when ESP32 is connected via USB zu Computer.
When ESP32 is connected to a Powerbank or another power source then Touchpins does not work

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32 DEVKIT
- [x] Tasmota binary firmware version number used: 8.5.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```
 
### TO REPRODUCE
Define
GPIO15 as Button_tc 1 and 
GPIO14 as Relay 1

### EXPECTED BEHAVIOUR
Touch pins should work with any power source 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
sometimes Touch pins working with battery connected, but very unreliable and only if another Finger touches ground.
probably the TOUCH_PIN_THRESHOLD value is not correct?

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-09-27:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2020-09-27:
> sometimes Touch pins working with battery connected, but very unreliable and only if another Finger touches ground.
probably the TOUCH_PIN_THRESHOLD value is not correct?

Seems to be a hardware issue more than a software one. 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-27 by ascillato2

mamaretti, 2020-09-27:
ok. see you in chat. Thanks

albnar, 2022-10-08:
I had the same problem. The solution was to bild a touch pad with two areas (for example two half circles) , one for touch pin and the other one connected to ground.  It seems that a pc has more "mass" and therfore a better ground for the circuit. With battery or another power suply the touch pad has to have an aditional pad to ground.