arendst/Tasmota #9411: TOUCH_PIN_THRESHOLD option parameter to change sensitivity

mamaretti, 2020-09-27:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
I found out that TOUCH_PIN_THRESHOLD is to low when you want to use a TouchPin on ESP32 with battery or any DC source

I've build my own tasmota32 firmware with TOUCH_PIN_THRESHOLD   set to 50 

https://github.com/arendst/Tasmota/blob/9a65d71ba9c82c50ca808653ffa6244b34bc005c/tasmota/support_button.ino#L28

TouchPins working now perfectly with all my ESP32 boards with battery or with USP-PC-Powered

would be great if there was an SetOption Parameter to have the possibility to change sensitivity of these Pins

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

closed 2020-11-15 by stale[bot]

czikus, 2020-11-23:
I noticed that TOUCH_PIN_THRESHOLD is too low for my device as well. Touch is not registered unless I set the value higher in the sources.

czikus, 2020-11-23:
It would be great to have the ability to configure the value.

tuct, 2021-04-21:
Running into the same issue, would be great if this can be set via a flag somehow.
Having to build an own version only to change this is a bit cumbersome.

Something similar to ButtonDebounce or a new SetOptionXX (TouchButtonTrashold) would be awesome