arendst/Tasmota #9412: Support for Shelly Dimmer 2

jamesturton, 2020-09-27:
**Have you looked for this feature in other issues and in the docs?**  

Yes, there is curently no support for the Shelly Dimmer 2

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

There is currently no support for the Shelly Dimmer 2. It would be nice for Tasmota to support the Shelly Dimmer 2.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

I would Tasmota to support the Shelly Dimmer 2 hardware.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Myself and others have worked hard to add support for the original Shelly Dimmer. The Shelly Dimmer 2, however, has some hardware changes thus needs its own support.

**Additional context**  
_Add any other context or screenshots about the feature request here._

This issue will be used to track mine, and others, work on making support for Shelly Dimmer 2 to happen.

**(Please, remember to close the issue when the problem has been addressed)**

mtroyernl, 2020-09-27:
Hi James,

I recently purchased a Shelly Dimmer 2 and am looking forward to work with it in Tasmota.  I just downloaded tasmota_shelly_dimmer_v0.8.bin.gz and loaded it on my SD2 with Tazmotizer expecting to go through the wifi SSID definition as well as the MQTT broker config.   The Wifi AP for the SD2 on this build does not show however after I load the firmware.

Am I missing a step?

Mark

jamesturton, 2020-09-28:
Hi @mtroyernl,

The firmware version v0.8 in issue [#6914](https://github.com/arendst/Tasmota/issues/6914) is for the original Shelly Dimmer, not the Shelly Dimmer 2. I have not yet tested that firmware on the Shelly Dimmer 2, nor have I started to reverse engineer the unit so I can't say what the expected behaviour of this would be.

However I would have thought the device would still start up and broadcast a WiFi AP even if the dimming function is unlikely to work...

Are you powering the device directly using 3.3V on the debug header when making these tests? Or after flashing, did you try powering from mains?

Is it possible you could try loading a normal version of Tasmota to see if the device starts up with that?

mtroyernl, 2020-09-28:
Hi James,

Not sure what happened.  I reflashed the code again and it worked this time--well sort of. 

With the Shelly Dimmer 2, the light came on but I observed the following:

1. The toggle button would not turn the light off.
2. The slider seemed to have the opposite function.  Sliding it to bright made it turn off.  Sliding it to dim, made it brighter.

If I can help with any other QA, please let me know!  Thanks for your work on this.  Great timing (for me)!

Mark

rca, 2020-09-29:
> [...] the Shelly Dimmer 2 has 7 extremely small female pin headers that require very high gage wire. It required a search effort to find some salvageable thin wire then to build a hook up jig to my usb serial module.

24 gauge CAT5 ethernet cable works great for the Shelly headers.

stefxx, 2020-10-07:
Can't wait to test this also. Tried the Dimmer version 0.8 just like @mtroyernl and got similar results. Off course I didn't expect it to work, but I am really looking forward for a specific Dimmer 2 version. Thanks for all your effort!

stefxx, 2020-10-07:
Oh, I forgot; this is what I use for those small pin headers connection: https://nl.aliexpress.com/item/32850757679.html. First perfectly.

joeH68, 2020-10-23:
Hi @jamesturton,

> There is currently no support for the Shelly Dimmer 2. It would be nice for Tasmota to support the Shelly Dimmer 2.

The great job done adding Sonoff ZigBee bridge support got me very excited about Tasmota, just to find out that my newly bought Shelly Dimmer 2 units aren't supported yet. Hope this support will be added soon. 

Does the Tasmota version supporting the original Shelly Dimmer "work with restrictions" or it just doesn't work? 
Thanks anyway for your support.

jamesturton, 2020-10-26:
So I have a first release for the Shelly dimmer 2 for testing!

[tasmota_shelly_dimmer_v1.0.bin.gz](https://github.com/arendst/Tasmota/files/5440406/tasmota_shelly_dimmer_v1.0.bin.gz)

This version of the driver v1.0 uses shelly-dimmer-stm32 version v51.0 which supports both Shelly dimmer 1 and 2. This is great as only one release is needed to support both these hardwares.

As this is the first release for the Shelly dimmer 2 there may be some issues with it so any feedback would be gratefully received.
Currently there is only really support for dimming, power monitoring (if possible) has not been implemented yet.

To save people from having to check issue #6914 I will list some installation tips here:
- For the first time installation of Tasmota, it is necessary to flash the ESP using the serial port and to pull GPIO_0 low.
- For subsequent updates it is possible OTA, but due to storing the stm32 firmware in the flash memory you must first upload `tasmota_minimal.bin.gz` to make some space in the flash memory and then upload `tasmota_shelly_dimmer.bin.gz` otherwise you will get the error `Upload Failed Upload buffer miscompare`.
- There is a special command `ShdLeadingEdge`. `ShdLeadingEdge 1` will set to leading edge dimming and `ShdLeadingEdge 0` will set it to trailing edge dimming. The default is trailing edge dimming.

Next steps:
- Investigate power monitoring support
- Investigate temperature monitoring
- Find which pin of the ESP32 is connected to the onboard LED.

kratz00, 2020-10-26:
@jamesturton Are you aware of https://github.com/Mollayo/Shelly-Dimmer-2
For temperature monitoring have a look at https://github.com/Mollayo/Shelly-Dimmer-2/blob/master/switches.cpp#L321

stefxx, 2020-10-26:
I loaded 1.0 into my Dimmer 2 succesfully. From the tasmota web interface, everything works fine. However, I can't seem to figure out how to configure the buttons. I like to control the dimmer with a single push button, but for now I would be happy it I got it to work with 2 push buttons. 

I will continue reading and testing, but if someone knows how to configure the buttons properly, let me know. And many, many thanks @jamesturton for your hard work!

stefxx, 2020-10-26:
Switchmode1 4 and switchmode2 4 enable me to turn the light on and off, with both buttons. No dimming. Tried many other options but local dimming doesn't work for me. So I either do not understand how it is suppose to work, or something is wrong.

jamesturton, 2020-10-26:
@kratz00 @Mollayo mentioned another one of their projects over in #6914 and provided some useful information on the difference in communication protocol between the esp8266 and stm32 when using stock firmware. Great to see someone else having fun hacking their Shelly dimmers! Looks like this should help with thermometer and LED pins!

@stefxx I am using `SwitchMode 11` with the following rule from the [Tasmota rules wiki page](https://tasmota.github.io/docs/Rules/#control-a-dimmer-with-one-switch) (even though it says `SwitchMode 5` on the wiki I use `SwitchMode 11` and `SetOption32 10`). Maybe this will help you getting it working?

```
ON switch1#state=2 DO publish cmnd/kitchenLights/POWER TOGGLE ENDON
ON switch1#state=4 DO publish cmnd/kitchenLights/DIMMER %mem1% ENDON
ON switch1#state=5 DO mem1 - ENDON
ON switch1#state=6 DO mem1 + ENDON
```

stefxx, 2020-10-28:
Hi! Sorry, I am still confused with the buttons. So I've set switchmode1 and switchmode2 both to 11. And put in the following rule:

rule1 on switch1#state do publish my/test1 %value% endon on switch2#state do publish my/test2 %value% endon

To my surprise, pressing button1 doesn't trigger the rule at all. But pressing button 2 publish the my/test**1** rule??

jamesturton, 2020-10-28:
@stefxx Thanks for testing! It looks like you are right - Shelly have printed the labels on the inputs the wrong way around! Looks like I will have to make a separate template for the Shelly dimmer 2 after all... Hopefully you can still continue testing knowing this?

stefxx, 2020-10-28:
Thanks. Note that SW2 is identified as SW1, but not the other way around. 

I will continue testing. For far, it seems to work just fine. Only issue I am having is that the workable dimming range is 60-100%. I am trying to build a rule that steps between 60-100, with steps smaller than the default 10. But this is not related to this firmware I guess.

Scoobler, 2020-10-29:
I've just loaded v 1.0 onto one of my Shelly D2's as a test - I'm struggling to get anything to work at the moment - the template available was the Shelly Dimmer 1 / 2. 

I currently have the switch into my Shelly Dimmer 2 as a live 240v input. I also use both switches with a momentary switch - SW1 short press is on, SW1 long press inc. brightness. SW2 short press is off, SW2 long press dec. brightness.

Although some places I don't want to rewire and only have feedback for a single switch, so using the rule above will work nicely. However I would like the ability to use N switching, instead of live switching - is that possible?

craigueh, 2020-10-29:
Is there any way for me to get this driver added to a dev branch 9.x of Tasmota? I'd like to use it with HASS's new Tasmota integration but needs a minimum of 9.0.0.3

jamesturton, 2020-10-29:
@Scoobler Glad you have got it working! I like the sound of your setup with the 'twin rocker switch'. For your single switch setup, do you have neutral connected?

@craigueh I can rebase my code against version 9.0.0.3 and make a new image for you to test with. Either tomorrow or at the weekend depending when I have enough time. I will have to do this anyway to get the code ready for a pull request soon too! If you can't wait that long then feel free to clone my branch of Tasmota here: https://github.com/jamesturton/Tasmota/tree/add_shelly_dimmer

@stefxx The first thing I would recommend you trying is to turn off the [Gamma Correction](https://tasmota.github.io/docs/Lights/#gamma-correction) with `LedTable 0`. I don't know who decided it should be on by default, but it is.
There are a couple more commands from the [Commands Page](https://tasmota.github.io/docs/Commands/#light) in the docs that might help you:
> DimmerRange: Change dimming range. < dimmerMin >,< dimmerMax > = set the internal dimming range from minimum to maximum value (0..100, 0..100)

> Fade: 0 = do not use fade (default), 1 = use fade See also SetOption91

Might be worth taking a read through that page as there is lots to learn!

stefxx, 2020-10-29:
Thanks @jamesturton  for the tips! I've read that page several times, played with dimmerrange (didn't do much, which makes sense as it clearly states it doesn't effect the Dimmer statement), speed and fade, as well, so I finally came up with a few rules (probably overly complicated but it works reasonably well):

rule1 on system#boot do backlog var1 up; var2 80 endon
on switch1#state=2 do POWER TOGGLE endon
on switch1#state=4 do backlog event updown=%var1%; DIMMER %var2% endon
on event#updown=down do sub2 4 endon
on event#updown=up do add2 4 endon
on var2#state>100 do var2 100 endon
on var2#state<60 do var2 60 endon
on switch1#state=5 do event upordown=%var1% endon
on switch1#state=6 do event upordown=down endon
on event#upordown=up do backlog var1 down; sub2 4 endon
on event#upordown=down do backlog var1 up; add2 4 endon

I've seen LedTable but never realised it would effect the actual dimmer range. Setting LedTable to Off increased the range from 60-100 to 25-100! So maybe I don't need those complicated rules after all... I'll do some more testing tomorrow. Thanks!

stefxx, 2020-10-29:
> Is it possible to change the rate the dimmer moves? At present it is 10 step increments, and jumps values very quickly. I guess it will take a rule to do this like @stefxx is looking for.

I guess we need to be able to set the rate, which is now fixed at 10 for + and -. A configuration option would be great!

stefxx, 2020-10-30:
> I guess we need to be able to set the rate, which is now fixed at 10 for + and -. A configuration option would be great!

https://github.com/arendst/Tasmota/issues/9669

stefxx, 2020-10-30:
> Thanks. Note that SW2 is identified as SW1, but not the other way around.

Ignore that comment. Turns out to be a broken switch. SW1 and SW2 are indeed simply reversed.

Scoobler, 2020-10-30:
@stefxx I have had a little play, I have used @jamesturton latest update from a couple of hours ago which now has the shelly dimmer 1 and 2 templates separate.

I have added a DimmerStep function, it allows a value from `1...10` it then uses the step value given on each step:

e.g.
```
14:59:31 CMD: dimmerstep 2
14:59:31 RSL: stat/tasmota_19F03D/RESULT = {"DimmerStep":2}
14:59:37 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
14:59:37 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"ON","Dimmer":64}
14:59:37 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
14:59:37 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"ON","Dimmer":66}
14:59:37 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
14:59:37 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"ON","Dimmer":68}
```

```
15:00:13 CMD: dimmerstep 5
15:00:13 RSL: stat/tasmota_19F03D/RESULT = {"DimmerStep":5}
15:00:18 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
15:00:18 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"ON","Dimmer":91}
15:00:18 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
15:00:19 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"ON","Dimmer":96}
15:00:19 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
15:00:19 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"ON","Dimmer":100}
```

@jamesturton this is where I could do with a spot of help, I used GitPod to do the changes, but then the skills have run out a little, I'm really not a GitHub user so not sure if I have done the right push back to you, or all the way up the tree 🤨 there is also the settings page - I'm not 100% I have adjusted the values correctly `settings.h`, things are working and prior to decrementing the number correctly nothing would save, so I went lower with the number and now everything seems to work.

```
  uint16_t      dimmer_step;             // F5E
  uint8_t        free_f5e[82];               // F6A - Decrement if adding new Setting variables just above and below
```

stefxx, 2020-10-30:
@Scoobler , I guess in order to try this I have to compile it myself right? Never done that, so I'll wait until there is a pre-compiled version to test. I will close https://github.com/arendst/Tasmota/issues/9669 as soon as I get it working.

Thanks!

Scoobler, 2020-10-30:
@stefxx I have a built [version here](https://we.tl/t-CeNs3toOvH), link will only last for a week. I have it loaded and working on my D2 on the bench.

Scoobler, 2020-10-30:
> @Scoobler Glad you have got it working! I like the sound of your setup with the 'twin rocker switch'. For your single switch setup, do you have neutral connected?
> 

@jamesturton I have all of my shelly's mounted in the ceiling by the light rose so there is a neutral, live and switched live there. Down to the light switch on the wall is just a twin and earth (live and switched live) for most of my older lights - hence the single switch use case. For the newer twin rockers I have a 3 core and earth - for these I still have the shelly in the ceiling rose, so neutral is available, for these I have used the neutral down to the switch and have two switched neutrals return.

stefxx, 2020-10-30:
@Scoobler , thanks! It works perfectly!

Scoobler, 2020-10-30:
@stefxx, further to #9669 I have just added another change, again pushed to @jamesturton, this one adds `DimmerStepRange`, defaults to `1...100` but can be changed to suit your dimming needs. I changed the memory location as I had used 16b int's and they didn't need to be, so when you load this over the earlier linked firmware, both `DimmerStep` and `DimmerStepRange` will need setting but would otherwise be set to a default of 10 for the steps and 1 to 100 for the range if it was a fresh install.

[Link Here](https://we.tl/t-LVWXTTO8Qr) - again only lasts a week..

```
16:53:05 CMD: dimmerstep 10
16:53:05 RSL: stat/tasmota_19F03D/RESULT = {"DimmerStep":10}
16:53:17 CMD: dimmersteprange 20,60
16:53:17 RSL: stat/tasmota_19F03D/RESULT = {"DimmerStepRange":{"Min":20,"Max":60}}
16:53:38 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
16:53:38 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":40}
16:53:39 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
16:53:39 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":50}
16:53:39 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
16:53:39 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":60}
16:53:40 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER +"
16:53:40 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":60}
...
16:54:13 RUL: SWITCH1#STATE=5 performs "mem1 -"
16:54:13 RSL: stat/tasmota_19F03D/RESULT = {"Mem1":"-"}
16:54:14 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER -"
16:54:14 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":50}
16:54:15 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER -"
16:54:15 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":40}
16:54:15 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER -"
16:54:15 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":30}
16:54:16 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER -"
16:54:16 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":20}
16:54:16 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_19F03D/DIMMER -"
16:54:16 RSL: stat/tasmota_19F03D/RESULT = {"POWER":"OFF","Dimmer":20}
```

stefxx, 2020-10-30:
@Scoobler , fantastic! It works perfect! Thank you so much!

@jamesturton , since I replaced the Dimmer 2 today (since I had issues with SW1, which turned out to be a bad switch) I noticed that the effective dimming range is quite different between the two. The "old" Dimmer 2 had a working range of about 30-90, the "new" one between 25-80. I purchased them both at the same time, no obvious differences. It there any calibration option that I missed? Or is this expected? The load was the same for both Dimmers.

jamesturton, 2020-10-31:
@craigueh Here is the Tasmota image for Shelly dimmer 1/2 based on Tasmota v9.0.0.3.
[tasmota_shelly_dimmer_v1.1.bin.gz](https://github.com/arendst/Tasmota/files/5468904/tasmota_shelly_dimmer_v1.1.bin.gz)
It seems that Tasmota now no longer accepts adding new device templates? Or maybe I've misunderstood the comments in the code, perhaps @ascillato2 or @arendst could explain why that is the case?

Because of this you should now use the following templates to configure the device:

**Shelly Dimmer 1**
```{"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5504,5536,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}```

**Shelly Dimmer 2**
```{"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5504,5536,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}```

@Scoobler Really nice work adding the `dimmerstep` and `dimmersteprange` commands! Looks like a really nice addition. Because this change would affect all dimmers used with Tasmota, I would say that these changes should be made in a separate pull request for clarity. If you would like, I could try setting up the pull request and making sure you are credited for your work? I actually think it would actually be nice if there was a separate issue tracking these changes: @stefxx is it possible for you to re-open #9669? Or maybe @ascillato2 could help?

@stefxx I'm not too sure why you are experiencing such a difference in performance between units - have you made sure you have set `LedTable 0` on both?

arendst, 2020-10-31:
The goal of tasmota now is to abandon internal templates as much as possible as it can be done externally most of the time leaving code space for more usefull items.

Since 9.0.0.1 the gpio numbers have also changed to cater for future gpio enhancements.

Regarding dimmerstep and dimmersteprange there must be overlap with the tuya code which already has range checks so a PR wouldn't be accepted as it is implemnted now (I guess, I never seen any Shelly dimmer code yet).

stefxx, 2020-10-31:
Thanks @jamesturton ! I have re-opened https://github.com/arendst/Tasmota/issues/9669.

Does the attached 1.1 still contain the DimmerStep and DimmerStepRange? Otherwise I will stay on 1.0 for now.

About the difference in performance between dimmers; I've read about calibration on the Shelly forums a few times, so that made me wonder if there is anything to calibrate with Tasmota...

jamesturton, 2020-10-31:
@stefxx v1.1 does not contain the `DimmerStep` or `DimmerStepRange` commands - I will keep these on a separate branch for now.

I agree with @arendst that `DimmerRange` (which exists currently) is too similar to `DimmerStepRange` so I shall implement `DimmerRange` in the Shelly driver. But it is obvious that there is no similar command to `DimmerStep` so this it still a very useful addition in my option rather than having a 'magic number' in the code that can never be changed.

stefxx, 2020-10-31:
@jamesturton Ok thanks. Are you sure DimmerRange and DimmerStepRange are overlapping? As the documentation clearly states that DimmerRange does NOT effect the Dimmer statement.  If the Dimmer command will also take DimmerRange into consideration, that would be good as well off course (and that would make perfect sense, but might not be backwards compatible?)

Scoobler, 2020-10-31:
@jamesturton I agree on the DimmerStepRange - after doing that I thought about it, we could just use the DimmerRange min and max values, instead of adding the extra ones - reduces the extra code and essentially shares a feature between Tuya and Shelly then.  

@arendst The DimmerRange I think is used somewhere else for the Tuya dimmers, however the function to handle + / - command of a dimmer is a little hard coded. It allows a dimmer dimmer to pass the DimmerRange - I guess somewhere in the Tuya code that gets caught and limited, maybe @s-hadinger can shed some light, I think he may have done some of the implementation of the special function?

```// Handle +/- special command
    if (1 == XdrvMailbox.data_len) {
      uint8_t channel = changeUIntScale(Light.current_color[light_index],0,255,0,100);
      if ('+' == XdrvMailbox.data[0]) {
        XdrvMailbox.payload = (channel > 89) ? 100 : channel + 10;
      } else if ('-' == XdrvMailbox.data[0]) {
        XdrvMailbox.payload = (channel < 11) ? 1 : channel - 10;
      }
    }
```
In this snippet you can see the Min / Max limit is hard coded to `1` / `100` - this would be a very easy change - swapping the hard coded `1` & `11` for `Settings.dimmer_hw_min` _(and a calculation of)_ and `89` & `100` for `Settings.dimmer_hw_man` _(and a calculation of)_ would mean the function uses the DimmerRange min and max values and ensures the function doesn't drive the dimmer outside the set range.

The DimmerStep essentially changes the hardcoded `10` step to a value between `1..10` to allow you to slow the speed the function changes the brightness of the light.

```
 // Handle +/- special command
  if (1 == XdrvMailbox.data_len) {
    if ('+' == XdrvMailbox.data[0]) {
      XdrvMailbox.payload = (dimmer > (Settings.dimmer_hw_max - (Settings.dimmer_step - 1))) ? Settings.dimmer_hw_max : dimmer + Settings.dimmer_step;
    } else if ('-' == XdrvMailbox.data[0]) {
      XdrvMailbox.payload = (dimmer < (Settings.dimmer_hw_min + (Settings.dimmer_step + 1))) ? Settings.dimmer_hw_min : dimmer - Settings.dimmer_step;
    }
  }
```

I don't think this would have any effect on the Tuya code as it essentially doesn't change anything - default values remain at `0..100` for the range and `10` for the step value, it just allows fine tuning of how the function works - something I think both Shelly and Tuya would benefit from.

jamesturton, 2020-11-03:
Here is version 1.2 of the Shelly dimmer Tasmota driver if anyone would like to test?
[tasmota_shelly_dimmer_v1.2.bin.gz](https://github.com/arendst/Tasmota/files/5482704/tasmota_shelly_dimmer_v1.2.bin.gz)

This release includes power monitoring on the Shelly dimmer 2. I would be interested to see how accurate the readings are for people with different mains voltages other than 240v.

I have also implemented constraining the dimmer value between `dimmer_hw_min` and `dimmer_hw_max`. What this means is that the dimmer value can still be set to a value anywhere between 0 and 100, but this will be scaled using the `changeUIntScale()` function to between the `dimmer_hw_min` and `dimmer_hw_max` values.

If there are no issues found with this version then I intend to submit a pull request to get the branch included in the main codebase for Tasmota!

stefxx, 2020-11-03:
Great work @jamesturton ! Just installed this on one of my Dimmer 2 devices and it seems to work just fine. Really miss the DimmerStep but having the DimmerRange effecting the Dimmer command makes up for it a little. And hopefully DimmerStep will be implemented as well by @Scoobler soon.

Thanks again!

kratz00, 2020-11-03:
@jamesturton 
I am currently using 1.0 , can I just OTA/upload upgrade or do I need to upgrade to tasmota_minimal.bin.gz first, like you mentioned earlier?

jamesturton, 2020-11-03:
@kratz00 you will have to upload a smaller Tasmota image first

craigueh, 2020-11-04:
@jamesturton Thanks for this. Seems to work great so far. 

Can you tell me what the differences are in commands to control the dimmer 2 vs the dimmer 1?
I’m working on another project to control a dimner2 and can’t find anywhere the differences are explained so I can adjust my code. I have full control over a dimmer1 but the same code doesn’t work for the 2.

jamesturton, 2020-11-04:
@craigueh This driver for the Shelly dimmer 2 uses a custom firmware on the stm32 which uses the same communication protocol as on the dimmer 1, so I won't be able to help very much unfortunately.

Others have tried reverse engineering the communication protocol like here: https://github.com/Mollayo/Shelly-Dimmer-2
Maybe it would be best to look their instead? @Mollayo has also made a repo especially for reverse engineering the communication protocol which looks really interesting! https://github.com/Mollayo/Shelly-Dimmer-2-Reverse-Engineering

If you would like you are more than welcome to use the open source firmware for the stm32 which has been used in the Tasmota driver here: https://github.com/jamesturton/shelly-dimmer-stm32/tree/shelly-dimmer-2

craigueh, 2020-11-04:
Are you saying the dimmer 1 and 2 both use an stm32 but the 2 has a different firmware from the factory? Just trying to get my head around the differences. 

So would flashing your tasmota build also flash the stm32 back to a dimmer 1 compatible firmware? In this case my code for a dimmer 1 should work as the stm32 is using equivalent of dimmer 1 firmware?

Or have I got it all wrong? :-)

jamesturton, 2020-11-04:
@stefxx Here is a new version with the newly included `DimmerStep` command :)
[tasmota_shelly_dimmer_v1.3.bin.gz](https://github.com/arendst/Tasmota/files/5489877/tasmota_shelly_dimmer_v1.3.bin.gz)

I'm not sure if you know, but another way you can make the transitions smoother is to use the [`Fade`](https://tasmota.github.io/docs/Commands/#fade) and [`Speed`](https://tasmota.github.io/docs/Commands/#speed) commands which work with `Dimmer +/-`

@craigueh 
> Are you saying the dimmer 1 and 2 both use an stm32 but the 2 has a different firmware from the factory?

I believe so, or it sends different commands when it knows it's loaded in a Shelly dimmer 2. Or maybe they now use the same protocol, but they didn't when I checked, I haven't bothered checking on the newer firmware versions.

> So would flashing your tasmota build also flash the stm32 back to a dimmer 1 compatible firmware? In this case my code for a dimmer 1 should work as the stm32 is using equivalent of dimmer 1 firmware?

Correct, this version of Tasmota actually flashes the stm32 with the new version of firmware. So loading Tasmota and then loading your own software to the ESP should work fine providing Shelly didn't change the protocol they have used on the later version of the of dimmer 1. If in doubt you can also check the `payload format` in the Tasmota driver code here: https://github.com/jamesturton/Tasmota/blob/add_shelly_dimmer/tasmota/xdrv_45_shelly_dimmer.ino

stefxx, 2020-11-04:
Thanks @jamesturton ! This means I can install the firmware on 4 more dimmers around the house since they passed the WAF with DimmerStep! ;)

craigueh, 2020-11-04:
@jamesturton I tried flashing your firmware 1.3 to a dimmer 2 and then reflashing with my other firmware but it still doesn't communicate with the stm32. Never mind, I may be able to use your tasmota build to do what I want.
Is it possible to use the two sw1 and sw2 inputs as something else in homeassistant and not to control the dimmer switch?
If I could just have the two switch inputs show as binary switches in HASS I can use automations to take care of the rest.

jamesturton, 2020-11-04:
@craigueh you should be able to manage that with [Switch Topics](https://tasmota.github.io/docs/Buttons-and-Switches/#switchtopic) and MQTT. I assume you are using MQTT to control Tasmota devices from HomeAssistant?

craigueh, 2020-11-04:
Yep I’m using mqtt. I had tried switchtopic 1 but didn’t realise you could send a custom topic and not change the power state. Thanks for the help.

stefxx, 2020-11-04:
@craigueh I use Switchmode1 11 for SW1, so that controls the dimmer. And Switchmode2 6 for SW2 so I have single, double and long press reported through MQTT.

craigueh, 2020-11-04:
Thanks @stefxx I'll give that a try and see if it works.

ChristusM, 2020-11-04:
At some point will the changes in tasmota_shelly_dimmer_v1.2.bin.gz be incorporated into the general Tasmota releases?

jamesturton, 2020-11-04:
@ChristusM yes, I am hoping to submit a merge request for these changes very soon. Once I have had some feedback from a few more people and I have done a bit more testing and I am confident that the driver works well I shall submit the PR.

ChristusM, 2020-11-04:
@jamesturton thank you.

Scoobler, 2020-11-04:
@jamesturton great work on the power monitoring 👍 I’ll do some testing with loads.

craigueh, 2020-11-04:
@jamesturton i notice my power calculation is off. With no load the device is currently showing 741.104 kWH energy used today. :-)

daxy01, 2020-11-05:
@jamesturton I'd love to try and test to give more confidence, but as my Dimmer (1) is in the kitchen and not very easily reachable, I need to have some level of confidence before I can flash. I've got a couple of Shelly 1 & 2.5's running Tasmota already. 

Can I just flash this image, configure it like the Shelly 1/2.5's, add it to HomeAssistant and it will just work (incl dimming)?

jamesturton, 2020-11-05:
@daxy01 This issue is for the Shelly dimmer 2. You should ask your question in issue #6914 for the Shelly dimmer 1.

@craigueh Maybe you can try [`EnergyReset1`](https://tasmota.github.io/docs/Commands/#power-monitoring)? Does the current power look correct?

daxy01, 2020-11-05:
I thought it was the same firmware file isn't it? https://github.com/arendst/Tasmota/issues/6914 only has v1.0 and not v1.3.
I'll ask there as well ;)

craigueh, 2020-11-05:
@jamesturton I tried EnergyReset1 but it didn’t seem to make any difference. Current power usage looks about right, 28w for 6 gu10 spots. It’s just the totals that are way off. 
It’s not a big deal for me but others might be more keen to have the totals working.

stefxx, 2020-11-05:
I had kWh usage for today and total (not yesterday) after initial installation. EnergyReset1 and EnergyReset3 solved this for me.

craigueh, 2020-11-05:
I tried both of those but doesn’t seem to have changed anything. Energy totals still show as 24291.527 kWh

[42 more replies]

closed 2020-11-16 by ascillato2