arendst/Tasmota #9416: Zigbee ZbPermitJoin incorrect behaviour

Courty40, 2020-09-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
ZbPermitJoin incorrect behaviour - doesnt matter it you send a 1 or 0 it just enables Join

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ZBBridge_____
- [x ] Tasmota binary firmware version number used:  Latest
 - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used:  Upload/OTA as per instruction_____
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:53:48 CMD: Backlog Template; Module; GPIO 255
11:53:48 MQT: stat/ZIGBEE_D7F7FA/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:53:49 MQT: stat/ZIGBEE_D7F7FA/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
11:53:49 MQT: stat/ZIGBEE_D7F7FA/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"165":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"166":"Zigbee Rx"},"GPIO4":{"215":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"158":"LedLinki"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}

```
- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

none
```
- [ x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

11:54:26 CMD:   STATUS 0
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"ZIGBEE_D7F7FA","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin","RestartReason":"Power On","Uptime":"1T20:14:11","StartupUTC":"2020-09-26T14:40:15","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2020-09-05T15:16:40","SaveCount":30,"SaveAddress":"F7000"}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS2 = {"StatusFWR":{"Version":"8.5.0(zbbridge)","BuildDateTime":"2020-09-09T11:44:49","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"353/699"}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hamara-IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006002","00000000"]}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":472,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F100786","04500001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,21,23,41","Sensors":""}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS5 = {"StatusNET":{"Hostname":"ZIGBEE_D7F7FA-6138","IPAddress":"192.168.111.134","Gateway":"192.168.111.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.111.254","Mac":"A4:CF:12:D7:F7:FA","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.111.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D7F7FA","MqttUser":"DVES_USER","MqttCount":7,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS7 = {"StatusTIM":{"UTC":"2020-09-28T10:54:26","Local":"2020-09-28T11:54:26","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:46","Sunset":"18:34"}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS10 = {"StatusSNS":{"Time":"2020-09-28T11:54:26"}}
11:54:26 MQT: stat/ZIGBEE_D7F7FA/STATUS11 = {"StatusSTS":{"Time":"2020-09-28T11:54:26","Uptime":"1T20:14:11","UptimeSec":159251,"Vcc":3.438,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":7,"Wifi":{"AP":1,"SSId":"Hamara-IOT","BSSId":"02:1D:AA:3C:4F:7E","Channel":9,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
11:44:28 CMD: ZbPermitJoin 0
11:44:28 MQT: stat/ZIGBEE_D7F7FA/RESULT = {"ZbPermitJoin":"Done"}
11:44:28 MQT: tele/ZIGBEE_D7F7FA/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
11:44:28 ZIG: {"ZbEZSPReceived":"45000500003600000000000000B9FF000000FFFF036D0001"}
11:44:28 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Send Join commands. happens 100% of the time

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

1 to enable 0 to disable

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Courty40, 2020-09-28:
looks similar to #9256 bit expected behaviour not the same as instructions

ascillato2, 2020-09-28:
Please try latest development version. A lot of fixes has been introduced since the release.

Courty40, 2020-09-28:
I can conform that the latest DEV version does indeed work as expected..
I can see a few other interesting bits too !

Any ETA for the official release of this version ?

Many Thanks.

```
14:47:17 CMD: ZbPermitJoin 0
14:47:17 MQT: stat/ZIGBEE_D7F7FA/RESULT = {"ZbPermitJoin":"Done"}
14:47:17 MQT: tele/ZIGBEE_D7F7FA/RESULT = {"ZbState":{"Status":20,"Message":"Pairing mode disabled"}}
14:47:17 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
```

Jason2866, 2020-09-28:
Next planned release is v9.0 due by November 05, 2020

Anyway the development zigbee version has new features and bug fixes since release v.8.5 
So there is no reason against to use.

closed 2020-09-28 by ascillato2

pickonedev, 2021-10-30:
I use Tasmota 10.0.0 by Theo Arends and the problem persists...

I have two ZbBridges and I want to split the number of devices 50% for one ZbBridge and 50% for another ZbBridge... But when I use one device from the first ZbBridce, it will register on the second one as well, pairing it to the Bridge...

There is any method in solve this issue??

s-hadinger, 2021-10-30:
If you have two coordinators with the same panid and the same encryption key, it is a violation of zigbee protocol and only bad things can happen. Is it your case?

pickonedev, 2021-10-30:
In my case, one bridge steals from another and so on... If I add a device to fist bridge, when I use that device, the second bridge will catch it too and then it will not work with the first bridge anymore, till I remote the device from the first one, then it will steal it from the second, and so on :-D Thats bad... 

It's bad because as far as I know, zigbee bridge sonoff can handle only 32 devices, thats why I try to use two bridges... but hard fail...

s-hadinger, 2021-10-30:
You are not answering my question. Do both coordinators have same panid? If yes you need to change this asap. 
And please stop cross posting

pickonedev, 2021-10-30:
Yes, I forgot... Same PanID, I changed it now and all good. Sorry