arendst/Tasmota #9419: ESP32, Guru Meditation Error, rebooting

GrendelWWU, 2020-09-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
device restarts in combination receiving IR commands

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU Developement Kit ESP32 (AZ-Delivery)
- [x] Tasmota binary firmware version number used: 8.5.0.1 (ir), 28.09.2020, 12:52:50 
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: initial epstool.py, latest version flashed OTA (webGUI/Upgrade)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`: 
14:54:13 CMD: Backlog Template; Module; GPIO 255
14:54:13 RSL: RESULT = {"NAME":"ESP32-Vogel","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
14:54:13 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
14:54:13 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"225":"Relay2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"544":"LedLink"},"GPIO17":{"1088":"IRrecv"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"256":"Relay_i1"},"GPIO33":{"192":"Switch_n1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}
```
  Configuration output here:
14:54:13 CMD: Backlog Template; Module; GPIO 255
14:54:13 RSL: RESULT = {"NAME":"ESP32-Vogel","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
14:54:13 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
14:54:13 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"225":"Relay2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"544":"LedLink"},"GPIO17":{"1088":"IRrecv"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"256":"Relay_i1"},"GPIO33":{"192":"Switch_n1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
14:55:30 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":52,"Free":459,"Rules":"on IrReceived#Data=0x00FFA25D do power1 toggle endon"}
14:55:30 RSL: RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":51,"Free":460,"Rules":"on IrReceived#Data=0x00FF38C7 do power2 blink endon"}
14:55:30 RSL: RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":150,"Free":361,"Rules":"on IrReceived#Data=0x00FFE21D do mem1 1 endon  on IrReceived#Data=0x00FF22DD do mem1 2 endon on IrReceived#Data=0x00FF9867 do power%mem1% toggle endon"}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
14:56:35 CMD: Status 0
14:56:35 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"ESP32-01","FriendlyName":["ESP32-01","Tasmota2"],"Topic":"tasmota_971EE4","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:56:35 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-ir.bin","RestartReason":"Software reset digital core","Uptime":"0T00:13:22","StartupUTC":"2020-09-28T13:43:13","Sleep":50,"CfgHolder":4617,"BootCount":79,"BCResetTime":"2020-09-14T19:52:49","SaveCount":729}}
14:56:35 RSL: STATUS2 = {"StatusFWR":{"Version":"8.5.0.1(ir)","BuildDateTime":"2020-09-28T12:52:50","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"443/699"}}
14:56:35 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["geleglu-point-01",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100FF0000005A00000000000000","00000008","00006000","00000000"]}}
14:56:35 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1075,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F5A0582","00000001","00000002","00000000","00000000","00000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12","Sensors":"2"}}
14:56:35 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_971EE4-7908","IPAddress":"10.11.12.14","Gateway":"10.11.12.254","Subnetmask":"255.255.255.0","DNSServer":"10.11.12.254","Mac":"24:6F:28:97:1E:E4","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:56:35 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-09-28T13:56:35","Local":"2020-09-28T14:56:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:46","Sunset":"18:34"}}
14:56:35 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-09-28T14:56:35","Switch1":"ON"}}
14:56:35 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-09-28T14:56:35","Uptime":"0T00:13:22","UptimeSec":802,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"geleglu-point-01","BSSId":"DC:39:6F:AE:B8:45","Channel":1,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
14:57:16 CMD: weblog 4
14:57:16 RSL: RESULT = {"WebLog":4}
14:57:32 RSL: RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FF9867","DataLSB":"0x00FF19E6","Repeat":0}}
14:57:32 RUL: IRRECEIVED#DATA=0X00FF9867 performs "power2 toggle"
14:57:32 RSL: RESULT = {"POWER2":"OFF"}
14:57:32 RSL: POWER2 = OFF
14:57:34 RSL: RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FF9867","DataLSB":"0x00FF19E6","Repeat":0}}
14:57:34 RUL: IRRECEIVED#DATA=0X00FF9867 performs "power2 toggle"
14:57:34 RSL: RESULT = {"POWER2":"ON"}
14:57:34 RSL: POWER2 = ON
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
[... 39 lines cut ...]
14:57:42 RSL: RESULT = {"POWER1":"ON"}
14:57:42 RSL: POWER1 = ON
14:57:42 RSL: RESULT = {"POWER2":"ON"}
14:57:42 RSL: POWER2 = ON
14:57:46 RSL: STATE = {"Time":"2020-09-28T14:57:46","Uptime":"0T00:00:13","UptimeSec":13,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":145,"MqttCount":0,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"geleglu-point-01","BSSId":"DC:39:6F:AE:B8:45","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:08"}}
14:57:46 RSL: SENSOR = {"Time":"2020-09-28T14:57:46","Switch1":"ON"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just pressing the '0' button of my IR remote some times to toggle Power1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect no crashing / rebooting.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
this crashing/rebooting appears with all flashed versions so far. The output is done with serial console, cause it provides more information (register output)

**(Please, remember to close the issue when the problem has been addressed)**

GrendelWWU, 2020-10-02:
Hello, I've tested 8.5.1 and this one, too;

Program Version | 9.0.0.1(ir)
-- | --
2020-10-02T10:44:16
1_0_4/v3.2.3-14-gd3e562907

Same problem....
Yes, I know, it's beta. Just wanted to let you know.

Jason2866, 2020-10-02:
Thx for reporting. The issue will be left open until (hopefully soon) it is solved

GrendelWWU, 2020-11-04:
V 9.0.0.3(ir) @ 2020-11-03T19:16:51still crashing...

arendst, 2020-11-04:
Just did a test with RC5 remote which went well:
```
17:55:46 MQT: tele/esp32c/INFO1 = {"Module":"ESP32-Devkit","Version":"9.0.0.3(tasmota)","FallbackTopic":"cmnd/DVES_36075C_fb/","GroupTopic":"cmnd/tasmotas/"}
17:55:46 MQT: tele/esp32c/INFO2 = {"WebServerMode":"Admin","Hostname":"esp32c","IPAddress":"192.168.2.145"}
17:55:46 MQT: tele/esp32c/INFO3 = {"RestartReason":"Software reset CPU"}
17:55:48 HTP: Main Menu
17:55:48 QPC: Reset
17:55:50 APP: Boot Count 100
17:55:51 CFG: Saved, Count 238, Bytes 4096
17:55:51 MQT: tele/esp32c/STATE = {"Time":"2020-11-04T17:55:51","Uptime":"0T00:00:11","UptimeSec":11,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"MqttCount":1,"Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:55:51 MQT: tele/esp32c/SENSOR = {"Time":"2020-11-04T17:55:51","DS18B20":{"Id":"01144A0CB2AA","Temperature":23.9},"TempUnit":"C"}
17:55:56 IRR: Echo 0, RawLen 26, Overflow 0, Bits 12, Value 0x803, Decode 1
17:55:56 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T17:55:56","IrReceived":{"Protocol":"RC5","Bits":12,"Data":"0x803"}}
17:55:56 IRR: Echo 0, RawLen 26, Overflow 0, Bits 12, Value 0x803, Decode 1
17:55:56 IRR: Echo 0, RawLen 26, Overflow 0, Bits 12, Value 0x803, Decode 1
17:55:58 IRR: Echo 0, RawLen 26, Overflow 0, Bits 12, Value 0x001, Decode 1
17:55:58 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T17:55:58","IrReceived":{"Protocol":"RC5","Bits":12,"Data":"0x001"}}
17:55:59 IRR: Echo 0, RawLen 26, Overflow 0, Bits 12, Value 0x001, Decode 1
17:55:59 IRR: Echo 0, RawLen 26, Overflow 0, Bits 12, Value 0x001, Decode 1
```
What happens if you disable rules and press some keys on the remote?

arendst, 2020-11-04:
Just found a remote that does NEC. Works OK too:
```
18:00:00 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00F750AF, Decode 3
18:00:00 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:00:00","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F750AF"}}
18:00:01 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00F750AF, Decode 3
18:00:01 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:00:01","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F750AF"}}
18:00:02 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00F750AF, Decode 3
18:00:02 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:00:02","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F750AF"}}
```
If you connect relays to the device do you use a decent power supply? A simple USB power connection won't do as it delivers not enough power to drive the relays.

Jason2866, 2020-11-04:
@arendst  since you are on. The failing version is variant IR

arendst, 2020-11-04:
Did you verify the fault?

Jason2866, 2020-11-04:
No! Tried with my Odroid with a added IR receiver. Could not reproduce

arendst, 2020-11-04:
BTW I don't think your Switch1 will work as it is connected to an Analog Input only GPIO.

arendst, 2020-11-04:
Just tried with your rules too. Again no issue at all:
```
18:16:41 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:16:41","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F76897"}}
18:16:41 RUL: IRRECEIVED#DATA=0X00F76897 performs "power1 toggle"
18:16:41 MQT: stat/esp32c/RESULT = {"POWER1":"ON"}
18:16:41 MQT: stat/esp32c/POWER1 = ON
18:16:45 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:16:45","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F76897"}}
18:16:45 RUL: IRRECEIVED#DATA=0X00F76897 performs "power1 toggle"
18:16:45 MQT: stat/esp32c/RESULT = {"POWER1":"OFF"}
18:16:45 MQT: stat/esp32c/POWER1 = OFF
18:16:53 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:16:53","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F76897"}}
18:16:53 RUL: IRRECEIVED#DATA=0X00F76897 performs "power1 toggle"
18:16:53 MQT: stat/esp32c/RESULT = {"POWER1":"ON"}
18:16:53 MQT: stat/esp32c/POWER1 = ON
18:16:55 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:16:55","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F76897"}}
18:16:55 RUL: IRRECEIVED#DATA=0X00F76897 performs "power1 toggle"
18:16:55 MQT: stat/esp32c/RESULT = {"POWER1":"OFF"}
18:16:55 MQT: stat/esp32c/POWER1 = OFF
18:16:57 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:16:57","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F7A857"}}
18:16:57 RUL: IRRECEIVED#DATA=0X00F7A857 performs "mem1 2"
18:16:57 MQT: stat/esp32c/RESULT = {"Mem1":"2"}
18:16:59 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:16:59","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F76897"}}
18:16:59 RUL: IRRECEIVED#DATA=0X00F76897 performs "power2 toggle"
18:16:59 MQT: stat/esp32c/RESULT = {"POWER2":"ON"}
18:16:59 MQT: stat/esp32c/POWER2 = ON
18:17:01 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:17:01","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F76897"}}
18:17:01 RUL: IRRECEIVED#DATA=0X00F76897 performs "power2 toggle"
18:17:01 MQT: stat/esp32c/RESULT = {"POWER2":"OFF"}
18:17:01 MQT: stat/esp32c/POWER2 = OFF
18:17:02 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:17:02","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F728D7"}}
18:17:02 RUL: IRRECEIVED#DATA=0X00F728D7 performs "mem1 1"
18:17:02 MQT: stat/esp32c/RESULT = {"Mem1":"1"}
18:17:04 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:17:04","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F76897"}}
18:17:04 RUL: IRRECEIVED#DATA=0X00F76897 performs "power1 toggle"
18:17:04 MQT: stat/esp32c/RESULT = {"POWER1":"ON"}
18:17:04 MQT: stat/esp32c/POWER1 = ON
18:17:06 MQT: tele/esp32c/RESULT = {"Time":"2020-11-04T18:17:06","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00F76897"}}
18:17:06 RUL: IRRECEIVED#DATA=0X00F76897 performs "power1 toggle"
18:17:06 MQT: stat/esp32c/RESULT = {"POWER1":"OFF"}
18:17:06 MQT: stat/esp32c/POWER1 = OFF
```

Jason2866, 2020-11-04:
Regarding power supply. Have you tried with disconnected relais?
ESP32 is very pitty with the power supply. 
Not only voltage and current, it reacts very badly with short spikes or noise.
A cheap USB wall wart or Smartphone charger will often not work

GrendelWWU, 2020-11-04:
Hello dear developers,

all relays are disconnected, the problem still persists. Just driving a few LEDs.

My rules so far:

> 20:50:13 CMD: Group 0, Index 1, Command "RULE", Data ""
> 20:50:13 RUL: Rule1 on System#Boot do backlog var1 1;blinkcount 1 endon on IrReceived#Data=0x00FFA25D do backlog power1 toggle;var1 1;blinkcount 1 endon on IrReceived#Data=0x00FF629D do backlog power2 toggle;var1 2;blinkcount 2 endon on IrReceived#Data=0x00FFE21D do backlog power3 toggle;var1 3;blinkcount 3 endon on IrReceived#Data=0x00FF22DD do backlog power4 toggle;var1 4;blinkcount 4 endon on IrReceived#Data=0x00FF02FD do backlog power5 toggle;var1 5;blinkcount 5 endon on IrReceived#Data=0x00FFC23D do backlog power6
> 20:50:13 RUL: Rule1 +toggle;var1 6;blinkcount 6 endon
> 20:50:13 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":537,"Free":327,"Rules":"on System#Boot do backlog var1 1;blinkcount 1 endon on IrReceived#Data=0x00FFA25D do backlog power1 toggle;var1 1;blinkcount 1 endon on IrReceived#Data=0x00FF629D do backlog power2 toggle;var1 2;blinkcount 2 endon on IrReceived#Data=0x00FFE21D do backlog power3 toggle;var1 3;blinkcount 3 endon on IrReceived#Data=0x00FF22DD do backlog power4 toggle;var1 4;blinkcount 4 endon on IrReceived#Data=0x00FF02FD do backlog power5 toggle;var1 5;blinkcount 5 endon on IrReceived#Data=0x00FFC23D do backlog power6 toggle;..."}
> 20:50:14 WIF: Checking connection...
> 20:51:05 CMD: Group 0, Index 2, Command "RULE", Data ""
> 20:51:05 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":313,"Free":198,"Rules":"on IrReceived#Data=0X00FF38C7 do backlog sleep;power%var1% toggle endon on IrReceived#Data=0x00FFB04F do backlog sleep;power1 on; power2 on; power3 on; power4 on; power5 on; power6 on endon on IrReceived#Data=0x00FF6897 do backlog sleep;power1 off; power2 off; power3 off; power4 off; power5 off; power6 off endon"}
> 20:51:14 WIF: Checking connection...
> 20:51:42 RSL: RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":363,"Free":148,"Rules":"on IrReceived#Data=0x00FF18E7 do power%var1% on endon on IrReceived#Data=0x00FF4AB5 do power%var1% off endon on IrReceived#Data=0X00FF9867 do backlog power7 blink;power8 blink;power9 blink endon on Var1#State do backlog blinkcount %Var1;blinktime 3;power7 blink endon on IrReceived#Data=0x00FF10EF do Dimmer - endon on IrReceived#Data=0x00FF5AA5 do Dimmer + endon"}
> 20:51:54 WIF: Checking connection...

When disabling ALL rules, it's only showing the IR codes.

Configuration: 

> 21:00:35 CMD: Group 0, Index 1, Command "STATUS", Data "0"
> 21:00:35 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"ESP32-01","FriendlyName":["ESP32-01","Tasmota2","Tasmota3","Tasmota4","Tasmota5","Tasmota6","Tasmota7","Tasmota8"],"Topic":"tasmota_971EE4","ButtonTopic":"0","Power":319,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> 21:00:35 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-ir.bin","RestartReason":"Software reset digital core","Uptime":"0T00:12:11","StartupUTC":"2020-11-04T19:48:24","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-09-14T19:52:49","SaveCount":57}}
> 21:00:35 RSL: STATUS2 = {"StatusFWR":{"Version":"9.0.0.3(ir)","BuildDateTime":"2020-11-03T19:16:51","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"452/699"}}
> 21:00:35 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["geleglu-point-01",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","280501000100FF0000005A00000000000000","00000009","00006000","00000000"]}}
> 21:00:35 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1080,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F5A0582","00000001","00000002","00000000","00000000","00000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12","Sensors":"2"}}
> 21:00:35 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_971EE4-7908","IPAddress":"10.11.12.14","Gateway":"10.11.12.254","Subnetmask":"255.255.255.0","DNSServer":"10.11.12.254","Mac":"24:6F:28:97:1E:E4","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
> 21:00:35 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-11-04T20:00:35","Local":"2020-11-04T21:00:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:51","Sunset":"17:15"}}
> 21:00:35 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-11-04T21:00:35","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","Switch5":"ON","Switch6":"ON"}}
> 21:00:35 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-11-04T21:00:35","Uptime":"0T00:12:11","UptimeSec":731,"Heap":111,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","POWER5":"ON","POWER6":"ON","POWER7":"OFF","POWER8":"OFF","POWER9":"ON","Dimmer":80,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"geleglu-point-01","BSSId":"DC:39:6F:AE:B8:45","Channel":1,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:07"}}}
> 21:00:54 WIF: Checking connection...
> 21:02:36 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
[... cut ...]
I've been assuming problems with power supply. But it doesn't make a change. I've been using a supply, that will deliver more then 1A current. Without relais, it doesn't pull more than 100mA, with 6 relays closed, it's current is about 0.5A.

I am experiencing this with 2 different ESP32 boards. I've been using many different pinouts.

When driving just one LED, the problem is still there, even with the strong power supply.
I will try to add some capacitors tomorrow, buffering the 3V part.

I don't have a clue, what's going on there.

Best regards,
Thomas

GrendelWWU, 2020-11-04:
Another point.... when performing power actions with the web interface, I won't reset the board.
For example, switching all 6 relays on and then performing on/off dozens of times rapidly....
When doing this with IR, it often crashes, when doing this with web interface, it's reacting as expected, without any problems.
My IR receiver is an TSOP4838 from Vishay. No matter if I'm using one or two (parallel)...
After testing with capacitor, I will try another receiver (no name).
But then, I'll be at the end of my wisdom....

GrendelWWU, 2020-11-04:
Another observation:
When doing a power blink (with the 0 button of the remote), the LED outputs a blinking as I want them to.
When powering off a powered on output, it makes sometimes trouble. When powering off an already off output, no problem. Setting power on is making more trouble. Spooky!

Jason2866, 2020-11-04:
1A power supply does say nothing about the quality. And for a ESP32 1A is not that much.
Use a known to be good power supply. Labor power supply would be good.
The weird issues you have are pointing in this direction.

GrendelWWU, 2020-11-05:
Okay, now I disconnected the LED, the relays, I am using a 1000µF capacitor and a good power supply for 3V path, so the ESP32 will have an excellent regulated voltage. But it's still crashing. I stripped this down and only the TX-LED onboard and two other Signal LEDs (LEDLINK and Power7) via 3k3 resistor (!) will be connected. It's still crashing. Sometimes after 1-5 toggles, sometimes after 20-30....

GrendelWWU, 2020-11-05:
Hello Jason, I'M using a Manson HSP2020 Labor power supply. And yes, I added a cap for 3V. 1000µF would be enough for an amperage of 200mA when driving this LED stuff.

GrendelWWU, 2020-11-05:
Okay, I've taken another different ESP32 board (AZ-Delivery ESPDevKit_V4 - DualCore). Just connected USB Power and IR-sensor. Minimal Rules....

> 16:49:41 CMD: rule1
16:49:41 RSL: stat/tasmota_B61D90/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":80,"Free":431,"Rules":"on IrReceived#Data=0x00FFA25D do backlog power1 toggle;var1 1;blinkcount 1 endon"}

Just configured LEDLINK. Power and IR-Receiver. Just connected IR-receiver. Firmware 9.1.0 (ir).

Still crashing...

Can it be, that the Arduino Core has a problem with the 2 MCU cores?

@Jason2866 If you'll need hardware to reproduce, I would send this configuration to you. In Germany I could send a parcel by hermes. I won't charge costs.

Jason2866, 2020-11-05:
Mhh, can you try a version from my special builds? The builds are based on latest esp32 stage core. https://github.com/Jason2866/Merge/tree/firmware
I would suggest the full rules variant, it is the version with the smallest changes to the standard build.
Tbh, sending the device to me will probably not solve. The only one is @arendst who knows where to digg. Sending to NL does not cost a fortune.

GrendelWWU, 2020-11-05:
Thanks for your quick response. I flashed the full rules variant, but it crashes, too.
Sure I can send it to @arendst  it's not really expensive.

Theo, which ESP-Board version do you prefer? Onboard with antenna or the 32U varaint with UFL-connector, pigtail and SMA-Antenna? My contact mail address is th-dieckmann@web.de

arendst, 2020-11-06:
Pls let me know what onboard antenna hardware you use. I have differnet kinds lying around so it may already be onsite.

I doubt though that it's software related as both Jason and me cannot reproduce on a standard ESP32-Devkit.

GrendelWWU, 2020-11-06:
Hello Theo,

I could reproduce this behaviour with these three different boards from two manufacturers.
The board with the UFL (from China) connector was used with an external antenna, of course.
I've tested several locations in my appartment, too. So, signal strength won't be
a problem. I'm using AVM Fritz!Box 7580 and Fritz!Repeater 3000. (latest Firmware)

Best regards,
Thomas

[image]

[image]

[image]

arendst, 2020-11-06:
OK. I have none of them ;-)

Let's try something different. I see you use the IR binary. For NEC that's not needed so take the "normal" tasmota binary and re-test pls.

GrendelWWU, 2020-11-06:
Unfortunately the standard version is crashing, too...

I've created a serial dump again:
```
12:09:12 RSL: RESULT = {"Time":"2020-11-06T12:09:12","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFA25D"}}
12:09:12 RUL: IRRECEIVED#DATA=0X00FFA25D performs "backlog power1 toggle;var1 1;blinkcount 1"
12:09:12 RSL: RESULT = {"POWER1":"OFF"}
12:09:12 RSL: POWER1 = OFF
12:09:12 RSL: RESULT = {"Var1":"1"}
12:09:12 RUL: VAR1#STATE performs "backlog blinkcount %Var1;blinktime 3;power7 blink"
12:09:12 RSL: RESULT = {"BlinkCount":1}
12:09:12 RSL: RESULT = {"BlinkCount":1}
12:09:12 RSL: RESULT = {"BlinkTime":3}
12:09:12 RSL: RESULT = {"POWER7":"Blink ON"}
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x4017e7f4  PS      : 0x00060034  A0      : 0x800821e4  A1      : 0x3ffbe790  
A2      : 0x3ffc543c  A3      : 0xb0000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ff42000  A7      : 0x7000003b  A8      : 0x8008154e  A9      : 0x00000001  
A10     : 0x3ffbee70  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x0000001c  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40062250  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4017e7f4

Backtrace: 0x4017e7f4:0x3ffbe790 0x400821e1:0x3ffbe7b0 0x400865ed:0x3ffbe7d0 0x4006224d:0x3ffb1b60 0x40094cf3:0x3ffb1b80 0x40094f60:0x3ffb1bb0 0x40094f93:0x3ffb1be0 0x40095310:0x3ffb1c00 0x40088bb1:0x3ffb1c30 0x40088e28:0x3ffb1c50 0x4011fcc1:0x3ffb1ca0 0x4011e2cd:0x3ffb1cc0 0x4011e4ae:0x3ffb1ce0 0x4011d44f:0x3ffb1d60 0x4011d9f2:0x3ffb1de0 0x4011cdd5:0x3ffb1e50 0x400d6b21:0x3ffb1e90 0x400d6b71:0x3ffb1ec0 0x400e58a1:0x3ffb1ee0 0x400f3315:0x3ffb1f00 0x400fc257:0x3ffb1f90 0x4010e921:0x3ffb1fb0 0x4008aa19:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
⸮00:00:00 CFG: Loaded, Count 266
00:00:00 QPC: Count 1
00:00:00 Project tasmota ESP32-01 Version 9.1.0(tasmota)-1_0_4
00:00:01 WIF: Connecting to AP1 geleglu-point-01 Channel 1 BSSId E0:28:6D:6D:A5:9F in mode 11  as tasmota_971EE4-7908...
00:00:04 WIF: Connect failed
00:00:04 WIF: Connecting to AP1 geleglu-point-01 in mode 11  as tasmota_971EE4-7908...
00:00:04 QPC: Reset
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on tasmota_971EE4-7908 with IP address 10.11.12.14
00:00:07 RSL: INFO1 = {"Module":"ESP32-DevKit","Version":"9.1.0(tasmota)","FallbackTopic":"cmnd/DVES_971EE4_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:07 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_971EE4-7908","IPAddress":"10.11.12.14"}
00:00:07 RSL: INFO3 = {"RestartReason":"Software reset digital core"}
00:00:07 RSL: RESULT = {"POWER1":"ON"}
00:00:07 RSL: POWER1 = ON
00:00:07 RSL: RESULT = {"POWER2":"ON"}
00:00:07 RSL: POWER2 = ON
00:00:07 RSL: RESULT = {"POWER3":"ON"}
00:00:07 RSL: POWER3 = ON
00:00:07 RSL: RESULT = {"POWER4":"ON"}
00:00:07 RSL: POWER4 = ON
00:00:07 RSL: RESULT = {"POWER5":"ON"}
12:09:22 RSL: POWER5 = ON
12:09:22 RSL: RESULT = {"POWER6":"ON"}
12:09:22 RSL: POWER6 = ON
12:09:22 RSL: RESULT = {"POWER7":"OFF"}
12:09:22 RSL: POWER7 = OFF
12:09:22 RSL: RESULT = {"POWER8":"OFF"}
12:09:22 RSL: POWER8 = OFF
12:09:22 RSL: RESULT = {"POWER9":"ON"}
12:09:22 RSL: POWER9 = ON
12:09:22 RUL: SYSTEM#BOOT performs "backlog var1 1;blinkcount 1"
12:09:22 RSL: RESULT = {"Var1":"1"}
12:09:22 RUL: VAR1#STATE performs "backlog blinkcount %Var1;blinktime 3;power7 blink"
12:09:22 RSL: RESULT = {"BlinkCount":1}
12:09:22 RSL: RESULT = {"BlinkCount":1}
12:09:22 RSL: RESULT = {"BlinkTime":3}
12:09:22 RSL: RESULT = {"POWER7":"Blink ON"}
12:09:23 RSL: RESULT = {"POWER7":"Blink OFF"}
12:09:23 RSL: RESULT = {"POWER7":"OFF"}
12:09:23 RSL: POWER7 = OFF
12:09:26 RSL: STATE = {"Time":"2020-11-06T12:09:26","Uptime":"0T00:00:14","UptimeSec":14,"Heap":111,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":100,"MqttCount":0,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","POWER5":"ON","POWER6":"ON","POWER7":"OFF","POWER8":"OFF","POWER9":"ON","Dimmer":60,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"geleglu-point-01","BSSId":"E0:28:6D:6D:A5:9F","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:08"}}
12:09:26 RSL: SENSOR = {"Time":"2020-11-06T12:09:26","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","Switch5":"ON","Switch6":"ON"}

```
I've successfully toggled Power1 for more than 20 times. He's telling something about cache.

Best regards,
Thomas

arendst, 2020-11-06:
OK managed to reproduce.

I will see what goes wrong here.

Jason2866, 2020-11-06:
@arendst  maybe https://github.com/espressif/arduino-esp32/issues/3634 ?

arendst, 2020-11-06:
I also read something about IRAM interrupts AND access to NVS (https://github.com/espressif/arduino-esp32/issues/855#issuecomment-423540387) used by Tasmota to store settings (changes). As every power change triggers a flash update this will be an issue.

As a workaround try to delay flash writes using command ``datasave 100`` and see what happens. I

GrendelWWU, 2020-11-06:
Hello Theo, you mean SaveData? I set to 0, so see, what happens then.
I guess, it doesn't crash any more. But Dimmer + and DImmer - don't seem to work any more.
I'll check, if it's the standard version and if this will work in 9.1.0(ir) again.

See output:
```
18:40:39 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00FF10EF, Decode 3
18:40:39 RSL: RESULT = {"Time":"2020-11-06T18:40:39","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FF10EF"}}
18:40:39 RUL: IRRECEIVED#DATA=0X00FF10EF performs "Dimmer -"
18:40:39 SRC: Rule
18:40:39 CMD: Group 0, Index 1, Command "DIMMER", Data "-"
18:40:39 RSL: RESULT = {"POWER9":"ON","Dimmer":50}
18:40:42 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00FF10EF, Decode 3
18:40:42 RSL: RESULT = {"Time":"2020-11-06T18:40:42","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FF10EF"}}
18:40:42 RUL: IRRECEIVED#DATA=0X00FF10EF performs "Dimmer -"
18:40:42 SRC: Rule
18:40:42 CMD: Group 0, Index 1, Command "DIMMER", Data "-"
18:40:42 RSL: RESULT = {"POWER9":"ON","Dimmer":50}
18:40:42 IRR: Echo 0, RawLen 4, Overflow 0, Bits 0, Value 0xFFFFFFFFFFFFFFFF, Decode 3
18:40:44 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00FF10EF, Decode 3
18:40:44 RSL: RESULT = {"Time":"2020-11-06T18:40:44","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FF10EF"}}
18:40:44 RUL: IRRECEIVED#DATA=0X00FF10EF performs "Dimmer -"
18:40:44 SRC: Rule
18:40:44 CMD: Group 0, Index 1, Command "DIMMER", Data "-"
18:40:44 RSL: RESULT = {"POWER9":"ON","Dimmer":50}
18:40:44 IRR: Echo 0, RawLen 4, Overflow 0, Bits 0, Value 0xFFFFFFFFFFFFFFFF, Decode 3
18:40:45 WIF: Checking connection...
18:40:59 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00FF10EF, Decode 3
18:40:59 RSL: RESULT = {"Time":"2020-11-06T18:40:59","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FF10EF"}}
18:40:59 RUL: IRRECEIVED#DATA=0X00FF10EF performs "Dimmer -"
18:40:59 SRC: Rule
18:40:59 CMD: Group 0, Index 1, Command "DIMMER", Data "-"
18:40:59 RSL: RESULT = {"POWER9":"ON","Dimmer":50}
18:41:02 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00FF10EF, Decode 3
18:41:02 RSL: RESULT = {"Time":"2020-11-06T18:41:02","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FF10EF"}}
18:41:02 RUL: IRRECEIVED#DATA=0X00FF10EF performs "Dimmer -"
18:41:02 SRC: Rule
18:41:02 CMD: Group 0, Index 1, Command "DIMMER", Data "-"
18:41:02 RSL: RESULT = {"POWER9":"ON","Dimmer":50}
18:41:02 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0x00FF10EF, Decode 3
18:41:02 RSL: RESULT = {"Time":"2020-11-06T18:41:02","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FF10EF"}}
18:41:02 RUL: IRRECEIVED#DATA=0X00FF10EF performs "Dimmer -"
18:41:02 SRC: Rule
18:41:02 CMD: Group 0, Index 1, Command "DIMMER", Data "-"
18:41:02 RSL: RESULT = {"POWER9":"ON","Dimmer":50}
18:41:05 WIF: Checking connection...
```

GrendelWWU, 2020-11-06:
The 9.0.0.3(ir) can perform Dimmer + and Dimmer -
the 9.1.0-versions can't.... I thought so. But I tried DimmerStep 5, and it worked again, with 9.1.0.
I set DimemrStep 10, it was also okay. Somewhere in up- and downgrading, changing versions the config possibly forgot the DimmerStep value.

So, now I will configure a IR button to make a SaveData and I'm satisfied for the moment.
Guys, I really appreciate, what you're doing there.
Is there a way to give a donation for some cups of coffee or whatever?

Jason2866, 2020-11-06:
Great. So the reason seems to be found. Theo will find for sure a solution, not just a workaround.
You find a donate badge in the readme of this github to support Tasmota.

GrendelWWU, 2020-11-07:
> You find a donate badge in the readme of this github to support Tasmota.

After searching a while, I finally found it. Thx for the support and your excellent work.

arendst, 2020-11-07:
Thx for the donation. Much appreciated.

closed 2020-12-11 by github-actions[bot]