arendst/Tasmota #9423: SetOption20 not working with Zigbee2Tasmota

psycho160, 2020-09-28:
### PROBLEM DESCRIPTION
Adjusting the brightness over mqtt results always a power on of the blub. Even if SetOption20 is set to 1. Tasmota Webif also says "off" when the light is powerd on with brightness.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Nodemcu + Webee CC2530 
- [x
[image]
] Tasmota binary firmware version number used: 8.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Platformio
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

Program Version | 8.5.0(tasmota)
-- | --
2020-09-28T11:08:04
2_7_4_1/2.2.2-dev(38a443e)

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
17:01:54 MQT: tele/tasmota_10CD56/0B52/SENSOR = {"ZbReceived":{"Licht_AZ":{"Device":"0x0B52","Name":"Licht_AZ","Power":0,"Dimmer":81,"Endpoint":11,"LinkQuality":101}}}
17:01:55 MQT: stat/tasmota_10CD56/RESULT = {"ZbSend":"Done"}
17:01:55 MQT: tele/tasmota_10CD56/RESULT = {"ZbResponse":{"Device":"0x0B52","Name":"Licht_AZ","Command":"0008!04","Status":0,"StatusMessage":"SUCCESS","Endpoint":11,"LinkQuality":101}}
17:01:57 MQT: tele/tasmota_10CD56/0B52/SENSOR = {"ZbReceived":{"Licht_AZ":{"Device":"0x0B52","Name":"Licht_AZ","Dimmer":160,"Endpoint":11,"LinkQuality":101}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set Power 0 to a Zigbee2Tasmota Device (eg. blub)
Adjust brigthness
Blub will turn on, even if SetOption20 is set to 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Blub should stay off

Tasmota webIf shows blub as "off" event if light is "on"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-09-28:
Please, try latest development version. Thanks.

s-hadinger, 2020-09-28:
I will try to explain. `Power` is in cluster 6 and `Dimmer` is in cluster 8. If you change Dimmer in cluster 8, and the bulb is not configured to report Power attribute, it will not update Z2T.

You can solve this by declaring the device as a light with `ZbLight <device>,<channels>`. Channels is the number of channels of the light (1=dimmer, 2=whiteCT, 3=RGB, 4=RGBW, 5=RGBCW).

Alternatively you can set the light to report any change in Power.
```
ZbSend {"Device":"<device>","Config":{"Power":{"MinInterval":1,"MaxInterval":600}}}
```

That was for reporting and updating the UI.

In the Zigbee protocol, there are 2 dimmer commands, one that changes the On/Off (default in Z2T) and one that doesn't. I will add a check on SetOption20 to select the right one automatically.

psycho160, 2020-09-29:
@ascillato2 now on current dev version 

```
Program Version | 8.5.0.1(tasmota)
-- | --
2020-09-29T07:32:14
STAGE/2.2.2-dev(38a443e)

```
same issue...

@s-hadinger : I set ZbLight to 1 but if i change dimm level, webif does not turn to "on"

`06:43:49 CMD: ZbLight Licht_AZ,1
06:43:49 MQT: stat/tasmota_10CD56/RESULT = {"ZbLight":{"Licht_AZ":{"Device":"0x0B52","Name":"Licht_AZ","Light":1,"Reachable":true,"Power":0,"Dimmer":200}}}
06:43:49 MQT: stat/tasmota_10CD56/RESULT = {"ZbLight":"Done"}
06:43:59 MQT: stat/tasmota_10CD56/RESULT = {"ZbSend":"Done"}
06:43:59 MQT: tele/tasmota_10CD56/RESULT = {"ZbResponse":{"Device":"0x0B52","Name":"Licht_AZ","Command":"0008!04","Status":0,"StatusMessage":"SUCCESS","Endpoint":11,"LinkQuality":119}}
06:44:00 MQT: tele/tasmota_10CD56/0B52/SENSOR = {"ZbReceived":{"Licht_AZ":{"Device":"0x0B52","Name":"Licht_AZ","Dimmer":200,"Endpoint":11,"LinkQuality":122}}}`

Lamp is now on 78% shining but this is what my webif says:
[image]

s-hadinger, 2020-09-29:
Unfortunately changing the Dimmer while the light is OFF has no effect; the IKEA bulb just ignores it. I'm afraid you need to pass on this feature.

For the Web UI, can you please try again with `weblog 3`?

psycho160, 2020-09-30:
`Unfortunately changing the Dimmer while the light is OFF has no effect; the IKEA bulb just ignores it. I'm afraid you need to pass on this feature.`

No, if i change dimming level, my philips hue is working fine, only the webif says off even if the blub shines bright... Maybe the ikea blub has a different behaviour..

weblog 3 - turn lamp "off" and change of dimming level:
```
06:37:14 CMD: weblog 3
06:37:14 MQT: stat/tasmota_10CD56/RESULT = {"WebLog":3}
06:37:15 CFG: Saved to flash at F8, Count 33, Bytes 4096
06:37:23 SRC: MQTT
06:37:23 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"device":"0x0B52","send":{"Power":"Off"}}"
06:37:23 ZIG: guessing endpoint 11
06:37:23 SendZCLCommand_P: zcl_cmd = 
06:37:23 ZigbeeZCLSend device: 0x0B52, group: 0x0000, endpoint:11, cluster:0x0006, cmd:0x00, send:""
06:37:23 ZbSend: shortaddr 0x0B52, groupaddr 0x0000, cluster 0x0006, endpoint 0x0B, cmd 0x00, data 
06:37:23 ZIG: ZbZNPSent 240202520B0000000000000B000001060018301E0300011800
06:37:23 MQT: stat/tasmota_10CD56/RESULT = {"ZbSend":"Done"}
06:37:23 ZIG: {"ZbZNPReceived":"640200"}
06:37:23 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:23 ZIG: {"ZbZNPReceived":"4480000118"}
06:37:24 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:24 ZIG: {"ZbZNPReceived":"448100000600520B0B01007C0054A03D00000518180B0000520B1D"}
06:37:24 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0B52","srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":0,"fc":"0x18","manuf":"0x0000","transact":24,"cmdid":"0x0B","payload":"0000"}}
06:37:24 MQT: tele/tasmota_10CD56/RESULT = {"ZbResponse":{"Device":"0x0B52","Name":"Licht_AZ","Command":"0006!00","Status":0,"StatusMessage":"SUCCESS","Endpoint":11,"LinkQuality":124}}
06:37:24 ZIG: ZbZNPSent 240202520B0000000000000B000001060019301E05000019000000
06:37:24 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:24 ZIG: {"ZbZNPReceived":"640200"}
06:37:24 ZIG: {"ZbZNPReceived":"448100000600520B0B01007C0070A13D00000718010A00001000520B1D"}
06:37:24 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0B52","srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":0,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x0A","payload":"00001000"}}
06:37:24 ZIG: ZbZCLRawReceived: {"0x0B52":{"0006/0000":0,"Endpoint":11,"LinkQuality":124}}
06:37:24 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:24 ZIG: {"ZbZNPReceived":"4480000119"}
06:37:24 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:24 ZIG: {"ZbZNPReceived":"448100000600520B0B01007C00A3A33D0000081819010000001000520B1D"}
06:37:24 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0B52","srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":0,"fc":"0x18","manuf":"0x0000","transact":25,"cmdid":"0x01","payload":"0000001000"}}
06:37:24 ZIG: ZbZCLRawReceived: {"0x0B52":{"0006/0000":0,"Endpoint":11,"LinkQuality":124}}
06:37:24 MQT: tele/tasmota_10CD56/0B52/SENSOR = {"ZbReceived":{"Licht_AZ":{"Device":"0x0B52","Name":"Licht_AZ","Power":0,"Endpoint":11,"LinkQuality":124}}}
06:37:30 SRC: MQTT
06:37:30 CMD: Group 0, Index 1, Command "ZBSEND", Data "{ "device":"0x0B52", "send":{"Dimmer":130} }"
06:37:30 ZIG: guessing endpoint 11
06:37:30 SendZCLCommand_P: zcl_cmd = 820A00
06:37:30 ZigbeeZCLSend device: 0x0B52, group: 0x0000, endpoint:11, cluster:0x0008, cmd:0x04, send:"820A00"
06:37:30 ZbSend: shortaddr 0x0B52, groupaddr 0x0000, cluster 0x0008, endpoint 0x0B, cmd 0x04, data 
06:37:30 ZIG: ZbZNPSent 240202520B0000000000000B00000108001A301E0600011A04820A00
06:37:30 MQT: stat/tasmota_10CD56/RESULT = {"ZbSend":"Done"}
06:37:30 ZIG: {"ZbZNPReceived":"640200"}
06:37:30 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:31 ZIG: {"ZbZNPReceived":"448000011A"}
06:37:31 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:31 ZIG: {"ZbZNPReceived":"448100000800520B0B01007C000CF63D000005181A0B0400520B1D"}
06:37:31 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":8,"srcaddr":"0x0B52","srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":0,"fc":"0x18","manuf":"0x0000","transact":26,"cmdid":"0x0B","payload":"0400"}}
06:37:31 MQT: tele/tasmota_10CD56/RESULT = {"ZbResponse":{"Device":"0x0B52","Name":"Licht_AZ","Command":"0008!04","Status":0,"StatusMessage":"SUCCESS","Endpoint":11,"LinkQuality":124}}
06:37:31 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:31 ZIG: {"ZbZNPReceived":"448100000600520B0B01007C0027F73D00000718020A00001001520B1D"}
06:37:31 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0B52","srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":0,"fc":"0x18","manuf":"0x0000","transact":2,"cmdid":"0x0A","payload":"00001001"}}
06:37:31 ZIG: ZbZCLRawReceived: {"0x0B52":{"0006/0000":1,"Endpoint":11,"LinkQuality":124}}
06:37:31 MQT: tele/tasmota_10CD56/0B52/SENSOR = {"ZbReceived":{"Licht_AZ":{"Device":"0x0B52","Name":"Licht_AZ","Power":1,"Endpoint":11,"LinkQuality":124}}}
06:37:31 ZIG: ZbZNPSent 240202520B0000000000000B00000108001B301E0500001B000000
06:37:31 ZIG: {"ZbZNPReceived":"640200"}
06:37:32 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:32 ZIG: {"ZbZNPReceived":"448000011B"}
06:37:32 ZIG: {"ZbZNPReceived":"45C4520B00"}
06:37:32 ZIG: {"ZbZNPReceived":"448100000800520B0B01007C003A033E000008181B010000002082520B1D"}
06:37:32 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":8,"srcaddr":"0x0B52","srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":0,"fc":"0x18","manuf":"0x0000","transact":27,"cmdid":"0x01","payload":"0000002082"}}
06:37:32 ZIG: ZbZCLRawReceived: {"0x0B52":{"0008/0000":130,"Endpoint":11,"LinkQuality":124}}
06:37:32 MQT: tele/tasmota_10CD56/0B52/SENSOR = {"ZbReceived":{"Licht_AZ":{"Device":"0x0B52","Name":"Licht_AZ","Dimmer":130,"Endpoint":11,"LinkQuality":124}}}
```

s-hadinger, 2020-09-30:
I'm confident the ZbLight command did the trick :)

There is no solution for the setoption20 issue. The zigbee protocol is against us and there is nothing we can do.

When I turn the bulb off, change the dimmer and turn it on again, it still display the previous dimmer settings. Maybe it's only a limitation of the IKEA bulbs, I would like someone to try with another brand's bulb.

What is you bulb brand?

psycho160, 2020-09-30:
I have both, IKea Tradfri and Philips

s-hadinger, 2020-09-30:
You can try to manually send a Dimmer command that does not trigger OnOff.

Power on and set dimmer to 80%:
`ZbSend {"Device":"Licht_AZ","Send":{"Dimmer":203}}`

Power off:
`ZbSend {"Device":"Licht_AZ","Send":{"Power":0}}`

Set Dimmer to 10% without turning on:
`ZbSend {"Device":"Licht_AZ,"Send":"0008!00/190A00"}`

Power On and check brightness (10% or 80%):
`ZbSend {"Device":"Licht_AZ","Send":{"Power":1}}`

If it succeeds I can merge, otherwise we have to drop the feature.

psycho160, 2020-10-01:
Hi,
if i run `ZbSend {"Device":"Licht_AZ,"Send":"0008!00/190A00"}` i get `RESULT = {"ZbSend":"Invalid JSON"}`

s-hadinger, 2020-10-01:
Right. Missing quote but I'm sure you figured it out. 

`ZbSend {"Device":"Licht_AZ","Send":"0008!00/190A00"}`

psycho160, 2020-10-01:
Ups, omg you´re right. it was too early.
But unfortunately after `ZbSend {"Device":"Licht_AZ","Send":"0008!00/190A00"}` and `ZbSend {"Device":"Licht_AZ","Send":{"Power":1}}` the light is at 80%    :-(

So i think we can close this

s-hadinger, 2020-10-01:
Closing. For future reference, this feature is not possible due to the Zigbee protocol limitations (or limitations in the implementation).

closed 2020-10-01 by s-hadinger