arendst/Tasmota #9426: SetOption19 toggle does not republish

errolsancaktar, 2020-09-29:
### PROBLEM DESCRIPTION
It seems that if setoption19 has been enabled and disabled, there is no way to re-enable.  i'm not sure if this is because the retained message was removed without tasmota's knowledge or not.  

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): several with same issue: eva logik dimmer, mj switch, etc
- [X ] Tasmota binary firmware version number used: _8.5.0____
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used: firmware upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
05:36:05 RSL: home/sonoff/light_outdoor_front/RESULT = {"NAME":"WF30 Switch","GPIO":[0,0,0,0,18,0,0,0,21,17,0,0,0],"FLAG":0,"BASE":18}
05:36:05 RSL: home/sonoff/light_outdoor_front/RESULT = {"Module":{"0":"WF30 Switch"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
05:37:02 CMD: status 0
05:37:02 MQT: home/sonoff/light_outdoor_front/STATUS = {"Status":{"Module":0,"DeviceName":"Front Outdoor Lights","FriendlyName":["Front Outdoor Lights"],"Topic":"light_outdoor_front","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:37:02 MQT: home/sonoff/light_outdoor_front/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"<redacted>":"Software/System restart","Uptime":"0T00:13:06","StartupUTC":"2020-09-29T04:23:56","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-09-28T22:47:58","SaveCount":102,"SaveAddress":"F5000"}}
05:37:02 MQT: home/sonoff/light_outdoor_front/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-09T11:41:02","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"452/699"}}
05:37:02 MQT: home/sonoff/light_outdoor_front/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<REDACTED>",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
05:37:02 MQT: home/sonoff/light_outdoor_front/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":0,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
05:37:02 RSL: home/sonoff/light_outdoor_front/STATUS5 = {"StatusNET":{"Hostname":"light_outdoor_front-7100","IPAddress":"172.16.10.174","Gateway":"172.16.10.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"<REDACTED>","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
05:37:02 RSL: home/sonoff/light_outdoor_front/STATUS6 = {"StatusMQT":{"MqttHost":"<REDACTED>","MqttPort":1883,"MqttClientMask":"%topic%","MqttClient":"%topic%","MqttUser":"<REDACTED>","MqttCount":781,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
05:37:02 RSL: home/sonoff/light_outdoor_front/STATUS7 = {"StatusTIM":{"UTC":"2020-09-29T04:37:02","Local":"2020-09-29T05:37:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:47","Sunset":"18:32"}}
05:37:02 RSL: home/sonoff/light_outdoor_front/STATUS10 = {"StatusSNS":{"Time":"2020-09-29T05:37:02"}}
05:37:02 RSL: home/sonoff/light_outdoor_front/STATUS11 = {"StatusSTS":{"Time":"2020-09-29T05:37:02","Uptime":"0T00:13:06","UptimeSec":786,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":781,"POWER":"ON","Wifi":{"AP":1,"SSId":"<REDACTED>","BSSId":"04:18:D6:C2:29:C5","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":2,"Downtime":"0T00:00:04"}}}
05:37:03 MQT: Attempting connection...

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
05:26:34 RSL: home/sonoff/light_outdoor_front/SO = {"SetOption19":"ON"}
05:26:35 MQT: Attempting connection...
05:26:35 MQT: Connected
05:26:35 MQT: home/sonoff/light_outdoor_front/LWT = Online (retained)
05:26:35 MQT: home/sonoff/light_outdoor_front/cmnd/POWER = 
05:26:35 RSL: homeassistant/switch/A37BBC_RL_1/config = {"name":"Front Outdoor Lights","stat_t":"home/sonoff/light_outdoor_front/STATE","avty_t":"home/sonoff/light_outdoor_front/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"home/sonoff/light_outdoor_front/cmnd/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"A37BBC_RL_1","dev":{"ids":["A37BBC"]}}
05:26:35 RSL: homeassistant/sensor/A37BBC_status/config = {"name":"Front Outdoor Lights status","stat_t":"home/sonoff/light_outdoor_front/HASS_STATE","avty_t":"home/sonoff/light_outdoor_front/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"home/sonoff/light_outdoor_front/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"A37BBC_status","dev":{"ids":["A37BBC"],"name":"Front Outdoor Lights","mdl":"WF30 Switch","sw":"8.5.0(tasmota)","mf":"Tasmota"}}
05:26:36 MQT: Attempting connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

- enable setoption19
- remove retained message from mqtt server
- disable and reenable setoption19, 59

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

my thought is that it should republish the homeassistant discovery config

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
not sure if this is as designed or not but only one of the messages shows up with topics:
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-09-29:
Please post the output of the console when doing setoption19 0
And then 
Setoption19 1

The zero is to disable the autodiscovery and the 1 is enabling it. To rediscover, you need to do both, first the 0 and then the 1.

effelle, 2020-09-29:
I'm failing to understand your issue. 
Using your config:
```json
12:24:43 CMD: backlog module; template; so19 1
12:24:44 MQT: stat/tasmota_49A3BC/RESULT = {"Module":{"0":"WF30 Switch"}}
12:24:44 MQT: stat/tasmota_49A3BC/RESULT = {"NAME":"WF30 Switch","GPIO":[0,0,0,0,18,0,0,0,21,17,0,0,0],"FLAG":0,"BASE":18}
12:24:44 MQT: stat/tasmota_49A3BC/RESULT = {"SetOption19":"ON"}
12:24:44 MQT: homeassistant/switch/49A3BC_RL_1/config = {"name":"Tasmota","stat_t":"tele/tasmota_49A3BC/STATE","avty_t":"tele/tasmota_49A3BC/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/tasmota_49A3BC/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"49A3BC_RL_1","dev":{"ids":["49A3BC"]}} (retained)
12:24:44 MQT: homeassistant/sensor/49A3BC_status/config = {"name":"Tasmota status","stat_t":"tele/tasmota_49A3BC/HASS_STATE","avty_t":"tele/tasmota_49A3BC/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/tasmota_49A3BC/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"49A3BC_status","dev":{"ids":["49A3BC"],"name":"Tasmota","mdl":"WF30 Switch","sw":"8.5.0(tasmota)","mf":"Tasmota"}} (retained)
```
Even removing manually the entities from mqtt (why?) and then doing  again so19 0 / so19 1 all is fine here.
> not sure if this is as designed or not but only one of the messages shows up with topics:

Is why we ask to send logs configuring the console with `weblog 4`. 
You will not see the empty topics because they are hidden.

ascillato, 2020-09-29:
@errolsancaktar 

From your log, it can be seen MQT and then RSL as type of logs, so that means that you are having a network issue or MQTT broker issue. Your broker is disconnecting and reconnecting. Please, check your network and broker.

ascillato2, 2020-09-29:
If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)