arendst/Tasmota #9431: Support multiple SR04 ultrasonic distance sensors

dka23, 2020-09-29:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
No - just need to use 2 SR04 sensors on a single ESP32

**Describe the solution you'd like**  
Support for multiple SR04 sensors like it already is present for buttons, e.g.

**Describe alternatives you've considered**  
Using a separate ESP per sensor would be an alternative, but more expensive and with additional cabling.

**Additional context**  
None

**(Please, remember to close the issue when the problem has been addressed)**

dka23, 2020-09-29:
I will finalize the pull request soon and submit accordingly. 

Little background on the project: I would like to use two SR04 sensors to monitor the occupancy of two parking spaces adjacent to each other inside a garage.

dka23, 2020-09-29:
PR looks good from my point of view. Happy to receive any feedback and improve/change the PR accordingly.

stefanbode, 2020-10-04:
My experience is that the sr04 are neither accurate nor stay healthy for more than a year. I have completely changed to the laser based vl53L0x. These support also more than one and work awesome.

dka23, 2020-11-09:
Anything I can do to support this further? Is there any problem with the PR that I can help to resolve?

ascillato2, 2020-11-09:
Closing this issue as the PR #9432 has been submitted for review. Thanks for your work!!!

It is missing just a test for the ESP8266.

closed 2020-11-09 by ascillato2

jgesser, 2023-08-08:
> Closing this issue as the PR #9432 has been submitted for review. Thanks for your work!!!
> 
> It is missing just a test for the ESP8266.

Please reopen this request, PR was closed and this is still a important feature for some users.

jgesser, 2024-03-05:
I have made the changes again to support multiple SR04 sensors on the latest development branch:

Please take a look on below comments on the old PR:
https://github.com/arendst/Tasmota/pull/9432#issuecomment-1978749484
https://github.com/arendst/Tasmota/pull/9432#issuecomment-1978755428

I have only one sensor at this moment and can't test it before raising a new PR. 
It will take a month at least to receive my new sensor here in Brazil, so if anyone is able to test please do it and I can raise the PR sooner.

PR #20975 (2024-03-31): Added support to multiple SR04
## Description:

Added support to multiple SR04 sensors.
Thanks to @belgotux for helping with the tests (https://github.com/arendst/Tasmota/pull/9432#issuecomment-2002438725)

**Related issue (if applicable):** fixes #9431

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.6
  - [x] The code change is tested and works with  [... cut]