arendst/Tasmota #9433: Compilation issue with NRF24 support

mrekin, 2020-09-30:
### PROBLEM DESCRIPTION
Get errors on compiling tasmota-sensors with _USE_SPI_ , _USE_NRF24_ and _USE_MIBLE_
Can be reproduced on release 8.5.0 and latest sources 8.5.0.1.

Errors:
`/workspace/Tasmota/tasmota/xsns_62_MI_HM10.ino:117:8: error: redefinition of 'struct mi_beacon_t'
 struct mi_beacon_t{
        ^
/workspace/Tasmota/tasmota/xsns_61_MI_NRF24.ino:166:8: error: previous definition of 'struct mi_beacon_t'
 struct mi_beacon_t{
        ^
/workspace/Tasmota/tasmota/xsns_62_MI_HM10.ino:146:8: error: redefinition of 'struct ATCPacket_t'
 struct ATCPacket_t{`
- and the same/ Looks like there was some types definition conflict.

Tested in Gitpod and local Platformio. (Latest gitpod image don't auto install platformio for some reason, need to install it manually..)

Compilation tasmota binary ('-e tasmota') works as expected with _USE_SPI_ , _USE_NRF24_ and _USE_MIBLE_ support   
Compilation tasmota-sensors binary ('-e tasmota-sensors') works as expected with _USE_SPI_ , _USE_NRF24_ and **without** _USE_MIBLE_ support   

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Define USE_SPI and USE_NRF24
2. Try to compile tasmota-sersors
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-30:
For a individual build you have to use always the variant Tasmota as base. All other variants (partially) resets your settings done in `my_user_config.h` or `my_user_config_override.h` So the behaviour you see is as expected.

With PR #9434 the Gitpod regression is fixed

mrekin, 2020-09-30:
@Jason2866  thanks. If it true way - compile 'tasmota' - this is ok. Older versions (8.2.x) had no compilation errors with  tasmota-sensors: this is a reason this issue opened.

closed 2020-09-30 by mrekin