arendst/Tasmota #9436: Invalid JSON for BLE(NRF2401l) sensors (8.5.0.1)

mrekin, 2020-09-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Tasmota 8.5.0.1, self compiled with USE_SPI and USE_NRF24.
Set of Xiaomi sensors, like LYWSD02 and others.

Some sensors too far from NRF2401L and some values not encoded from advrs packets - they has **null** values
[image]

With SetOption19 1   HAAS autodiscovery doesn't work, because tasmota generates json with empty entry (in "LYWSD02-8250f4" object):  
`14:23:08 MQT: tele/tasmota_4C80D5/SENSOR = {"Time":"2020-09-30T14:23:08","LYWSD02-8250f4":{,"Battery":55},"CGD1-51cf95":{"Temperature":24.4,"Humidity":34.3,"DewPoint":7.6},"TempUnit":"C"}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): __Wemos D1 mini +NRF2401l+___
- [x] Tasmota binary firmware version number used: _8.5.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mrekin, 2020-10-01:
Looks like this code may be redesigned
\tasmota\xsns_61_MI_NRF24.ino
`          if(!isnan(MIBLEsensors[i].temp) && !isnan(MIBLEsensors[i].hum)){
              ResponseAppendTHD(MIBLEsensors[i].temp,MIBLEsensors[i].hum);
          }
          if(MIBLEsensors[i].bat!=0x00){ // this is the error code -> no battery
            ResponseAppend_P(PSTR(",\"Battery\":%u"), MIBLEsensors[i].bat);
          }`
add additional 'if' to check are comma needed 
or do not send Battery stat at all
or append temp\hum\dewpoint separately.

Staars, 2020-10-01:
Yes, that is not intended.
I already changed this part of the code in the other BLE-drivers and will port it to the NRF24-version soon.

Staars, 2020-10-03:
I made a GIST for testing:
https://gist.github.com/Staars/6ec50d01c082c75186f47fef163f884a

Just replace the current driver code.

This version would also change the way of publishing the data - away from the teleperiod to a direct bridge mode. With the current options, you will not get the full data at teleperiod, but every time the NRF receives a packet. This will start after 3 successful receive operations to rule out fake sensors due to data corruption.

But you can change this in the code:
 MINRF.option.noSummary = 1; (-> full output at teleperiod)

The NRF24 needs more exemptions than the other BLE-drivers and this makes driver maintenance harder too.

mrekin, 2020-10-04:
Just installed (with 8.5.0 base).
Works,but..  
I did clear install (remove old topics from mqtt server), set autodiscovery to true (SetOption19 1)
Tasmota set  only _Battery topics in homeassistant/ 
[image]

and published every packet (as you described)
So as result I get only "battery" objects in HAAS with 'blinking' values :) 
[image]

Staars, 2020-10-05:
When I do a `SO19 1` after enough time to read all sensors for at least 3 times I get this in the console (loglevel 4):

```
16:46:52 MQT: homeassistant/sensor/11DA0C_ATC-a9fcf0_Temperature/config = {"name":"Tasmota ATC-a9fcf0 Temperature","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_ATC-a9fcf0_Temperature","dev":{"ids":["11DA0C"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['ATC-a9fcf0']['Temperature']}}"} (retained)
16:46:52 MQT: homeassistant/sensor/11DA0C_ATC-a9fcf0_Humidity/config = {"name":"Tasmota ATC-a9fcf0 Humidity","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_ATC-a9fcf0_Humidity","dev":{"ids":["11DA0C"]},"unit_of_meas":"%","dev_cla":"humidity","frc_upd":true,"val_tpl":"{{value_json['ATC-a9fcf0']['Humidity']}}"} (retained)
16:46:52 MQT: homeassistant/sensor/11DA0C_ATC-a9fcf0_DewPoint/config = {"name":"Tasmota ATC-a9fcf0 DewPoint","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_ATC-a9fcf0_DewPoint","dev":{"ids":["11DA0C"]},"unit_of_meas":"°C","ic":"mdi:weather-rainy","frc_upd":true,"val_tpl":"{{value_json['ATC-a9fcf0']['DewPoint']}}"} (retained)
16:46:52 MQT: homeassistant/sensor/11DA0C_ATC-a9fcf0_Battery/config = {"name":"Tasmota ATC-a9fcf0 Battery","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_ATC-a9fcf0_Battery","dev":{"ids":["11DA0C"]},"unit_of_meas":"%","dev_cla":"battery","frc_upd":true,"val_tpl":"{{value_json['ATC-a9fcf0']['Battery']}}"} (retained)
16:46:52 MQT: homeassistant/sensor/11DA0C_MHOC303-421c8f_Tempera/config = {"name":"Tasmota MHOC303-421c8f Temperature","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_MHOC303-421c8f_Tempera","dev":{"ids":["11DA0C"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['MHOC303-421c8f']['Temperature']}}"} (retained)
16:46:52 MQT: homeassistant/sensor/11DA0C_MHOC303-421c8f_Humidit/config = {"name":"Tasmota MHOC303-421c8f Humidity","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_MHOC303-421c8f_Humidit","dev":{"ids":["11DA0C"]},"unit_of_meas":"%","dev_cla":"humidity","frc_upd":true,"val_tpl":"{{value_json['MHOC303-421c8f']['Humidity']}}"} (retained)
16:46:53 MQT: homeassistant/sensor/11DA0C_MHOC303-421c8f_DewPoin/config = {"name":"Tasmota MHOC303-421c8f DewPoint","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_MHOC303-421c8f_DewPoin","dev":{"ids":["11DA0C"]},"unit_of_meas":"°C","ic":"mdi:weather-rainy","frc_upd":true,"val_tpl":"{{value_json['MHOC303-421c8f']['DewPoint']}}"} (retained)
16:46:53 MQT: homeassistant/sensor/11DA0C_MHOC303-421c8f_Battery/config = {"name":"Tasmota MHOC303-421c8f Battery","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_MHOC303-421c8f_Battery","dev":{"ids":["11DA0C"]},"unit_of_meas":"%","dev_cla":"battery","frc_upd":true,"val_tpl":"{{value_json['MHOC303-421c8f']['Battery']}}"} (retained)
16:46:53 MQT: homeassistant/sensor/11DA0C_Flora-6ab577_Temperatu/config = {"name":"Tasmota Flora-6ab577 Temperature","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_Flora-6ab577_Temperatu","dev":{"ids":["11DA0C"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['Flora-6ab577']['Temperature']}}"} (retained)
16:46:53 MQT: homeassistant/sensor/11DA0C_Flora-6ab577_Illuminan/config = {"name":"Tasmota Flora-6ab577 Illuminance","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_Flora-6ab577_Illuminan","dev":{"ids":["11DA0C"]},"unit_of_meas":"LX","dev_cla":"illuminance","frc_upd":true,"val_tpl":"{{value_json['Flora-6ab577']['Illuminance']}}"} (retained)
16:46:53 MQT: homeassistant/sensor/11DA0C_Flora-6ab577_Moisture/config = {"name":"Tasmota Flora-6ab577 Moisture","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_Flora-6ab577_Moisture","dev":{"ids":["11DA0C"]},"unit_of_meas":"%","ic":"mdi:cup-water","frc_upd":true,"val_tpl":"{{value_json['Flora-6ab577']['Moisture']}}"} (retained)
16:46:53 MQT: homeassistant/sensor/11DA0C_Flora-6ab577_Fertility/config = {"name":"Tasmota Flora-6ab577 Fertility","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_Flora-6ab577_Fertility","dev":{"ids":["11DA0C"]},"unit_of_meas":" ","ic":"mdi:eye","frc_upd":true,"val_tpl":"{{value_json['Flora-6ab577']['Fertility']}}"} (retained)
16:46:53 MQT: homeassistant/sensor/11DA0C_Flora-6ab577_Battery/config = {"name":"Tasmota Flora-6ab577 Battery","stat_t":"tele/tasmota_11DA0C/SENSOR","avty_t":"tele/tasmota_11DA0C/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"11DA0C_Flora-6ab577_Battery","dev":{"ids":["11DA0C"]},"unit_of_me [... cut ...] pl":"{{value_json['CGD1-508af6']['Battery']}}"} (retained)
```

The automatic first autodiscovery should be too early for this driver.
Do you get a comparable output for your sensors?

The other issue with the shortened JSON at teleperiod is related to my suggestion to switch to a pure bridge-mode, which means that at the time of receiving a BLE data packet, you immediately translate this to a mqtt message. Thus the problem of resending old data (at teleperiod) is easily solved, at least for me.

mrekin, 2020-10-05:
Tamota was online about 5 hours.
In UI I got 6 devices (5 real and one fake with very similar mac for real one)
In console
`19:22:00 CMD: SO19 1
19:22:00 SRC: WebConsole from 192.168.1.11
19:22:00 CMD: Group 0, Index 19, Command "SO", Data "1"
19:22:00 MQT: stat/tasmota_4C80D5/RESULT = {"SetOption19":"ON"}
19:22:00 MQT: homeassistant/sensor/4C80D5_LYWSD02-81cd1f_Battery/config = {"name":"TasmotaBLEGW LYWSD02-81cd1f Battery","stat_t":"tele/tasmota_4C80D5/SENSOR","avty_t":"tele/tasmota_4C80D5/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"4C80D5_LYWSD02-81cd1f_Battery","dev":{"ids":["4C80D5"]},"unit_of_meas":"%","dev_cla":"battery","frc_upd":true,"val_tpl":"{{value_json['LYWSD02-81cd1f']['Battery']}}"} (retained)
19:22:00 MQT: homeassistant/sensor/4C80D5_ATC-e4cf93_Battery/config = {"name":"TasmotaBLEGW ATC-e4cf93 Battery","stat_t":"tele/tasmota_4C80D5/SENSOR","avty_t":"tele/tasmota_4C80D5/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"4C80D5_ATC-e4cf93_Battery","dev":{"ids":["4C80D5"]},"unit_of_meas":"%","dev_cla":"battery","frc_upd":true,"val_tpl":"{{value_json['ATC-e4cf93']['Battery']}}"} (retained)
19:22:00 MQT: homeassistant/sensor/4C80D5_ATC-2bcf99_Battery/config = {"name":"TasmotaBLEGW ATC-2bcf99 Battery","stat_t":"tele/tasmota_4C80D5/SENSOR","avty_t":"tele/tasmota_4C80D5/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"4C80D5_ATC-2bcf99_Battery","dev":{"ids":["4C80D5"]},"unit_of_meas":"%","dev_cla":"battery","frc_upd":true,"val_tpl":"{{value_json['ATC-2bcf99']['Battery']}}"} (retained)
19:22:00 MQT: homeassistant/cover/4C80D5_SHT_1/config =  (retained)
19:22:00 MQT: homeassistant/cover/4C80D5_SHT_2/config =  (retained)
19:22:00 MQT: homeassistant/cover/4C80D5_SHT_3/config =  (retained)
19:22:00 MQT: homeassistant/cover/4C80D5_SHT_4/config =  (retained)
19:22:00 MQT: homeassistant/light/4C80D5_LI_1/config =  (retained)
19:22:00 MQT: homeassistant/switch/4C80D5_RL_1/config =  (retained)
19:22:00 MQT: homeassistant/light/4C80D5_LI_2/config =  (retained)
19:22:00 MQT: homeassistant/switch/4C80D5_RL_2/config =  (retained)
19:22:00 MQT: homeassistant/light/4C80D5_LI_3/config =  (retained)
19:22:00 MQT: homeassistant/switch/4C80D5_RL_3/config =  (retained)
19:22:00 MQT: homeassistant/light/4C80D5_LI_4/config =  (retained)
19:22:00 MQT: homeassistant/switch/4C80D5_RL_4/config =  (retained)
19:22:00 MQT: homeassistant/light/4C80D5_LI_5/config =  (retained)
19:22:00 MQT: homeassistant/switch/4C80D5_RL_5/config =  (retained)
19:22:00 MQT: homeassistant/light/4C80D5_LI_6/config =  (retained)
19:22:00 MQT: homeassistant/switch/4C80D5_RL_6/config =  (retained)
19:22:00 MQT: homeassistant/light/4C80D5_LI_7/config =  (retained)
19:22:00 MQT: homeassistant/switch/4C80D5_RL_7/config =  (retained)
19:22:00 MQT: homeassistant/light/4C80D5_LI_8/config =  (retained)
19:22:00 MQT: homeassistant/switch/4C80D5_RL_8/config =  (retained)
19:22:00 MQT: homeassistant/sensor/4C80D5_status/config = {"name":"TasmotaBLEGW status","stat_t":"tele/tasmota_4C80D5/HASS_STATE","avty_t":"tele/tasmota_4C80D5/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/tasmota_4C80D5/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"4C80D5_status","dev":{"ids":["4C80D5"],"name":"TasmotaBLEGW","mdl":"Generic","sw":"8.5.0(tasmota)","mf":"Tasmota"}} (retained)
19:22:00 CFG: Saved to flash at F5, Count 17, Bytes 4096`

HAAS found only "_battery" objects  without any value (empty).
When tasmota send "immediately" message like
`19:27:49 MQT: tele/tasmota_4C80D5/SENSOR = {"Time":"2020-10-05T19:27:49","ATC-2bcf99":{"Temperature":25.8,"Humidity":32.0,"DewPoint":7.8,"Battery":92},"TempUnit":"C"}
`
haas set object to value from message for a second and switch back to 'unavaliable' again 
[image]

Telemessage contains only battery values
`19:30:49 MQT: tele/tasmota_4C80D5/SENSOR = {"Time":"2020-10-05T19:30:49","LYWSD02-8250f4":{"Battery":50},"LYWSD02-81cd1f":{"Battery":27},"ATC-e4cf93":{"Battery":87},"CGD1-51cf95":{},"ATC-2bcf99":{"Battery":92},"LYWSD02-81cd1f":{"Battery":27}}
`

I'll try to build 8.5.1 version with new driver..

mrekin, 2020-10-05:
Build with Tasmota 9.0.0.1 and seems it works!
Will check a little time

Staars, 2020-10-05:
Yes, please play with the options in init function.

The LYWSD02 „twitters“ very frequently and what you see is the real situation of incoming messages. The „old“ method (all sensor values only at teleperiod) hides the real world and that’s why I think it is better to do a real-time translation of the messages to mqtt.

mrekin, 2020-10-05:
The second LYWSD02 transmit not such frequently, but it has only 27% battery. 
Does HM10 have a better sensitivity rather than nrf24? 
I have one and will try, but have no a lot of time to hobby..)
Can hm10 work with ble and beacon devices in parallel?

Staars, 2020-10-05:
HM-10 may perform a little bit better, but this is not guaranteed.

To my best knowledge the HM-10 could only do Xiaomi-sensor readings and beacon functions in parallel, if the existence of the beacon is parsed from the continuously scanned BLE advertisements.

What do expect from a „beacon function“ in Tasmota? Only showing, that a BLE-device with certain MAC-address is visible?

mrekin, 2020-10-05:
> What do expect from a „beacon function“ in Tasmota? Only showing, that a BLE-device with certain MAC-address is visible?

Yep, do some automation in haas if beacon visible or not.

Staars, 2020-10-05:
Okay, I will take a look (simple MAC-based „beacon“ and standardized iBeacon should be doable).

But before we get too much off-topic here, please take your time for some checks of this new potential driver version.
I would like to get this issue closed and then we can move on.

mrekin, 2020-10-06:
As summary:
1) Teleperiod messages contains only battery values
`08:44:36 MQT: tele/tasmota_4C80D5/SENSOR = {"Time":"2020-10-06T08:44:36","LYWSD02-81cd1f":{"Battery":27},"LYWSD02-8250f4":{"Battery":50},"ATC-e4cf93":{"Battery":84},"CGD1-51cf95":{},"ATC-2bcf99":{"Battery":93}}
`
- this message drops other sensor values and causes values "blinking" in hass.
Why it contains only battery values ?  Looks like 
` if((!MINRF.mode.triggeredTele && !MINRF.option.minimalSummary)||MINRF.mode.triggeredTele){`
not working, because all internal checks have (!MINRF.mode.triggeredTele) conditions, but Battery values appending have other 'true' conditions ?
Do you plan to send telemessage ? If no  - need to remove battery state for teleperiod, if yes -need to fill it with all values
2) Is it hard to implement MINRF.option.noSummary switch by cmd , not by code changing? I don't know if this really needed, just asking.
3) My HASS instance starts freezing\lagging with intermediate messaging when I open integrations->mqtt-><my ble device> (here is all BLE objects with values).  Not a bit problem (other pages\scenes works normally)

Staars, 2020-10-06:
Hmm, I will try gather more information about HA and how it expects the data.
At least I see no unintended behaviour of the driver in your posts. It just does not play nice with HA at the moment.

Staars, 2020-10-08:
After some lengthy talks with other developers there seems to be chance to improve interaction with HASS. Due to lack of time I can not not promise a time frame, but it is not forgotten.

mrekin, 2020-10-08:
Thanks a lot ! No new issues found with new driver at this time, all works as expected.

closed 2020-10-09 by ascillato2

Staars, 2020-10-15:
@mrekin The new version is online and I went back to presenting sensor data only at TELEPERIOD by default. Not all ideas are realized yet.
But I already updated the GIST from above to a new version, which (hopefully) publishes 'null' for every missing sensor value.
So please test both versions (from official development and the GIST) and tell me, which one ist better for HASS, if you find the time. I can only make educated guesses as a happy and problem-free non-HASS-user 😀 .

mrekin, 2020-10-15:
@Staars Hi:) I'm on vacations now and will back 20.10) So I will test new version on the next week. 
Thanks!