arendst/Tasmota #9438: SetOption1 2 to allow only single and hold events to remove lag?

mzealey, 2020-09-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes, doesn't appear to be supported

**Is your feature request related to a problem? Please describe.**  
I'm building some switches that want to support press and hold but with immediate triggering. Due to multi-press functionality a single press doesn't respond immediately which is a bit off-putting.

**Describe the solution you'd like**  
`SetOption1 2` be similar to `SetOption1 1` but further restricting to only press and hold, and hence can action as soon as the button has been released

**Describe alternatives you've considered**  
`SetOption13 1` provides instant feedback, but as it's leading edge triggered it cannot detect holds

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-30:
Sorry, will be most probably not done. The button and switch stuff has tons of variants and options. There is already a lot of very complex code for it. There will be always wishes for "another" variant.

digiblur, 2020-10-02:
Set your Sonoff Basic to use the button as switch1, then set switchmode1 5 and you'll have instant pushes and still have holds.  I use this one a lot and even my last Sonoff basic in production has this.

closed 2020-10-09 by ascillato2