arendst/Tasmota #9448: Diymore 4 csatornás relétábla

ZoleeBB, 2020-10-02:
### PROBLEM DESCRIPTION
Helló!
Ezt a terméket vásároltam: https://www.aliexpress.com/item/32950217718.html
Mivel ESP8266 vezérlővel rendelkezik, gondolkodás nélkül telepítettem rá a legújabb Tasmota firmware-t!
A probléma az, hogy bármit is választok az eszköz beállításaiban, egyik relé sem működik. Kipróbáltam az "Általános" beállítást, és megpróbáltam beállítani az összes GPIO relét. Sajnos egyik relé sem kapcsol be soha! :( Tudna segíteni? Hogyan működhet ez?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): diymore DC 12V ESP8266 4 Channel Relay Board ESP-01 WIFI Module
- [x] Tasmota binary firmware version number used: 8.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
None of the relays turn on

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-02:
Only English.

Jason2866, 2020-10-02:
Proabably this one will work. Please in English next time. Thx!
https://templates.blakadder.com/LC-ESP01-4R-12V.html

ZoleeBB, 2020-10-04:
> Proabably this one will work. Please in English next time. Thx!
> https://templates.blakadder.com/LC-ESP01-4R-12V.html

Thanks, this is how it works!

closed 2020-10-04 by ZoleeBB