arendst/Tasmota #9449: MQTT SENSOR messages stops working when "thermostat" is enabled

cwesystems, 2020-10-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have flashed my Shelly 1PM with DS18B20-hat with my own compiled Tasmota with thermostat feature.

When I activate the thermostat function with command "THERMOSTATMODESET 1" the MQTT SENSOR message only reports "{"Time"

When thermostat is disabled the MQTT SENSOR message is correct, "{"Time":"2020-10-02T10:57:43","Switch1":"OFF","ANALOG":{"Temperature":27.5},"ENERGY":{"TotalStartTime":"2020-09-28T10:13:30","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0},"DS18B20":{"Id":"01193838E3D1","Temperature":24.2},"TempUnit":"C"}"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [*] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [*] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [*] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [*] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [*] Device used (e.g., Sonoff Basic): Shelly 1PM with official DS18B20-hat
- [*] Tasmota binary firmware version number used: 8.5.0.1
  - [ ] Pre-compiled
  - [*] Self-compiled
    - [*] IDE / Compiler used: Atom/PlatformIO
- [*] Flashing tools used: esptool.py
- [*] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:09:56 CMD: Backlog Template; Module; GPIO 255
11:09:57 MQT: stat/termostat_19C654/RESULT = {"NAME":"Shelly 1PM U2","GPIO":[255,255,0,255,82,134,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}
11:09:57 MQT: stat/termostat_19C654/RESULT = {"Module":{"0":"Shelly 1PM U2"}}
11:09:57 MQT: stat/termostat_19C654/RESULT = {"GPIO0":{"192":"DS18x20o"},"GPIO1":{"0":"Ingen"},"GPIO2":{"0":"Ingen"},"GPIO3":{"4":"DS18x20"},"GPIO4":{"82":"Omkopplare1n"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"Ingen"},"GPIO10":{"0":"Ingen"},"GPIO12":{"0":"Ingen"},"GPIO13":{"0":"Ingen"},"GPIO14":{"0":"Ingen"},"GPIO15":{"21":"Relä1"},"GPIO16":{"0":"Ingen"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:11:27 CMD: STATUS 0
11:11:27 MQT: stat/termostat_19C654/STATUS = {"Status":{"Module":0,"DeviceName":"element","FriendlyName":["element"],"Topic":"termostat_19C654","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:11:27 MQT: stat/termostat_19C654/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:14:46","StartupUTC":"2020-10-02T09:56:41","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2020-09-28T10:13:30","SaveCount":82,"SaveAddress":"FA000"}}
11:11:27 MQT: stat/termostat_19C654/STATUS2 = {"StatusFWR":{"Version":"8.5.0.1(tasmota)","BuildDateTime":"2020-09-24T19:35:36","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"406/699"}}
11:11:27 MQT: stat/termostat_19C654/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":"Automation",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
11:11:27 MQT: stat/termostat_19C654/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":404,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["0000041D","8FCAE797","04368001","000000CD","010013C0","C000F981","00004404","00000000"],"Drivers":"1,2,3,4,5,6,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,39","Sensors":"1,2,3,4,5,6"}}
11:11:27 MQT: stat/termostat_19C654/STATUS5 = {"StatusNET":{"Hostname":"termostat_19C654-1620","IPAddress":"192.168.0.151","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"D8:BF:C0:19:C7:34","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:11:27 MQT: stat/termostat_19C654/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_19C654","MqttUser":"user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:11:27 MQT: stat/termostat_19C654/STATUS7 = {"StatusTIM":{"UTC":"2020-10-02T10:11:27","Local":"2020-10-02T11:11:27","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:52","Sunset":"18:26"}}
11:11:27 MQT: stat/termostat_19C654/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:11:27 MQT: stat/termostat_19C654/STATUS10 = {"StatusSNS":{"Time":"2020-10-02T11:11:27","Switch1":"OFF","ANALOG":{"Temperature":36.9},"ENERGY":{"TotalStartTime":"2020-09-28T10:13:30","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0},"DS18B20":{"Id":"01193838E3D1","Temperature":24.6},"TempUnit":"C"}}
11:11:27 MQT: stat/termostat_19C654/STATUS11 = {"StatusSTS":{"Time":"2020-10-02T11:11:27","Uptime":"0T00:14:46","UptimeSec":886,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Automation","BSSId":"BSSId":"AA:2F:3E:92:B2:AA","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
10:57:23 MQT: tele/termostat_19C654/STATE = {"Time":"2020-10-02T10:57:23","Uptime":"0T00:00:42","UptimeSec":42,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Automation","BSSId":"AA:2F:3E:92:B2:AA","Channel":6,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:06"}}
10:57:23 MQT: tele/termostat_19C654/SENSOR = {"Time":"2020-10-02T10:57:23","Switch1":"OFF","ANALOG":{"Temperature":26.8},"ENERGY":{"TotalStartTime":"2020-09-28T10:13:30","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0},"DS18B20":{"Id":"01193838E3D1","Temperature":24.2},"TempUnit":"C"}
10:57:23 MQT: tele/termostat_19C654/HASS_STATE = {"Version":"8.5.0.1(tasmota)","BuildDateTime":"2020-09-24T19:35:36","Module or Template":"Shelly 1PM U2","RestartReason":"Power On","Uptime":"0T00:00:42","Hostname":"termostat_19C654-1620","IPAddress":"192.168.0.151","RSSI":"54","Signal (dBm)":"-73","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"LoadAvg":20}
10:57:33 MQT: tele/termostat_19C654/STATE = {"Time":"2020-10-02T10:57:33","Uptime":"0T00:00:52","UptimeSec":52,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Automation","BSSId":"AA:2F:3E:92:B2:AA","Channel":6,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:06"}}
10:57:33 MQT: tele/termostat_19C654/SENSOR = {"Time":"2020-10-02T10:57:33","Switch1":"OFF","ANALOG":{"Temperature":27.2},"ENERGY":{"TotalStartTime":"2020-09-28T10:13:30","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0},"DS18B20":{"Id":"01193838E3D1","Temperature":24.2},"TempUnit":"C"}
10:57:33 MQT: tele/termostat_19C654/HASS_STATE = {"Version":"8.5.0.1(tasmota)","BuildDateTime":"2020-09-24T19:35:36","Module or Template":"Shelly 1PM U2","RestartReason":"Power On","Uptime":"0T00:00:52","Hostname":"termostat_19C654-1620","IPAddress":"192.168.0.151","RSSI":"54","Signal (dBm)":"-73","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"LoadAvg":19}
10:57:43 MQT: tele/termostat_19C654/STATE = {"Time":"2020-10-02T10:57:43","Uptime":"0T00:01:02","UptimeSec":62,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Automation","BSSId":"AA:2F:3E:92:B2:AA","Channel":6,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:06"}}
[... cut ...]
5. Disable thermostat with "THERMOSTATMODESET 0"
6. MQTT SENSOR message is correct.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
MQTT SENSOR message should be the same even if thermostat is enabled. I need to monitor temperature.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-02:
You are using a not supported flash size. 
`"ProgramSize":600,"Free":404,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"`
Please compile with the standard (provided) platformIO settings. This will generate a firmware with the standard 1024 ProgramFlashSize. If you build different the compiled binary is not working correctly.

cwesystems, 2020-10-02:
OK, I have now flashed it with -fs 1MB, last I flashed it with auto flash-size.
Build-size is the same, I have not changed anything from standard Tasmota.

Now I have, **"ProgramSize":600,"Free":404,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"**

Still the same problem:
"SENSOR = {"Time"

ascillato2, 2020-10-09:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

arendst, 2020-10-10:
Can replicate on a wemos. Will investigate.

Jason2866, 2020-10-11:
@cwesystems Have you tried the fix? Please close when the issue is solved for you.

cwesystems, 2020-10-11:
I can confirm it works with latest dev bransch!
Good work! Thank you!

closed 2020-10-11 by cwesystems