arendst/Tasmota #9451: Attempted OTA upgrade from 8.3.1 fails with 2 units left 'stuck' at 8.5.0-minimal

georgep56, 2020-10-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Today I have tried to OTA upgrade 2 of my test units (Sonoff S25 switches) from version 8.3.1 .
Both units ostensibly went through the normal upgrade sequence but both are now 'stuck' at version 8.5.0-minimal and I can't get them to upgrade further. Neither an automatic upgrade (from GUI or console) nor a manual firmware upload will work 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

16:45:05 CMD: Backlog Template; Module; GPIO 255
16:45:06 MQT: stat/zztasmota1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:45:06 MQT: stat/zztasmota1/RESULT = {"Module":{"8":"Sonoff S2X"}}
16:45:06 MQT: stat/zztasmota1/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
16:48:32 CMD: status 0
16:48:32 MQT: stat/zztasmota1/STATUS = {"Status":{"Module":8,"DeviceName":"zzTasmota1","FriendlyName":["zzTasmota1"],"Topic":"zztasmota1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:48:32 MQT: stat/zztasmota1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Blocked Loop","Uptime":"0T00:08:51","StartupUTC":"2020-10-02T15:39:41","Sleep":50,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2020-06-25T15:05:49","SaveCount":112,"SaveAddress":"FB000"}}
16:48:32 MQT: stat/zztasmota1/STATUS2 = {"StatusFWR":{"Version":"8.5.0(minimal)","BuildDateTime":"2020-09-09T11:41:33","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"400/699"}}
16:48:32 MQT: stat/zztasmota1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VodafoneConnect52496345",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600001E5A00000000000000","00000000","00006000","00000000"]}}
16:48:32 MQT: stat/zztasmota1/STATUS4 = {"StatusMEM":{"ProgramSize":378,"Free":624,"Heap":30,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","00000580","04000003","00000002","00000000","00000000","00000000","00000000"],"Drivers":"1,2","Sensors":""}}
16:48:32 MQT: stat/zztasmota1/STATUS5 = {"StatusNET":{"Hostname":"zztasmota1-7301","IPAddress":"192.168.0.219","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"CC:50:E3:1C:5C:85","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:48:32 MQT: stat/zztasmota1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1C5C85","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:48:32 MQT: stat/zztasmota1/STATUS7 = {"StatusTIM":{"UTC":"2020-10-02T15:48:32","Local":"2020-10-02T16:48:32","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99}}
16:48:32 MQT: stat/zztasmota1/STATUS10 = {"StatusSNS":{"Time":"2020-10-02T16:48:32"}}
16:48:32 MQT: stat/zztasmota1/STATUS11 = {"StatusSTS":{"Time":"2020-10-02T16:48:32","Uptime":"0T00:08:51","UptimeSec":531,"Vcc":3.484,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VodafoneConnect52496345","BSSId":"E4:95:6E:45:15:9C","Channel":7,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}
16:48:32 MQT: stat/zztasmota1/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Blocked Loop","EPC":["4020173e","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4020173c","402548fc","40223c11","4021dd36","402426a0","4021e53e","402426ad","402426f2","4021b934","40000f49","40000f49","40000f49","40000e19","40104d3d","40104d43","4010000d","4023b124","4023b0d5","40104b85","401001f4","40101db5","40101984","4010476a","401040b9","4010426b","4010466d","40104945","40102c0e","401001f4","40101984","40104c37"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

<<not applicable>>

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Attempt to upgrade fully working Sonoff S26 switches from v8.3.1 to the latest release 8.5.1 using 'upgrade 1' from the console.
OTAUrl is set to default "http://thehackbox.org/tasmota/release/tasmota.bin"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Units should automatically load a minimal software, reboot, then load the latest version, upgrade and reboot.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

The key to the problem seems to be 'blocked loop' message as show in Status12; googling this it seems to be some kind of memory corruption and only resolvable by manual re-flashing (i.e. dismantle the switch, add wires to serial ports.......)

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-10-02:
Wrong OtaUrl.

Pls read releasenotes for correct OtaUrl.

georgep56, 2020-10-02:
Thank you Theo.

I had seen the different OtaUrl but (maybe naively) thought that fetching the binary from a different place would not make any difference?
I have now changed the OtaUrl value but the 'damaged' units now won't recover - I guess that they will need to be wiped and re-flashed using esptool?
I'd be interested to know just what has happened here - the release notes suggest that the new url holds the same binaries as the original?
-- 
George

Jason2866, 2020-10-02:
It has to be a other issue. There files hosted at thehackbox.org are the version before this release but still valid and okay. I just OTA updated a device from thehackbox.org and after that to the newest release from ota.tasmota.com

georgep56, 2020-10-03:
I have tried to recover both of the units which are now running 'minimal' firmware by manually choosing and uploading the firmware but they are both 'stuck' at 8.5.0-minimal. My understanding is that it should be valid to do this yet they are both failing in the same way. I have also tried a 'reset 6' yet the units still behave the same and won't upgrade.

Obviously I can open them up and wipe/flash but having upgraded a lt of units over time I am concerned that there could be an issue here that will affect others.

I have obtained this latest output which should help to at least determine what state these units have become 'stuck' in...

[11:58:32] stat/zztasmota1/STATUS12 {"StatusSTK":{"Exception":28,"Reason":"Blocked Loop","EPC":["4020173e","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4020173c","4010539d","4021dd36","402426a0","4021e53e","402426ad","402426f2","4021b934","40000f49","40000f49","40000f49","40000e19","40104d3d","40104d43","4010000d","4023b124","4023b0d5","401001f4","40101db5","40101984","40104c1f","40103dfa","4010426b","401001f4","40101db5","40101984","40103d33","40103dfa","40102bf8","4000050c","401028e0"]}}

schrej-zz, 2020-10-05:
Hi,
did you try a local update with the compressed versions?
I also failed with uncompressed version (tasmota-IR in that case)

ascillato2, 2020-10-09:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

hcsaustrup, 2020-10-26:
I'm having the same issues upgrading from 8.3.1 to 8.5.1. I have about 20 units around the house, and currently about 5 are stuck with this issue. OTA URL seems fine. Have also tried uploading both the standard and minimal release to no avail - both compressed and uncompressed.

```
house/tasmota/cellar-ventilator/stat/RESULT {"OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-minimal.bin.gz"}
house/tasmota/cellar-ventilator/stat/RESULT {"Upgrade":"Version 8.3.1 from http://thehackbox.org/tasmota/release/tasmota-minimal.bin.gz"}
...
house/tasmota/cellar-ventilator/stat/UPGRADE {"Upgrade":"Failed Update error: ERROR[6]: Stream Read Timeout"}
```

or..

```
house/tasmota/livingroom-fishtank/tele/INFO3 {"RestartReason":{"Exception":28,"Reason":"Blocked Loop","EPC":["402022ca","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["402022c8","4028476c","4024e6ed","402479a2","4026d274","402489f2","4026d281","4026d2c6","4024570c","40000f49","40000f49","40000f49","40000e19","40105a41","40105a47","4010000d","40265d14","40265cc5","4010593b","4010321f","40103400","40102688","401038c2","40103400","4000050c","4024b039","4024b6a2","4024b5d8","4024bfd9","4024b6a2","4024b5d8"]}}
```

georgep56, 2020-10-26:
I wasn't happy that the issue was closed without any more discussion. It's kind of good to see someone else with this problem.

ascillato2, 2020-10-26:
> I wasn't happy that the issue was closed without any more discussion.

Reopening issue as requested.

ascillato2, 2020-10-26:
Please try using latest version from development branch (v 9.0.0.2)

Jason2866, 2020-10-26:
The ota url is still the outdated one.

hcsaustrup, 2020-10-26:
I tried with the latest development binary (9.0.0.0 I think), available at http://ota.tasmota.com/tasmota/tasmota.bin.gz - while other units successfully upgraded to the latest version, one running 8.3.1 appears to upgrade, but it stuck at 8.3.1:

```
house/tasmota/cellar-ventilator/stat/RESULT {"OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz"}
house/tasmota/cellar-ventilator/stat/RESULT {"Upgrade":"Version 8.3.1 from http://ota.tasmota.com/tasmota/tasmota.bin.gz"}
house/tasmota/cellar-ventilator/tele/LWT Offline
house/tasmota/cellar-ventilator/tele/LWT Online
house/tasmota/cellar-ventilator/cmnd/POWER (null)
house/tasmota/cellar-ventilator/tele/LWT Offline
...
house/tasmota/cellar-ventilator/tele/LWT Online
house/tasmota/cellar-ventilator/cmnd/POWER (null)
house/tasmota/cellar-ventilator/stat/UPGRADE {"Upgrade":"Successful. Restarting"}
...
house/tasmota/cellar-ventilator/tele/LWT Offline
house/tasmota/cellar-ventilator/tele/LWT Online
house/tasmota/cellar-ventilator/cmnd/POWER (null)
house/tasmota/cellar-ventilator/tele/INFO1 {"Module":"Sonoff Basic","Version":"8.3.1(tasmota)","FallbackTopic":"cmnd/DVES_1C7A50_fb/","GroupTopic":"house/tasmota/tasmotas/cmnd/"}
house/tasmota/cellar-ventilator/tele/INFO2 {"WebServerMode":"Admin","Hostname":"tasmota-cellar-ventilator","IPAddress":"10.69.0.147"}
house/tasmota/cellar-ventilator/tele/INFO3 {"RestartReason":"Software/System restart"}
```

Jason2866, 2020-10-26:
Try for the problem device:
download tasmota.bin.gz and do a local firmware file upload
If wifi connection is not good (enough) OTA from server does not always work

hcsaustrup, 2020-10-26:
Same result with image upload. I've succesfully upgraded 6 units now, with 3 still stuck at 8.3.1. There doesn't seem to be a pattern in the unit-to-access-point range. RSSIs on the three failing units:

- 62%, -69 dBm
- 66%, -67 dBm
- 80%, -60 dBm

One of the successfully upgraded units is in my garden shed, way across the lawn - and it's reporting 42%, -79 dBm, which seems to indicate connection is not a problem. Not sure if this is valuable at all, but the access points are Unifi AC PROs.

georgep56, 2020-10-27:
Very similar to my experience. I have 2 units 'stuck' on minimal firmware and unable to be upgraded. I have tried different versions and manual uploads but nothing works.
Obviously I can disassemble and re-flash but there is obviously a 'real' problem here.

sfromis, 2020-10-27:
I'd suggest trying http://ota.tasmota.com/tasmota/tasmota.bin instead of the .gz version. I think I've had a case of upgrading from 8.3.1 fail until I did that.

georgep56, 2020-10-27:
Good suggestion but I've tried that already :-)

gufonero, 2020-11-20:
I experienced the same problem described here and after a lot of trial-and-error I came to the conclusion that it is connection-related and not a Tasmota bug. 

My situation was similar to the one described above, with three devices being upgraded. One was successful, the other two got stuck after upgrading to a custom tasmota-minimal v.8.5.1. Any attempt to upgrade firmware from here resulted in exception:28, reason:"Blocked Loop". After trying several times I succeded in upgrading another device, but the third one wouldn't upgrade or downgrade, neither by OTA nor by file upload.

Finally I switched off the first of the two access points configured as default. This AP is not very far away from the device, but is separated by a couple of walls. Switching it off forced the Tasmota device to connect to the second AP, which is closer and has no obstacles in between. After that the upgrade process worked like a charm.

The first AP has been working flawlessly for months with all three said devices, exchanging MQTT messages and even occasional firmware updates, but I guess the connection is not fully reliable and sending big files may cause lots of transmission errors and retries that put too much stress on the upgrade process' resources; this would be consistent with being able to occasionally upgrade to the minimal firmware, but not to the (bigger) standard build. Maybe the latest Tasmota 8 releases are more sensitive to this problem for some reason and end up crashing, but there seems not to be a fundamental problem in Tasmota.

So my suggestion if you experience this problem is: ensure that you have a good WiFi connection and try again.

closed 2020-11-27 by ascillato2

Nilmoty, 2023-01-05:
Hi, I have the same, have you find solution ? 
Regards