arendst/Tasmota #9457: GPIO03 / Switch1(9) triggers Power toggle in NodeMcu with GND, on Sonoff basic V2 only with TX

yossibeck, 2020-10-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
With Tasmota 6, connecting a physical switch had to be done with Rx connected to Gnd, now (8.5.1) it works with Rx to Tx
however - the same binary flashed into a NodeMcu works Rx to Gnd.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
 {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
 {"Module":{"1":"Sonoff Basic"}}
 {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"9":"Switch1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
  STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_CA94F1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
  STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:26:29","StartupUTC":"2020-10-03T21:07:06","Sleep":50,"CfgHolder":4617,"BootCount":48,"BCResetTime":"2020-10-03T13:48:12","SaveCount":74,"SaveAddress":"FA000"}}
  STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"373/699"}}
  STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ob-wan-k","TP-LINK_AP_60D4"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
  STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
  STATUS5 = {"StatusNET":{"Hostname":"sonoff-ParentsLightSW","IPAddress":"10.100.102.190","Gateway":"10.100.102.1","Subnetmask":"255.255.255.0","DNSServer":"10.100.102.1","Mac":"60:01:94:CA:94:F1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
  STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CA94F1","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
  STATUS7 = {"StatusTIM":{"UTC":"2020-10-03T21:33:35","Local":"2020-10-03T22:33:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:53","Sunset":"18:24"}}
  STATUS10 = {"StatusSNS":{"Time":"2020-10-03T22:33:35","Switch1":"OFF"}}
  STATUS11 = {"StatusSTS":{"Time":"2020-10-03T22:33:35","Uptime":"0T00:26:29","UptimeSec":1589,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"ob-wan-k","BSSId":"20:B0:01:32:9B:0D","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}  

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
flash tasmota.bin to a sonoff basic RF R2
select sonoff basic template
set gpio03 to switch9

connect Rx to Gnd - nothing happens
connect Rx to Tx - power toggles

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
connect Rx to Gnd - power should toggle

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-04:
Yes, since the circuit design from NodeMCU and Sonoff Basic is different.
Not a issue. You have to adapt the settings for the use case/device.

yossibeck, 2020-10-04:
@Jason2866  - thanks, but this WAS the old setting - (Rx to Gnd) - I wonder - if I upgrade a switch that's already installed (which has Rx to Gnd - and working currently) from 6 to 8 - the physical switch will stop working ?

Jason2866, 2020-10-04:
A lot of stuff changed in Tasmota since v.6.x. Gpio pin handling too. So it is possible due to different circuit designs in different hardware that behaviour can change. 
Connecting extern switches without a pull up resistor is not a good design choice.
This is a usual design in electronics. Connecting RX to TX is only a (bad) workaround.
If you add a resistor it will work.

yossibeck, 2020-10-04:
thanks a lot - I checked and indeed it works with a pull up resistor, closing as both solutions are workable
either Tx/Rx or a pull up resistor

closed 2020-10-04 by yossibeck