arendst/Tasmota #9458: roller shutters timing problem

pawol, 2020-10-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pawol, 2020-10-03:
On the latest 8.5.1 there is roller shutters timing problem.
For instance full real shutter cycle takes 30 s. 
On 8.5.0 it was set by commands shutteropenduration and shuttercloseduration
and ir worked well. On 8.5.1 it is necessary to add more or less additional 5s
to open and close shutters fully.

pawol, 2020-10-03:
when restored 8.5.0 then shutters works well again

kugelkopf123, 2020-10-03:
Sound like my problem:
https://github.com/tasmota/docs/pull/440#issuecomment-699653496

pawol, 2020-10-04:
I thing that new feature of 8.5.1 is the root cause:

Add new shutter modes #9244

stefanbode, 2020-10-04:
I will take a look into it. I haven't seen this behavior. 5 seconds are quite much. Can you add a log file. Just to get a better idea.

kugelkopf123, 2020-10-04:
> Sounds like my problem:
> [tasmota/docs#440 (comment)](https://github.com/tasmota/docs/pull/440#issuecomment-699653496)

@stefanbode 
I think that the problem of TO is the same problem as mine. 
I can open another issue with my problem if that is desired, otherwise I would join this issue.
The complete behaviour and how it can be reproduced I already described in the linked thread. 
Unfortunately nobody has answered within 1 week. BTW Not even in the support chat.
Now the weekend is over and I can't reach the Shelly at the moment. Therefore I cannot send you any logs.
Thank you very much for taking care of it.

pawol, 2020-10-04:
I think you will not see anything in logs.
My shutter rollers are normal type, AC powered pipe motor.
Like I said time of full closing and opening is 40 s. I worked perfectly on 8.5.0 and earlier. After update onto 8.5.1 simply set time looks too short and I had to add a few second. Of course set time is longer than real time.

stefanbode, 2020-10-05:
I flushed one with 30sec up and down. as I can see in the logs it is:
08:57:19 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"ON"}
...
08:57:50 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"OFF"}

This is exactly 30 seconds maybe 31sec max. I also used a stopwatch and came to the same value. Please share a log so we can see if the logs report the correct time at least

```
08:57:19 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER2":"ON"}
08:57:19 MQT: hm/status/tasmota_A9D47F/POWER2 = ON (retained)
08:57:19 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"ON"}
08:57:19 MQT: hm/status/tasmota_A9D47F/POWER1 = ON (retained)
08:57:19 MQT: hm/status/tasmota_A9D47F/RESULT = {"ShutterClose1":0}
08:57:19 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
08:57:20 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":98,"Direction":-1,"Target":0}}
08:57:21 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":94,"Direction":-1,"Target":0}}
08:57:22 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":91,"Direction":-1,"Target":0}}
08:57:23 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":88,"Direction":-1,"Target":0}}
08:57:24 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":85,"Direction":-1,"Target":0}}
08:57:25 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":81,"Direction":-1,"Target":0}}
08:57:26 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":78,"Direction":-1,"Target":0}}
08:57:27 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":74,"Direction":-1,"Target":0}}
08:57:28 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":71,"Direction":-1,"Target":0}}
08:57:29 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":68,"Direction":-1,"Target":0}}
08:57:30 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":64,"Direction":-1,"Target":0}}
08:57:31 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":61,"Direction":-1,"Target":0}}
08:57:32 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":58,"Direction":-1,"Target":0}}
08:57:33 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":54,"Direction":-1,"Target":0}}
08:57:34 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":51,"Direction":-1,"Target":0}}
08:57:35 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":48,"Direction":-1,"Target":0}}
08:57:36 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":45,"Direction":-1,"Target":0}}
08:57:37 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":41,"Direction":-1,"Target":0}}
08:57:38 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":38,"Direction":-1,"Target":0}}
08:57:39 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":34,"Direction":-1,"Target":0}}
08:57:40 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":31,"Direction":-1,"Target":0}}
08:57:41 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":28,"Direction":-1,"Target":0}}
08:57:42 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":24,"Direction":-1,"Target":0}}
08:57:43 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":21,"Direction":-1,"Target":0}}
08:57:44 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":18,"Direction":-1,"Target":0}}
08:57:45 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":15,"Direction":-1,"Target":0}}
08:57:46 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":11,"Direction":-1,"Target":0}}
08:57:47 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":8,"Direction":-1,"Target":0}}
08:57:48 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":5,"Direction":-1,"Target":0}}
08:57:49 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":1,"Direction":-1,"Target":0}}
08:57:50 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"OFF"}
08:57:50 MQT: hm/status/tasmota_A9D47F/POWER1 = OFF (retained)
08:57:50 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER2":"OFF"}
08:57:50 MQT: hm/status/tasmota_A9D47F/POWER2 = OFF (retained)
08:57:50 MQT: hm/status/tasmota_A9D47F/SHUTTER1 = 0 (retained)
08:57:50 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}

```

stefanbode, 2020-10-05:
I found indeed an error in the calibration calculation. As long as the calibration matrix is 0 everything is fine. Any calibration causes a mess. Update come soon.

stefanbode, 2020-10-05:
Correct gents. With the 8.5.1 change, I had to increase internal resolution from 100 to 1000 to support accelerating and decelerating of specific shutters. The new high numbers in the calibration matrix created an overflow of the int16 values. +-32.000 max. All fixed with the latest submit. Thanks for reporting and bringing it up. Also made an enhancement because the old algorithm was with one calibration value (20.000) with 30 seconds quite near the overflow.

kugelkopf123, 2020-10-05:
Great! I’m just able to test it on the next weekend. Will report.
Thanks!

pawol, 2020-10-05:
when the fix will be available in a release? I can test it at once.

kugelkopf123, 2020-10-05:
@pawol just download the latest dev Version. The fix is merged.

pawol, 2020-10-05:
uuuu... I rather avoid to use master versions. that is my silent rule :)

kugelkopf123, 2020-10-05:
@pawol its just for the test =) to check if its okay now.

pawol, 2020-10-05:
ok. I shall try tommorow

pawol, 2020-10-06:
Done.
Verified and works well on master 9.0.01. 
Update onto 9.0.01 is correct, roller shutter timing is correct.
At the end downgrade onto 8.5.0 is correct

kugelkopf123, 2020-10-06:
Great! Thx @stefanbode @pawol 
Will try it, when Im home on Saturday.

stefanbode, 2020-10-07:
I found an additional bug after the change. sorry. submitted and waiting for merge. Line 194:
New:
`		return (Settings.shuttercoeff[2][index] * 5 > realpos) ? SHT_DIV_ROUND(realpos/10, Settings.shuttercoeff[2][index]) : SHT_DIV_ROUND(realpos/10-Settings.shuttercoeff[0][index]*10, Settings.shuttercoeff[1][index]);`

old:
`		return (Settings.shuttercoeff[2][index] * 5 > realpos/10) ? SHT_DIV_ROUND(realpos/10, Settings.shuttercoeff[2][index]) : SHT_DIV_ROUND(realpos/10-Settings.shuttercoeff[0][index]*10, Settings.shuttercoeff[1][index]);`

stefanbode, 2020-10-09:
@pawol : Can you close if everything is fine?

pawol, 2020-10-09:
works ok

closed 2020-10-09 by pawol

kugelkopf123, 2020-10-09:
Hm. Yes its working better now. 
Latest Dev Version on it, Done a reset 1.
Set Interlock 1
Set Shutterrelay1 1
Set Shuttermode 1
Set shutteropenduration to 12.4
set shuttercloseduration to 12.2

But there is still something strange with the "ShutterSetHalfway" Stuff. After setting the Value everything was okay. From close to 50 works. Back to close and then to Shutteropen. And the Shutter stops around 10cm befor Totally Open Position. 
Set SO80 0 and Power1 1 to open it totally. 
Set SO80 back on and call Shutterposition 50. Now its 10 cm too high in the Mid Position. Ok. Further to the "shutterclose" position. Thats okay. Now back to shutterposition 50. Still 10cm to high. 

Back to close. ShutterSethalfway 50 to reset the Mid position. (as the docs says)
via Console: Shutterposition 73. Thats normaly my Half open Position. The Shutter runs 5 Secs open. and is still nearly closed.  Okay... Tried shutterposition 90. Runs 4 more seconds and now its still 80% closed. 
Call "shutteropen" -> Runs one more second. Now i would say its stll 75% closed
Damn irritating. Its not possible to open it anymore. 
So back to shutterclose.
Shutterclose fits. 
Tried a restart and than shutteropen...
Doesnt help.
The only thing that helps, is to set: Shuttercalibration back to 30 50 70 90 100. 
I have not done any fine calibration before. So this was the first time I run the command Shuttercalibration. 
So i would suggest to reset the shuttercalibration with the command "sethalfway",too.

stefanbode, 2020-10-29:
The sethalfway uses the callibration matrix to store its values. I made changes to avoid overflow there. I have to check if sethalfway 50 resets the matrix in the correct way. This might be an issue. anyhow anything with sethalfway 60,70,90 whatever should work without a problem. I will check this today. Using our settings:
Set Interlock 1
Set Shutterrelay1 1
Set Shuttermode 1
Set shutteropenduration to 12.4
set shuttercloseduration to 12.2

cu

stefanbode, 2020-10-29:
I have pushed a small change, Indeed the calibration mask was not reset if you go back to 50 and this causes the system to do strange things. Anyhow everything else than 50 as shuttersethalfway should also work before.

kugelkopf123, 2020-10-30:
Great! Thanks that you have had a look at it. 
I can not check if it’s working the next 2 weeks. I’m not at home. But as soon as I checked it, I will update you.