arendst/Tasmota #9462: Tasmota STM32

fabiofusco, 2020-10-04:
Hi,
I have a ethernet relay board.
This board have a n stm32, can i fash it with Tasmota?
Does Tasmota support wired ethermet connection?
This is the board link:

https://www.amazon.it/internet-Ethernet-controller-Remote-switch/dp/B07C2VRZ4D/ref=sr_1_19_sspa?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=4OQYCAVSKF2S&dchild=1&keywords=ethernet+relay+i%2Fo&qid=1601804851&sprefix=eternet+%2Caps%2C172&sr=8-19-spons&psc=1&smid=A2F7JN9V2AWTSH&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExVUJSV0FJUURZUlUmZW5jcnlwdGVkSWQ9QTAyMTA1MTYxUk45T0hYR0pNMEdQJmVuY3J5cHRlZEFkSWQ9QTA1MDkyMTBCMk1TR1JKM1MwS1Umd2lkZ2V0TmFtZT1zcF9tdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

Thanks 
Fabio

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-04:
No, since Tasmota is for ESP8266 and ESP32 only.

sfromis, 2020-10-04:
Tasmota can use the Ethernet on some ESP32 PoE boards.

closed 2020-10-06 by arendst