arendst/Tasmota #9464: No connection to Zigbee devices after upgrade (via Router)

schrej-zz, 2020-10-04:
### PROBLEM DESCRIPTION
I upgraded from Tasmota 8.2.03. (Zigbee enabled) to Tasmota 8.5.1 (Zigbee enabled)
After restart, I do not get any connection to the Zigbee router and all devices.
Devices are visible via zbstatus, but no MQTT messages in log.
At least router should transmit regular messages.
I haven't found any documentation, if CC253x coordinator and/or router need a dedicated update after tasmota update. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic):  WMOS D1 Mini with CC2531 router
- [x] Tasmota binary firmware version number used: 8.5.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x ] IDE / Compiler used: Tasmocompiler
- [x] Flashing tools used: Via tasmota device from local files
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
18:28:20 MQT: Haus/OT/WL_WD1M_03/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
18:28:21 MQT: Haus/OT/WL_WD1M_03/stat/RESULT = {"Module":{"18":"Generic"}}
18:28:21 MQT: Haus/OT/WL_WD1M_03/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"166":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"165":"Zigbee Tx"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS = {"Status":{"Module":18,"DeviceName":"WL_WD1M_03","FriendlyName":["WL_WD1M_03"],"Topic":"Haus/OT/WL_WD1M_03","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:34","StartupUTC":"2020-10-04T16:27:40","Sleep":50,"CfgHolder":4617,"BootCount":83,"BCResetTime":"2020-04-05T11:51:23","SaveCount":1629,"SaveAddress":"FA000"}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-04T15:33:21","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"433/699"}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Schreiner",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["04008009","2805C8000100060000005AFF000000000000","000000C8","00006000","00000000"]}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS4 = {"StatusMEM":{"ProgramSize":558,"Free":444,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F804582","04660001","000000C1","00000000","C000F981","00000004","00000000"],"Drivers":"1,2,4,6,8,9,10,18,19,22,23,24,26,27,29,30,35,37","Sensors":"1,2,4,5,6"}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS5 = {"StatusNET":{"Hostname":"WL-WMOS-D1-Mini-03","IPAddress":"192.168.2.213","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"BC:DD:C2:24:82:0B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_24820B","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-10-04T16:29:14","Local":"2020-10-04T18:29:14","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:55","Sunset":"19:22"}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS10 = {"StatusSNS":{"Time":"2020-10-04T18:29:14"}}
18:29:14 MQT: Haus/OT/WL_WD1M_03/stat/STATUS11 = {"StatusSTS":{"Time":"2020-10-04T18:29:14","Uptime":"0T00:01:34","UptimeSec":94,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"98:9B:CB:BE:A6:BF","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
N/A
```

### TO REPRODUCE
Upgrade device to 8.5.1 (Zigbee) and restart

### EXPECTED BEHAVIOUR
Connection to Zigbee router should appear in console as MQTT message

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-04:
Can you please paste logs of startup?

schrej-zz, 2020-10-04:
Sure
` 00:00:00 CFG: Loaded from flash at F8, Count 1631
00:00:00 QPC: Count 1
00:00:00 SNS: Hardware Serial
00:00:00 Project tasmota WL_WD1M_03 Version 8.5.1(tasmota)-2_7_4_1
00:00:00 ZbZNPReceived: received frame of wrong size FE, len 1, expected 5
00:00:04 WIF: Connecting to AP1 Schreiner Channel 1 BSSId 98:9B:CB:BE:A6:BF in mode 11N as WL-WMOS-D1-Mini-03...
00:00:05 WIF: Connected
20:50:23 HTP: Web server active on WL-WMOS-D1-Mini-03 with IP address 192.168.2.213
20:50:23 QPC: Reset
20:50:24 MQT: Attempting connection...
20:50:24 MQT: Connected
20:50:24 MQT: Haus/OT/WL_WD1M_03/tele/LWT = Online (retained)
20:50:24 MQT: Haus/OT/WL_WD1M_03/cmnd/POWER = 
20:50:24 MQT: Haus/OT/WL_WD1M_03/tele/INFO1 = {"Module":"Generic","Version":"8.5.1(tasmota)","FallbackTopic":"cmnd/DVES_24820B_fb/","GroupTopic":"tasmotas/cmnd/"}
20:50:24 MQT: Haus/OT/WL_WD1M_03/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"WL-WMOS-D1-Mini-03","IPAddress":"192.168.2.213"}
20:50:24 MQT: Haus/OT/WL_WD1M_03/tele/INFO3 = {"RestartReason":"Software/System restart"}
20:50:27 ZIG: rebooting CC2530 device
20:50:28 MQT: Haus/OT/WL_WD1M_03/tele/STATE = {"Time":"2020-10-04T20:50:28","Uptime":"0T00:00:13","UptimeSec":13,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"98:9B:CB:BE:A6:BF","Channel":1,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:07"}}
20:50:29 MQT: Haus/OT/WL_WD1M_03/tele/RESULT = {"ZbState":{"Status":1,"Message":"CC2530 booted","RestartReason":"Watchdog","MajorRel":2,"MinorRel":6}}
20:50:29 MQT: Haus/OT/WL_WD1M_03/tele/RESULT = {"ZbState":{"Status":50,"MajorRel":2,"MinorRel":6,"MaintRel":3,"Revision":20190608}}
20:50:29 MQT: Haus/OT/WL_WD1M_03/tele/RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
20:50:30 MQT: Haus/OT/WL_WD1M_03/tele/RESULT = {"ZbState":{"Status":40,"NewState":9,"Message":"Started as coordinator"}}
20:50:30 MQT: Haus/OT/WL_WD1M_03/tele/RESULT = {"ZbState":{"Status":51,"IEEEAddr":"0x00124B00199DF8BF","ShortAddr":"0x0000","DeviceType":7,"DeviceState":9,"NumAssocDevices":0}}
20:50:30 MQT: Haus/OT/WL_WD1M_03/tele/RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
20:50:30 ZIG: Zigbee started
20:50:30 ZIG: Zigbee devices data in Flash (836 bytes) `

Riza-Aslan, 2020-10-05:
I've also upgraded my Sonoff ZbBridge from 8.5.0 to 8.5.1 and 3 of 4 Zigbee devices just disappeared. I'm pretty happy that I am still testing and haven't yet moved all my devices over from z2m to z2t.

Riza-Aslan, 2020-10-05:
Looks like the devices (Xiaomi Temperature, Vibration and a Osram Socket) are coming back with its regular update. But the friendlynames are lost.

schrej-zz, 2020-10-05:
@Riza-Aslan,
I think tasmota.bin with zogbee2tasmota enabled ist something different than tasmota-zigbee-bridge.
You have to open a separate issue.
Joerg

schrej-zz, 2020-10-05:
@s-hadinger,
as I have to compile the bin by my own:
Are there any other eg. sensors or features that have to be part of the bin file as zigbee relies on them?

Riza-Aslan, 2020-10-05:
@schrej-zz Oh, sorry. I thought it is the same source. I'll open a seperate issue.

schrej-zz, 2020-10-05:
That's what I did. As Zigbee commands are accepted, I think, the compilation is fine.

schrej-zz, 2020-10-08:
Is anyone successfully running Tasmota with Zigbee2Tasmota enabled after 8.2.0.3?
I still don't get any connection to devices after that version. Going back to 8.2.0.3 successfully gets connection.

s-hadinger, 2020-10-08:
Ah, I suspect that there is an issue with PanID or network key. Starting with 8.2.0.3, the zigbee network configuration moved from compile time constants to Flash settings. Settings were automatically copied from the defaults from compile time. But it's possible that something went wrong here.

Use `ZbConfig` command and double-check the PanID, ExtPanID and network key. Chances are that they are different from the one you used before 8.2.0.3

schrej-zz, 2020-10-08:
@s-hadinger,
thanks for your support.
ZbConfig seems to be still an unknown command in version 8.2.0.3.
Any other chance to read PAN ID or network key with this version?

s-hadinger, 2020-10-08:
In the source code if the command does not exist. Why not try the latest V9.0?

schrej-zz, 2020-10-08:
@s-hadinger,
does this help me in any kind to get back the connection to the devices?
As I said, I see the devices (ZbStatus; 8.5.1 & 9.0.1) but cannot connect them.
Do I have to reset everything and reconnect the devices?
Thanks Joerg

closed 2020-10-08 by s-hadinger