arendst/Tasmota #9466: Crash when PWM1i is defined when using command wakeup

Jason2866, 2020-10-05:
Define a GPIO with function PWM1i. 
Use command `LedTable 1` and `Dimmer 3` after that fire command `wakeup` -> crash.
Sometimes it already crashes with the `Dimmer` command.
Compiling Tasmota @160Mhz for the ESP makes it even worse.

The issue is already discussed in Discord. The crash is proabably caused from the used PWM code.
Tasmota actual uses PR https://github.com/esp8266/Arduino/pull/7022 for Tasmota core 2.7.4.1
( https://github.com/tasmota/Arduino/tree/2.7.4.1-tasmota )

With some! tests it seems the issue is fixed using PR https://github.com/esp8266/Arduino/pull/7231 instead.
( https://github.com/tasmota/Arduino/tree/2.7.4.1-PR7231 )

The shutter part and the AC-Zero-Cross Dimmer needs testing for compability for the changed PWM part.
@stefanbode and everyone else is invited to do tests. Thx in advance for doing.

For doing this download the `provided platformio_override.ini.txt`. Rename it to `platformio_override.ini`and copy it to your
root file of the Tasmota project folder. Rename your override file before ;-) 

[platformio_override.ini.txt](https://github.com/arendst/Tasmota/files/5325646/platformio_override.ini.txt)

-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: latest dev
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: latest Platformio / VSC
- [x] Flashing tools used: Esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
see above

### EXPECTED BEHAVIOUR
no crash

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-05:
The crash with PWM 7022 was a WDT reboot with no stack trace (even not with serial attached). Is the crash with ws2812 similar?

arendst, 2020-10-05:
The crashes with wd2812 were exceptions 0. Bad thing now is I cannot reproduce... Investigating.

arendst, 2020-10-05:
Wakeup exception 0 fixed by latest commit

Jason2866, 2020-10-06:
Did test with div 0 fix. Wakeup does work now in general.
The problem with dimmer and PWM1i inverted and random crash is still there when using Tasmota core (PR 7022 used)
I can generate no crash when using Core with PR7231. I suggest to change for development build to core with PR7231
to get feedback if there are any negative side effects

arendst, 2020-10-06:
Agree. Pls provide pr

s-hadinger, 2020-10-06:
+1. We just need to check the max frequency of 7231 because of impact on pwm fan

closed 2020-10-09 by ascillato2

Jason2866, 2020-10-16:
@s-hadinger  did a test (pwm fan).  Delivers a good signal at frequencys high enough for pwm fans.
So if there is not rising a hidden bug in 7231 it LGTM