arendst/Tasmota #9471: Setting the sensitivity for Aqara Vibration Sensor doesn't work

Riza-Aslan, 2020-10-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If I try to set the sensitivity of the Aqara Vibration Sensor like discribed on the blakadder website, I'm getting an error message

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Join a Aqara Vibration Sensor and try to set the sensitivity with the command: ZbSend {“Device”:”…”,”Manuf”:”0x115F”,”Write”:{“0000/FF0D%20”:1}}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Change of the sensitivity

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-05:
What error message do you get?

Riza-Aslan, 2020-10-06:
Input:
`ZbSend {“Device”:”0x24D5”,”Manuf”:”0x115F”,”Write”:{“0000/FF0D%20”:1}}`

Output:
```
09:07:57 CMD: ZbSend {“Device”:”0x24D5”,”Manuf”:”0x115F”,”Write”:{“0000/FF0D%20”:1}}
09:07:57 SRC: WebConsole from 10.14.8.164
09:07:57 CMD: Group 0, Index 1, Command "ZBSEND", Data "{“Device”:”0x24D5”,”Manuf”:”0x115F”,”Write”:{“0000/FF0D%20”:1}}"
09:07:57 MQT: stat/ZigBee2Tasmota/RESULT = {"ZbSend":"Invalid JSON"}
```

s-hadinger, 2020-10-07:
Oh I see. The quotes were replaced by typographic quotes which are not valid JSON.

Use:
`ZbSend {"Device":"0x24D5","Manuf":"0x115F","Write":{"0000/FF0D%20":1}}`

Riza-Aslan, 2020-10-09:
Oh...good to know.

I'm getting another issue now

`14:39:16 CMD: ZbSend {"Device":"0x24D5","Manuf":"0x115F","Write":{"0000/FF0D%20":1}}
14:39:16 SRC: WebConsole from 10.14.8.164
14:39:16 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x24D5","Manuf":"0x115F","Write":{"0000/FF0D%20":1}}"
14:39:16 ZIG: guessing endpoint 0
14:39:16 MQT: stat/ZigBee2Tasmota/RESULT = {"ZbSend":"Missing endpoint"}`

s-hadinger, 2020-10-09:
The missing endpoint means that Z2T failed to get the endpoint list at pairing time. So you need either to re-pair, or specify exxplicitly the endpoint (1 for Aqara).

Can you try `ZbStatus2 0x24D5` to check internal information about the device.

`ZbSend {"Device":"0x24D5","Endpoint":1,"Manuf":"0x115F","Write":{"0000/FF0D%20":1}}`

Riza-Aslan, 2020-10-09:
Here is the output:
`MQT: stat/ZigBee2Tasmota/RESULT = {"ZbStatus2":[{"Device":"0x24D5","Name":"Wohnzimmer_Vibration","IEEEAddr":"0x0000000000000000","Endpoints":[]}]}`

If I touch the sensor i'm getting this:
`tele/Wohnzimmer_Vibration/SENSOR = {"ZbReceived":{"Wohnzimmer_Vibration":{"Device":"0x24D5","Name":"Wohnzimmer_Vibration","BatteryVoltage":3.01,"BatteryPercentage":100,"Xiaomi_98":40,"Xiaomi_99":209,"Xiaomi_9A":"0x04EF006BFFD2","Endpoint":1,"LinkQuality":108}}}`

s-hadinger, 2020-10-09:
Ok, so the command should work. Press the button and send `ZbSend {"Device":"0x24D5","Endpoint":1,"Manuf":"0x115F","Write":{"0000/FF0D%20":1}}` right after.

ascillato2, 2020-10-11:
Hi, any news on this?

closed 2020-10-11 by ascillato2