arendst/Tasmota #9473: ADC button doesn't work

e9hack, 2020-10-05:
### PROBLEM DESCRIPTION
ADC Button doesn't work anymore after redesign of GPIO's.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund SP112
- [x] Tasmota binary firmware version number used: 9.0.0.1(sensors)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSC / PIO
	- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
21:18:02 MQT: stat/Gosund 1/RESULT = {"NAME":"Generic","GPIO":[321,1,320,1,2656,2720,0,0,2624,257,224,0,0,1],"FLAG":4,"BASE":18}
21:18:02 MQT: stat/Gosund 1/RESULT = {"Module":{"0":"Generic"}}
21:18:02 MQT: stat/Gosund 1/RESULT = {"GPIO0":{"321":"Led_i2"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"257":"Relay_i2"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
```
- [x] Provide the output of this command: `Status 0`:
```
21:18:49 MQT: stat/Gosund 1/STATUS = {"Status":{"Module":0,"DeviceName":"Gosund 1 (Power)","FriendlyName":["Gosund 1 (Power)","Tasmota2"],"Topic":"Gosund 1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
21:18:49 MQT: stat/Gosund 1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:46","StartupUTC":"2020-10-05T19:17:03","Sleep":50,"CfgHolder":4618,"BootCount":7,"BCResetTime":"2020-10-05T21:11:53","SaveCount":21,"SaveAddress":"F7000"}}
21:18:49 MQT: stat/Gosund 1/STATUS2 = {"StatusFWR":{"Version":"9.0.0.1(sensors)","BuildDateTime":"2020-10-05T20:09:41","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8285","CR":"475/699"}}
21:18:49 MQT: stat/Gosund 1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["The Internet of Shitty Things","AndroidAP"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
21:18:49 MQT: stat/Gosund 1/STATUS4 = {"StatusMEM":{"ProgramSize":606,"Free":396,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"xxxxxx","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F8A8787","0405A005","B7FFBFCF","01DA9BC4","24367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,8,9,10,14,16,17,20,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
21:18:49 MQT: stat/Gosund 1/STATUS5 = {"StatusNET":{"Hostname":"Gosund 1-xxx","IPAddress":"192.168.255.34","Gateway":"192.168.255.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.255.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:18:49 MQT: stat/Gosund 1/STATUS6 = {"StatusMQT":{"MqttHost":"mosquitto.osexpert.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_xxxxxx","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:18:49 MQT: stat/Gosund 1/STATUS7 = {"StatusTIM":{"UTC":"2020-10-05T19:18:49","Local":"2020-10-05T21:18:49","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:27","Sunset":"18:51"}}
21:18:49 MQT: stat/Gosund 1/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:18:49 MQT: stat/Gosund 1/STATUS10 = {"StatusSNS":{"Time":"2020-10-05T21:18:49","ENERGY":{"TotalStartTime":"2020-10-05T21:11:53","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":237,"Current":0.000}}}
21:18:49 MQT: stat/Gosund 1/STATUS11 = {"StatusSTS":{"Time":"2020-10-05T21:18:49","Uptime":"0T00:01:46","UptimeSec":106,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"The Internet of Shitty Things","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"RSSI":20,"Signal":-90,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE

### EXPECTED BEHAVIOUR
Button press shall switch on/off the relay or with double press switch on/off the USB charger.

### SCREENSHOTS

### ADDITIONAL CONTEXT
I did setup the configuration on the 'configuration template'. It looks like that the setting of GPIO 17 isn't stored. I did change the entry for FLAG to 4 on 'Configuration Others'.

**(Please, remember to close the issue when the problem has been addressed)**

viktordaniel, 2020-10-06:
I can confirm the same behavior. Worked  on 8.5.1, after upgrade to 9.0.0.1 the button is not working any more.

arendst, 2020-10-06:
Thx. I thought I had a SP112 but couldn't find it so wasn't tested.

Just had an epiphany where I can easily emulate the situation on a wemos.

Will investigate and resolve ASAP.

arendst, 2020-10-06:
Pls give it a try. 

Remember to use the correct template as made available on the template website https://templates.blakadder.com/gosund_SP112_v3_4.html
```
{"NAME":"Gosund 112v3.4","GPIO":[56,0,57,0,132,134,0,0,131,30,21,0,0],"FLAG":4,"BASE":18}
```
or 
```
{"NAME":"Gosund 112v3.4","GPIO":[320,0,321,0,2656,2720,0,0,2624,257,224,0,0,4800],"FLAG":0,"BASE":18}
```
Notice the latter has FLAG set to 0.

viktordaniel, 2020-10-06:
Confirmed, working.

e9hack, 2020-10-06:
It's working now. 
I try to add a template to the SW. I'm able to configure all necessary pin's with exception of ADC button. How can I define the ADC configuration?

closed 2020-10-08 by arendst

4nuts, 2020-11-01:
Same problem with Gosund SP112 on 8.5.1.
Solved by a hard reset and reconfiguring with the original template.
https://templates.blakadder.com/gosund_SP112.html
{"NAME":"SHP5","GPIO":[57,145,56,146,255,22,0,0,255,255,21,255,17],"FLAG":0,"BASE":18}