arendst/Tasmota #9475: scripting json parsing problems - first key/value pair unavailable

jsgotham, 2020-10-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Scripting with USE_WEBSEND_RESPONSE,  in >E section first json pair is not available.  If first key is duplicated, parse always picks up second instance.  Ignores first key/value pair ... adding ':,' to start of json makes the next pair available.  Seems to be the same in >T section for parsing of mqtt payloads.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Pow
- [x] Tasmota binary firmware version number used: 9.0.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:32:09 RSL: stat/tasmota_1A62B3/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:32:09 RSL: stat/tasmota_1A62B3/RESULT = {"Module":{"6":"Sonoff Pow"}}
00:32:10 RSL: stat/tasmota_1A62B3/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2592":"HLWBL SEL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"2656":"HLWBL CF1"},"GPIO14":{"2688":"HLW8012 CF"},"GPIO15":{"288":"Led1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:33:22 CMD: STATUS 0
00:33:22 RSL: stat/tasmota_1A62B3/STATUS = {"Status":{"Module":6,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1A62B3","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T09:20:34","StartupUTC":"2020-10-05T14:12:48","Sleep":50,"CfgHolder":4617,"BootCount":285,"BCResetTime":"2020-10-04T03:59:09","SaveCount":454,"SaveAddress":"F7000"}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS2 = {"StatusFWR":{"Version":"9.0.0.1(tasmota)","BuildDateTime":"2020-10-04T23:01:28","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"345/699"}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gnomenclature2G4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAE797","043E8001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS5 = {"StatusNET":{"Hostname":"tasmota_1A62B3-0691","IPAddress":"10.1.1.21","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"2C:3A:E8:1A:62:B3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1A62B3","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS7 = {"StatusTIM":{"UTC":"2020-10-05T23:33:22","Local":"2020-10-06T00:33:22","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:57","Sunset":"18:17"}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS10 = {"StatusSNS":{"Time":"2020-10-06T00:33:22","ENERGY":{"TotalStartTime":"2020-10-04T03:59:09","Total":0.016,"Yesterday":0.009,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS11 = {"StatusSTS":{"Time":"2020-10-06T00:33:22","Uptime":"0T09:20:34","UptimeSec":33634,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":125,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Gnomenclature2G4","BSSId":"54:25:EA:CF:0C:60","Channel":4,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:05"}}}
00:33:22 RSL: stat/tasmota_1A62B3/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["40247311","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4024751c","40247992","40247a93","40211da2","40253532","40291beb","4024751c","40291beb","40291bec","4025371c","40211b5e","40253532","40211b5e","40252cd1","402131f4","402139f4","402186a9","40252d88","401018fb","40101b83","402271bc","402483ba","40252d88","40101bba","40252d88","40101c01","401018fb","40255e1c","40227869","40101b83","402357da"]}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Multiple different json payloads (see websend RESULT for details in each case)

00:52:27 CMD: websend [10.1.1.5:80] /test.json
00:52:27 SRC: WebConsole from 10.1.1.15
00:52:27 CMD: Group 0, Index 1, Command "WEBSEND", Data "[10.1.1.5:80] /test.json"
00:52:27 RSL: stat/tasmota_1A62B3/RESULT = {"key1":"111","key2":22,"key3":"333"}
00:52:27  Strings - 'key1'='' 'key2'='' 'key3'='333'
00:52:27  Numeric - 'key1'='0.00' 'key2'='22.00' 'key3'='333.00'

00:54:45 CMD: websend [10.1.1.5:80] /test.json
00:54:45 SRC: WebConsole from 10.1.1.15
00:54:45 CMD: Group 0, Index 1, Command "WEBSEND", Data "[10.1.1.5:80] /test.json"
00:54:45 RSL: stat/tasmota_1A62B3/RESULT = {"key1":11,"key2":"222","key3":"333"}
00:54:45  Strings - 'key1'='' 'key2'='222' 'key3'='333'
00:54:45  Numeric - 'key1'='0.00' 'key2'='222.00' 'key3'='333.00'

00:55:52 CMD: websend [10.1.1.5:80] /test.json
[... 35 lines cut ...]
sVar3=""
nVar1=0
nVar2=0
nVar3=0

>E
; reset variables to stop retaining previous values (for testing)
sVar1=""
sVar2=""
sVar3=""
nVar1=0
nVar2=0
nVar3=0
; get json values
sVar1=key1
sVar2=key2
sVar3=key3
nVar1=key1
nVar2=key2
nVar3=key3
; print json values
=>print Strings - 'key1'='%sVar1%' 'key2'='%sVar2%' 'key3'='%sVar3%'
=>print Numeric - 'key1'='%nVar1%' 'key2'='%nVar2%' 'key3'='%nVar3%'

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-10-06:
there was a severe bug in scripter json import (introduced by jsmn) causing erratic behavior and crashes, that was fixed 2 days ago. did you use the very recent dev version ?
if not please try again.

else this seems to be a "jsmn" issue and we have to consult  @s-hadinger 

ps
max Internal string size is limited to 48

jsgotham, 2020-10-06:
Just my luck that I cloned the repository whilst the issue was active :(
JSON parsing seems to be working correctly (and a lot more reliably) with a fresh clone of the development version.

Thanks!

closed 2020-10-06 by jsgotham