arendst/Tasmota #9477: ESP32 BME280 issues (not updateing and bad data)

baerrs, 2020-10-06:
### PROBLEM DESCRIPTION
Hi All !, I'm having issues with a BME280 running on the dev downloaded on 10/5. I'm using Tasmota32-sensor dev build from 10/5. I've set the GIOP4&5 to I2C SDA/L. It appears the data never gets re-read (the data that comes back is not correct). I even place the device in the frig for an hour... still reported the same (bad)data. I have two BME sensor(only one connected at a time), and confirmed both chips on a PI-zero are working and get good results... I used Pi-Zero and BME280pi to test the chips. I'm using an ESP32 Wroom32E

Program Version: 9.0.0.1(sensors)
Build &Time/Date: 2020-10-05T17:07:37
Core/SDK Version: 1_0_4/v3.2.3-14-gd3e562907

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-Wroom32E_____
- [x] Tasmota binary firmware version number used: _9.0.0.1(sensors)____
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA & esptool.py _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
07:44:09 WIF: Checking connection...
07:44:29 WIF: Checking connection...
07:44:49 WIF: Checking connection...
07:45:09 WIF: Checking connection...
07:45:19 CMD: Backlog Template; Module; GPIO 255:
07:45:19 SRC: WebConsole from 192.168.1.141
07:45:19 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255:"
07:45:19 SRC: Backlog
07:45:19 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
07:45:19 MQT: stat/tasmota_CF14A8/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
07:45:19 SRC: Backlog
07:45:19 CMD: Group 0, Index 1, Command "MODULE", Data ""
07:45:19 MQT: stat/tasmota_CF14A8/RESULT = {"Module":{"1":"ESP32-DevKit"}}
07:45:19 SRC: Backlog
07:45:19 CMD: Group 0, Index 1, Command "GPIO", Data "255:"
07:45:19 MQT: stat/tasmota_CF14A8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None ...
07:45:29 WIF: Checking connection...
07:45:42 MQT: tele/tasmota_CF14A8/HASS_STATE = {"Version":"9.0.0.1(sensors)","BuildDateTime":"2020-10-05T17:07:37","Module or Template":"ESP32-DevKit","RestartReason":"Software reset CPU","Uptime":"0T01:10:02","Hostname":"tasmota_CF14A8-5288","IPAddress":"192.168.1.144","RSSI":"90","Signal (dBm)":"-55","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:05","MqttCount":9,"LoadAvg":22}
07:45:49 WIF: Checking connection...
07:45:51 MQT: tele/tasmota_CF14A8/STATE = {"Time":"2020-10-06T07:45:51","Uptime":"0T01:10:11","UptimeSec":4211,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":9,"Wifi":{"AP":1,"SSId":"SmartOne2","BSSId":"14:91:82:30:50:8A","Channel":11,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:05"}}
07:45:51 MQT: tele/tasmota_CF14A8/SENSOR = {"Time":"2020-10-06T07:45:51","BME280":{"Temperature":21.5,"Humidity":93.0,"DewPoint":20.3,"Pressure":647.4},"PressureUnit":"hPa","TempUnit":"C"}
07:45:59 MQT: Attempting connection...
07:45:59 MQT: Connected
07:45:59 MQT: tele/tasmota_CF14A8/LWT = Online (retained)
07:45:59 MQT: cmnd/tasmota_CF14A8/POWER = 
07:45:59 MQT: Subscribe to cmnd/tasmota_CF14A8/#
07:45:59 MQT: Subscribe to cmnd/tasmotas/#
07:45:59 MQT: Subscribe to cmnd/DVES_CF14A8_fb/#
07:45:59 MQT: Subscribe to homeassistant/status
07:46:09 WIF: Checking connection...
  Configuration output here:

```
- [x]  If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
07:50:03 CMD: Backlog Rule1; Rule2; Rule3
07:50:03 SRC: WebConsole from 192.168.1.141
07:50:03 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
07:50:03 SRC: Backlog
07:50:03 CMD: Group 0, Index 1, Command "RULE", Data ""
07:50:03 MQT: stat/tasmota_CF14A8/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
07:50:04 SRC: Backlog
07:50:04 CMD: Group 0, Index 2, Command "RULE", Data ""
07:50:04 MQT: stat/tasmota_CF14A8/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
07:50:04 SRC: Backlog
07:50:04 CMD: Group 0, Index 3, Command "RULE", Data ""
07:50:04 MQT: stat/tasmota_CF14A8/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
[... 15 lines cut ...]
07:46:57 MQT: stat/tasmota_CF14A8/STATUS7 = {"StatusTIM":{"UTC":"2020-10-06T06:46:57","Local":"2020-10-06T07:46:57","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:57","Sunset":"18:17"}}
07:46:57 MQT: stat/tasmota_CF14A8/STATUS10 = {"StatusSNS":{"Time":"2020-10-06T07:46:57","BME280":{"Temperature":21.5,"Humidity":93.0,"DewPoint":20.3,"Pressure":647.4},"PressureUnit":"hPa","TempUnit":"C"}}
07:46:57 MQT: stat/tasmota_CF14A8/STATUS11 = {"StatusSTS":{"Time":"2020-10-06T07:46:57","Uptime":"0T01:11:17","UptimeSec":4277,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":10,"Wifi":{"AP":1,"SSId":"SmartOne2","BSSId":"14:91:82:30:50:8A","Channel":11,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-10-06:
Cannot reproduce. 

My BMP280 works as expected:
```
12:33:40 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:33:40","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.6,"Humidity":39.9,"DewPoint":9.1},"BMP280":{"Temperature":24.6,"Pressure":996.8},"PressureUnit":"hPa","TempUnit":"C"}
12:34:07 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:34:07","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.6,"Humidity":39.8,"DewPoint":9.1},"BMP280":{"Temperature":24.7,"Pressure":996.8},"PressureUnit":"hPa","TempUnit":"C"}
12:34:27 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:34:27","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.7,"Humidity":40.1,"DewPoint":9.3},"BMP280":{"Temperature":24.7,"Pressure":996.8},"PressureUnit":"hPa","TempUnit":"C"}
12:34:47 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:34:47","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.7,"Humidity":47.1,"DewPoint":11.7},"BMP280":{"Temperature":24.8,"Pressure":996.8},"PressureUnit":"hPa","TempUnit":"C"}
12:35:07 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:35:07","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.8,"Humidity":41.7,"DewPoint":10.0},"BMP280":{"Temperature":24.8,"Pressure":996.8},"PressureUnit":"hPa","TempUnit":"C"}
12:35:27 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:35:27","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.7,"Humidity":40.9,"DewPoint":9.6},"BMP280":{"Temperature":25.8,"Pressure":996.8},"PressureUnit":"hPa","TempUnit":"C"}
```
same for BME680:
```
12:41:04 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:41:04","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.5},"AM2301":{"Temperature":23.7,"Humidity":40.5,"DewPoint":9.5},"BME680":{"Temperature":24.8,"Humidity":54.1,"DewPoint":14.9,"Pressure":996.9,"Gas":20.32},"PressureUnit":"hPa","TempUnit":"C"}
12:41:44 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:41:44","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.6,"Humidity":40.8,"DewPoint":9.5},"BME680":{"Temperature":25.0,"Humidity":52.3,"DewPoint":14.6,"Pressure":996.8,"Gas":44.45},"PressureUnit":"hPa","TempUnit":"C"}
12:42:04 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:42:04","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.7,"Humidity":40.5,"DewPoint":9.5},"BME680":{"Temperature":25.0,"Humidity":52.0,"DewPoint":14.5,"Pressure":996.8,"Gas":57.51},"PressureUnit":"hPa","TempUnit":"C"}
12:42:24 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:42:24","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.6,"Humidity":40.2,"DewPoint":9.3},"BME680":{"Temperature":25.1,"Humidity":51.7,"DewPoint":14.5,"Pressure":996.8,"Gas":70.89},"PressureUnit":"hPa","TempUnit":"C"}
12:42:44 MQT: tele/esp32c/SENSOR = {"Time":"2020-10-06T12:42:44","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.6,"Humidity":40.2,"DewPoint":9.3},"BME680":{"Temperature":25.1,"Humidity":51.7,"DewPoint":14.4,"Pressure":996.9,"Gas":84.53},"PressureUnit":"hPa","TempUnit":"C"}
```
I have no BME280.

on v9.0.0.1:
```
12:37:24 CMD: Group 0, Index 1, Command "STATUS", Data "0"
12:37:24 MQT: stat/esp32c/STATUS = {"Status":{"Module":1,"DeviceName":"Esp32C","FriendlyName":["Esp32C"],"Topic":"esp32c","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:37:24 MQT: stat/esp32c/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-sensors.bin","RestartReason":"Software reset CPU","Uptime":"0T00:03:57","StartupUTC":"2020-10-06T10:33:27","Sleep":50,"CfgHolder":4617,"BootCount":66,"BCResetTime":"2020-04-16T14:39:15","SaveCount":128}}
12:37:24 MQT: stat/esp32c/STATUS2 = {"StatusFWR":{"Version":"9.0.0.1(sensors)","BuildDateTime":"2020-10-06T08:30:15","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"358/699"}}
12:37:24 MQT: stat/esp32c/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["indebuurt","indebuurt_IoT"],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","000000C0","00006000","00000000"]}}
12:37:24 MQT: stat/esp32c/STATUS4 = {"StatusMEM":{"ProgramSize":1110,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","00152005","B7FFBFCF","01DA9BC4","60360CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,14,17,20,21,24,29,34","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,64,66,67,74"}}
12:37:24 MQT: stat/esp32c/STATUS5 = {"StatusNET":{"Hostname":"esp32c","IPAddress":"192.168.2.145","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"84:0D:8E:36:07:5C","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
12:37:24 MQT: stat/esp32c/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_36075C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
[... cut ...]
12:37:24 MQT: stat/esp32c/STATUS10 = {"StatusSNS":{"Time":"2020-10-06T12:37:24","DS18B20":{"Id":"01144A0CB2AA","Temperature":22.6},"AM2301":{"Temperature":23.7,"Humidity":40.2,"DewPoint":9.3},"BMP280":{"Temperature":24.9,"Pressure":996.7},"PressureUnit":"hPa","TempUnit":"C"}}
12:37:24 MQT: stat/esp32c/STATUS11 = {"StatusSTS":{"Time":"2020-10-06T12:37:24","Uptime":"0T00:03:57","UptimeSec":237,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```

sfromis, 2020-10-06:
I'm having no problems with BME280 on Tasmota32 9.0.0.1. While using gipo4/5 is the "standard" pins on ESP8266, I'd avoid that on ESP8266, as the "usual" suggestion is SDA on gpio21 and SCL on gpio22, as seen [here](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/). That page also has a note about gpio5 outputting a PWM signal at boot, I speculate that this might confuse the sensor. I'd suggest trying again with gpio21/22 and see if you get better results.

arendst, 2020-10-06:
Those were my original GPIOs too.

baerrs, 2020-10-07:
I tried gpio21&22 with the same results.  I'm going to verify my hardware set up, I'll keep this post updated.

ascillato2, 2020-10-09:
Hi, any news on this?

closed 2020-10-11 by ascillato2

baerrs, 2020-10-14:
I still have the issue, and am still looking.  I'm ramping up on i2c, and c... 
I have since replaced the bme280 again, so this is my 3rd one, all with the same results.     What I've read and am currently looking at  isthe humidity has to be set before the temp & pressure, and then give it a min of 4ms delay before reading.. other wise you may get the old results (which is one of my two problems.

czikus, 2020-11-22:
I can confirm that I'm seeing the exact same issue with my BME280 and ESP32.

czikus, 2020-11-22:
Can we reopen this issue?

Solution-Builders-Alex, 2021-06-15:
I have same issue

00:05:28 MQT: tele/sens1/SENSOR = {"Time":"1970-01-01T00:05:28","BME280":{"Temperature":22.4,"Humidity":42.6,"DewPoint":9.0,"Pressure":656.6},"PressureUnit":"hPa","TempUnit":"C"}

robmool, 2021-06-27:
I have the same issue with my BME280. I have tried with two BME280's and both have the same issue. The problem occurs randomly after some hours or days. 

Additional information:
Device used: ESP8266-01
Pre-compiled version: 9.4.0 sensors
Flash tools used: NodeMCU PyFlasher 4.0

17:57:28.076 CMD: Backlog Template; Module; GPIO 255
17:57:28.111 MQT: myhome/stat/tasmota_238875/RESULT = {"NAME":"BME280","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:57:28.325 MQT: myhome/stat/tasmota_238875/RESULT = {"Module":{"0":"BME280"}}
17:57:28.579 MQT: myhome/stat/tasmota_238875/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"608":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"640":"I2C SDA"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

Sensor readings:
{"Time":"2021-06-26T04:07:35","BME280":{"Temperature":23.8,"Humidity":40.5,"DewPoint":9.5,"Pressure":945.7,"SeaPressure":1018.1},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:12:35","BME280":{"Temperature":23.8,"Humidity":40.4,"DewPoint":9.5,"Pressure":945.8,"SeaPressure":1018.1},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:17:36","BME280":{"Temperature":23.8,"Humidity":40.3,"DewPoint":9.5,"Pressure":945.8,"SeaPressure":1018.2},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:22:35","BME280":{"Temperature":23.8,"Humidity":40.3,"DewPoint":9.5,"Pressure":945.9,"SeaPressure":1018.2},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:27:35","BME280":{"Temperature":23.8,"Humidity":40.4,"DewPoint":9.5,"Pressure":945.9,"SeaPressure":1018.3},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:32:35","BME280":{"Temperature":21.1,"Humidity":79.8,"DewPoint":17.5,"Pressure":661.9,"SeaPressure":706.0},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:37:36","BME280":{"Temperature":21.1,"Humidity":79.8,"DewPoint":17.5,"Pressure":661.9,"SeaPressure":706.0},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:42:36","BME280":{"Temperature":21.1,"Humidity":79.8,"DewPoint":17.5,"Pressure":661.9,"SeaPressure":706.0},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:47:36","BME280":{"Temperature":21.1,"Humidity":79.8,"DewPoint":17.5,"Pressure":661.9,"SeaPressure":706.0},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:52:36","BME280":{"Temperature":21.1,"Humidity":79.8,"DewPoint":17.5,"Pressure":661.9,"SeaPressure":706.0},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-06-26T04:57:35","BME280":{"Temperature":21.1,"Humidity":79.8,"DewPoint":17.5,"Pressure":661.9,"SeaPressure":706.0},"PressureUnit":"hPa","TempUnit":"C"}

Could someone re-open this issue?

Kind regards,
Rob

ascillato, 2021-06-27:
@robmool

Hi, sorry but the GPIOs you are using are not recommended for I2C due to at boot time the ESP8266 sends a lot of information (hardware limitation) and that can confuse your sensor.

Use other GPIOs are recommend in the Tasmota's documentation.

Thanks.

robmool, 2021-06-28:
So basically you're saying that I cannot use the ESP8266-01 chip. The available GPIO pins on this chip are: GPIO0, GPIO1, GPIO2, GPIO3.

Any other suggestions?

Best regards,
Rob

ascillato, 2021-06-28:
GPIO3 is RX and GPIO2 is normal I/O. Try those.

robmool, 2021-06-28:
Thank you for your fast response! I will change the hardware setup and I keep you posted with new info.

robmool, 2021-07-10:
Changed the GPIO ports as indicated above. After running the ESP8266-01 for 9 days I got wrong readings again:
{"Time":"2021-07-08T18:41:29","BME280":{"Temperature":25.0,"Humidity":36.3,"DewPoint":9.0,"Pressure":949.9,"SeaPressure":1022.6},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-07-08T18:46:29","BME280":{"Temperature":25.0,"Humidity":36.5,"DewPoint":9.1,"Pressure":949.9,"SeaPressure":1022.7},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-07-08T18:51:29","BME280":{"Temperature":21.1,"Humidity":79.8,"DewPoint":17.5,"Pressure":661.9,"SeaPressure":706.0},"PressureUnit":"hPa","TempUnit":"C"}
{"Time":"2021-07-08T18:56:29","BME280":{"Temperature":21.1,"Humidity":79.8,"DewPoint":17.5,"Pressure":661.9,"SeaPressure":706.0},"PressureUnit":"hPa","TempUnit":"C"}

I might think that this is a SW issue. Any suggestions? 

By the way I am also a SW engineer and could help debugging this issue if necessary.

ascillato, 2021-07-10:
I couldn't make my device to fail. Have you tried another sensor?

robmool, 2021-07-10:
I already tried with 2 different devices.  Both have the same problem. I will try with a 3th device.

As a temporary solution I send the command "restart 1" to the device, this will give me the correct readings.

ascillato, 2021-07-10:
How are you powering your device?

sfromis, 2021-07-10:
To me, the patterns I see with similar failures looks like a hardware issue, not a software fault. Possibly related to power fluctuations, adding a suitable decoupling capacitor might be helpful. Maybe the driver could be amended to better handle this type of frozen bad values, that would depend on the details.

robmool, 2021-07-10:
The powering is as follows:
220 VAC - DC 5V adapter - 3.3V regulator - ESP8266-01 + BME280.

robmool, 2021-07-10:
By the way, my BME280 device has an on-board 3.3V LDO regulator. I am using this device:
https://www.amazon.es/gp/product/B07FS95JXT/ref=ppx_yo_dt_b_search_asin_image?ie=UTF8&psc=1

ascillato, 2021-07-10:
How much current they can provide. Or what is the power rating in W of yours? Voltage specification is not enough to determine if the power source is suitable. For example, a 3.3V power source of less than 1W and without ripple filtered output won't be enough for your ESP8266+SENSORS.

robmool, 2021-07-10:
My DC 5V adapter can provide 2A (10W).

sfromis, 2021-07-10:
It is not important what the 5V adapter can provide, as long as it is not too low. What counts is what the LDO supplying the ESP8266 can deliver.

robmool, 2021-07-10:
I am using a 7333-A LDO which can deliver a maximum output current of 250mA.

ascillato, 2021-07-10:
>I am using a 7333-A LDO which can deliver a maximum output current of 250mA

That is good, but from the ESP8266 datasheet, it consumes 50mA and can reach peaks to close to 500mA while transmitting. So, your LDO have a capacitor at the output to provide this extra peaks when needed? If not, maybe your issue is just a dropout of power in some scenarios that make your device to fail (we saw this scenario with some sonoff Basics that have a bad design in its power regulator and it is all the time in the border between operational and failure). This is just one possibility. We have to be sure that the hardware is working ok due to we can't reproduce this issue. I'm not saying that it is bug free, I think that we must be sure it is not hardware related to find the exact condition that makes it fail.

robmool, 2021-07-10:
I have a 2.2uF ceramic capacitor at the LDO output.

robmool, 2021-07-11:
I just replaced the 2.2uF ceramic capacitor with two 10uF electrolytic capacitors on the LDO input/output according to the LDO datasheet.

sfromis, 2021-07-22:
Assuming that you do not have INA219 or a Multi Gas Sensor, which have quite different I2C addresses, those popping up looks like some sort of hardware issue. Could be defects, bad wiring/connections or maybe power quality issues. Not all gpios are quite the same, but I'd expect an inappropriate selection to fail fast.

robmool, 2021-09-03:
Since I replaced the capacitors on the LDO  (see post July 11) I haven't seen any bad reading.

Koddra, 2023-03-03:
I have fixed this issue of data not being updated by changing the FORCED sampling mode to NORMAL. Hope that this helps.