arendst/Tasmota #9481: Support for 20€ BLE Thermostat Equiva CC-RT-BLE

Damontir, 2020-10-06:
**Have you looked for this feature in other issues and in the docs?**  

Jep

**Is your feature request related to a problem? Please describe.**  
No Problem

**Describe the solution you'd like**  
Like to see my Thermostats in Tasmota + HM10

**Describe alternatives you've considered**  
Should be possible, if not, thats ok.

**Additional context**  
Hey :)

Here since 5.1 GREAT JOB! Cant live without.

Just installed the HM10 Module with Tasmota to test and happy with my MiFlora Sensors.

Unfortunately i have 5 CC-RT-BLE Thermostats.

There are scripts out there, but nothing goes over Tasmota so WOULD LOVE TO SEE IT HERE <3

https://github.com/rytilahti/python-eq3bt
https://github.com/Heckie75/eQ-3-radiator-thermostat
https://github.com/softypit/esp32_mqtt_eq3
https://github.com/zewelor/bt-mqtt-gateway
and some more.

Thanks guys 👍

**(Please, remember to close the issue when the problem has been addressed)**

Pewibe, 2020-10-12:
I like to support this request of @Damontir. I'm wondering that Tasmota (target: use cheap devices with comfort by modular software) has no category "heating & radiators" as focus because nearly 30 mio. households in Germany have the demand for such control.  

The EQ-3 CC-RT-BLE thermostats are a good choice with a reasonable price of 20 € less. If you don't have them in stock, comparable and even cheaper is EUROtronic Genius BLE 100 for 15€ less. Additionally Eurotronic has a family of similar devices with alternative radio systems to BLE, like DECT, ZigBee, Zwave, WiFi (lowest power consumption of all).

If it comes to a decision that Tasmota somehow supports radiator control, the restriction to EQ-3 CC-RT-BLE is OK for a beta test, but the software specification should be build more general to support as much as possible radio systems. To profit from existing Tasmota time schedules, etc. on one hand and to lower the effort for controlling the thermostat on the other hand, the development should start with a subset of commands/status (actual temperature, target temperature, boost on/off, battery power).

Pewibe, 2020-10-13:
After a phone call with EUROtronic concerning thermostats "Comet Wifi" I like to add the info that this product is brand new and first pieces are delivered to Hornbach (Germany) this week. In accordance with the data sheet the product has lowest power consumption of all their radiator controls. Some questions dedicated to the product are still open and I will post the answers somehow, if they are of value for Tasmota.

Damontir, 2020-10-13:
Please let me know your finds, they look promising. Will try to help testing, thats all i can do.

Damontir, 2020-11-14:
Hello Mr. Stale Bot, i am quite interested in this solution...

closed 2020-12-15 by github-actions[bot]

Ysbrand, 2021-04-11:
> **Have you looked for this feature in other issues and in the docs?**
>
> Jep
>
> **Is your feature request related to a problem? Please describe.**
> No Problem
>
> **Describe the solution you'd like**
> Like [...]

Have a look here (still in development, but getting closer):
[https://github.com/btsimonh/Tasmota/tree/EQ3-2021-04-08](url)

MBadberg, 2021-12-06:
> Have a look here (still in development, but getting closer):
> [https://github.com/btsimonh/Tasmota/tree/EQ3-2021-04-08](url)

There is nothing? Anywhere else some update about that?
I have the comet wifi now, and the software sucks and it would be great, if tasmota supports such things

Ysbrand, 2021-12-06:
It's part of the TAS 10.0 ESP32-bluetooth release, [http://sidweb.nl/tasmota32/release-10.0.0/tasmota32-bluetooth.bin](url)