arendst/Tasmota #9487: Shutter<x>#Target is not triggered at start of movement (Tasmota 8.2)

andy5macht, 2020-10-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Shutter#Position is not triggered at start und during movement reporting actual position (%value%=0-100)
It is triggered only on end of movement.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Luani - ESP8266 230V I/O Modul
- [x] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use This Rule on Master:
Rule3 ON Shutter1#Target DO Backlog WebSend [192.168.0.201] ShutterPosition1 %value%; WebSend [192.168.0.202] ShutterPosition1 %value%; WebSend [192.168.0.203] ShutterPosition1 %value% ENDON Rule3 1 
Set Shutter to any Target via slider in Web-Interface
When master stops move, the other shutter start to move to target.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Shutter#Position is triggered at start, during and at the end of movement reporting actual position (%value%=0-100)
=> Master should send target on start of movement

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-10-07:
have you thought about using a group topic to move shutters together? I can see every second a message sending all information. I do not know if rules really catch this up, but this is a rules problem
```

08:57:19 MQT: hm/status/tasmota_A9D47F/POWER1 = ON (retained)
08:57:19 MQT: hm/status/tasmota_A9D47F/RESULT = {"ShutterClose1":0}
08:57:19 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
08:57:20 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":98,"Direction":-1,"Target":0}}
08:57:21 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":94,"Direction":-1,"Target":0}}
08:57:22 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":91,"Direction":-1,"Target":0}}
08:57:23 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":88,"Direction":-1,"Target":0}}

```

andy5macht, 2020-10-07:
I dont want to use a MQT-Server.
If my "master Tasmota device" can be used as server. I will have a look into it.

stefanbode, 2020-10-07:
just created an invalid rule and it gets fired all the time. see logfile

.
```
17:42:39 RUL: Stored uncompressed, would compress from 34 to 26 (-24%)
17:42:39 MQT: hm/status/tasmota_A9D47F/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":34,"Free":477,"Rules":"on Shutter1#target do %val1% endon"}
17:42:44 CMD: rule1 on
17:42:45 MQT: hm/status/tasmota_A9D47F/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":34,"Free":477,"Rules":"on Shutter1#target do %val1% endon"}
17:42:57 CMD: shutterposition 10
17:42:57 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER2":"ON"}
17:42:57 MQT: hm/status/tasmota_A9D47F/POWER2 = ON (retained)
17:42:58 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"ON"}
17:42:58 MQT: hm/status/tasmota_A9D47F/POWER1 = ON (retained)
17:42:58 MQT: hm/status/tasmota_A9D47F/RESULT = {"ShutterPosition1":10}
17:42:58 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":10}}
17:42:58 RUL: SHUTTER1#TARGET performs "%val1%"
17:42:58 RUL: SHUTTER1#TARGET performs "%val1%"
17:42:58 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":98,"Direction":-1,"Target":10}}
17:42:58 RUL: SHUTTER1#TARGET performs "%val1%"
17:42:59 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":93,"Direction":-1,"Target":10}}
17:42:59 RUL: SHUTTER1#TARGET performs "%val1%"
17:43:00 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":88,"Direction":-1,"Target":10}}
```

stefanbode, 2020-10-07:
All fine on my side
```
18:07:29 CMD: shutterposition 30
18:07:29 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER2":"OFF"}
18:07:29 MQT: hm/status/tasmota_A9D47F/POWER2 = OFF (retained)
18:07:29 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"ON"}
18:07:29 MQT: hm/status/tasmota_A9D47F/POWER1 = ON (retained)
18:07:29 MQT: hm/status/tasmota_A9D47F/RESULT = {"ShutterPosition1":30}
18:07:29 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":20,"Direction":1,"Target":30}}
18:07:29 RUL: SHUTTER1#TARGET performs "30"
18:07:29 MQT: hm/status/tasmota_A9D47F/RESULT = {"Command":"Unknown"}
18:07:30 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":25,"Direction":1,"Target":30}}
18:07:30 RUL: SHUTTER1#TARGET performs "30"
18:07:30 MQT: hm/status/tasmota_A9D47F/RESULT = {"Command":"Unknown"}
18:07:31 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"OFF"}
18:07:31 MQT: hm/status/tasmota_A9D47F/POWER1 = OFF (retained)
18:07:31 MQT: hm/status/tasmota_A9D47F/SHUTTER1 = 30 (retained)
18:07:31 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":30,"Direction":0,"Target":30}}
18:07:31 RUL: SHUTTER1#TARGET performs "30"
18:07:31 MQT: hm/status/tasmota_A9D47F/RESULT = {"Command":"Unknown"}

```

stefanbode, 2020-10-07:
I assume your problem is, that the command is fired every second. To overcome the issue you have to change the rule to ONCE. than you can create another rule shutter#moved and use this to activate your first rule again
```
rule1 on shutter1#target do WebSend [192.168.0.203] ShutterPosition1 %value% ENDON
rule1 5
rule2 on shutter#moved do rule1 5 endon
rule2 1
```

stefanbode, 2020-10-07:
and @andy5macht  : can you check this on the latest release? I see 8.2. this might be a bit old.

ascillato2, 2020-10-09:
@andy5macht 

Please, when you have time, tell us if updating solves your issue. Thanks.

closed 2020-10-09 by ascillato2

andy5macht, 2020-10-09:
Sorry for late reply. I will tell you if this is working with newer version or modified rules when i get time to test (again). Sorry, sometimes I need to do other things first - renovating a house atm. But i will reply as fast as I can.

andy5macht, 2020-10-11:
> I assume your problem is, that the command is fired every second. To overcome the issue you have to change the rule to ONCE. than you can create another rule shutter#moved and use this to activate [...]

this did not help. Same behavior as without ONCE.
An update to 8.5.1 solved the issue.
Thank you and sorry for the already fixed "issue"