arendst/Tasmota #9489: sunset/sunrise floating point done every minute instead of once per day

cybermaus, 2020-10-07:
### PROBLEM DESCRIPTION
ApplyTimerOffsets() and below that DuskTillDawn() is called every minute from TimerEverySecond(), even though they remain unchanged the entire day. DuskTillDawn() contain quite a few floating and trigonometry functions and are thus not ideal for performance and power consumption.

We could save 1439 out of 1440 calls to these floating and trigonometry functions calls.

It would require calling ApplyTimerOffsets() only once at 00:00 and storing them in a array, similar like what was done for TimerSetRandomWindows() 

Additionally of course, also calling it when a timer configuration is changed).

I would not mind doing this myself, but it would be my first actual PR, but I would like to know if I am not completely wrong, and also that the PR has some chance of being accepted.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Basic 
- [X] Tasmota binary firmware version number used: 8.0.0.1
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VisualStudio/PlatformIO
- [X] Flashing tools used: Tasmotiser
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
13:58:48 CMD: Backlog Template; Module; GPIO 255
13:58:49 RSL: RESULT = {"NAME":"Generic","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
13:58:49 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
13:58:49 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
13:59:59 CMD: Backlog Rule1; Rule2; Rule3
13:59:59 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
14:00:00 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
14:00:00 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [X] Provide the output of this command: `Status 0`:
```
14:00:51 CMD: STATUS 0
14:00:51 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_02E838","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:00:51 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Power On","Uptime":"75T20:59:56","StartupUTC":"2020-10-06T16:00:55","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2020-10-06T18:01:05","SaveCount":5,"SaveAddress":"F7000"}}
14:00:51 RSL: STATUS2 = {"StatusFWR":{"Version":"9.0.0.1(tasmota)","BuildDateTime":"2020-10-06T17:41:53","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"278/699"}}
14:00:51 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Devices",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008000","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
14:00:51 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1420BA","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
14:00:51 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_02E838-2104","IPAddress":"192.168.189.24","Gateway":"192.168.189.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.189.1","Mac":"CC:50:E3:02:E8:38","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
14:00:51 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-12-21T13:00:51","Local":"2020-12-21T14:00:51","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"255:2","Sunset":"255:2"}}
14:00:51 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-12-21T14:00:51"}}
14:00:51 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-12-21T14:00:51","Uptime":"75T20:59:56","UptimeSec":6555596,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Devices","BSSId":"A2:05:43:13:53:B9","Channel":13,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
NA
```

### TO REPRODUCE
NA

### EXPECTED BEHAVIOUR
Reduce calls to ApplyTimerOffsets()

### SCREENSHOTS
NA

### ADDITIONAL CONTEXT
I am actually working on some personal timer mods, while I found this.
It seemed like a nice, contained, yet usefull sidequest to go a bit more official rather then having personal mods.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-07:
No fear :-)  Provide a PR. If the idea is right the reviewers will you advice what to change.
But still there is no merging gurantee.
Thx for sharing your findings and ideas!

s-hadinger, 2020-10-07:
It depends on how much memory you need. Memory tend to be more precious than CPU on esp8266.

sfromis, 2020-10-07:
Activating a timer on sunset did not increase the LoadAvg on a bulb, in a small test. Actual benefit may be more important than doing premature optimization.

cybermaus, 2020-10-07:
See, I am learning already on what to consider for these things.
Maybe memory is more important then CPU.
Than again, but for people trying battery and sleep modes, CPU also means energy. Though that is not my personal use case.

Anyway, got it mostly working (struggling to see where to pick up the web-based config update, though command and device init are already OK)
Seems it cost 32 bytes (16 timers x uint16_t)

s-hadinger, 2020-10-07:
We can definitely afford 32 bytes.

cybermaus, 2020-10-08:
Ok, seems to work. I am studying how to do a PR.
The price is 32 bytes RAM and 112 bytes code, the profit 1439 calls/day to a heavy trigonometry routine, per timer.
I am not quite sure how to demonstrate actual real-world benefits, other then this us my learning curve.
````
Updated
RAM:   [======    ]  58.0% (used 47524 bytes from 81920 bytes)
Flash: [======    ]  59.5% (used 609472 bytes from 1023984 bytes)
Original
RAM:   [======    ]  58.0% (used 47492 bytes from 81920 bytes)
Flash: [======    ]  59.5% (used 609360 bytes from 1023984 bytes)
````
Pretty surprising on those 32 bytes. I added an extra uint32_t to remember the day, so I was sure it would have been 32+4=36 bytes.
Can anyone give me a hint why that global variable is not in the RAM? Somehow optimized onto the stack? Lucky longword alignment? Something?
I added the 2nd and 4th line. Why only 32 bytes extra RAM used?
````
int16_t timer_last_minute = 60;
uint32_t timer_last_day = 0;                 // date for which timer windows and duskdawn is calculated
int8_t timer_window[MAX_TIMERS] = { 0 };
int16_t timer_duskdawn[MAX_TIMERS] = { 0 };  // dusk or dawn calculated for timer
````

s-hadinger, 2020-10-08:
This is strange. It should be indeed 36 bytes. One reason might be that you declared the variable but actually didn't use it in reachable code. You can check in the map file if `timer_last_day` was allocated in RAM.

cybermaus, 2020-10-08:
Found it. 

It does use 36 bytes (0x24 bytes) in the map. But then a half a dozen pages further down, this indeed has a lucky disappearance of a 4-byte filler to align to a longword for the .bss.Settings  declaration

s-hadinger, 2020-10-08:
Good finding!

Staars, 2020-10-08:
>I am not quite sure how to demonstrate actual real-world benefits, other then this is my learning curve.

You could try to measure the ticks before/after the changes. Like described here:
https://sub.nanona.fi/esp8266/timing-and-ticks.html

It would be interesting to see your results.

arendst, 2020-10-08:
I see no real gain in making changes at the cost of memory and more important code space.

To make you even more eary the function is called for every possible timer at every minute so it's 1440 * 16.

A simple change I'll do anyway is moving the calculation below the arm test so that it is only executed when a timer is actually armed.

```
      for (uint32_t i = 0; i < MAX_TIMERS; i++) {
        Timer xtimer = Settings.timer[i];
        if (xtimer.arm) {
#ifdef USE_SUNRISE
          if ((1 == xtimer.mode) || (2 == xtimer.mode)) {      // Sunrise or Sunset
            ApplyTimerOffsets(&xtimer);
          }
#endif
          int32_t set_time = xtimer.time + timer_window[i];  // Add random time offset
```

Doing the ApplyTimerOffset only at midnight will fail when a restart occurs during the day. it also needs storage for every timer as you already noticed.

cybermaus, 2020-10-08:
Ok, so RAM is really on ration then.

Ok a bit of a let down, but no real problem. I will not probably not even keep this one as personal mod, because of the hassle it would bring each merge. 

As to your suggestion: ```Doing the ApplyTimerOffset only at midnight will fail when a restart occurs```

Well, yeah, but I had that covered by also remembering the day the offsets were calculated for. A simple 
```if (RtcTime.days != timer_last_day) {  ```
to replace     
```if (!RtcTime.hour && !RtcTime.minute && !RtcTime.second) { ```
dealt with all cases: Midnight as well as INIT, NTP adjustments, TZ and DST changes, and time \<x\> commands.
I made sure to test them all, I think I covered all cases. But yes, it did cost 36 bytes.

cybermaus, 2020-10-08:
I guess this can be closed now

closed 2020-10-08 by cybermaus