arendst/Tasmota #9492: Shutter problem

Tinus78, 2020-10-07:
### PROBLEM DESCRIPTION
Using a 2 relay switch as rolling shutter switch, connecting to Domoticz.
When the open command is given with domoticz, the shutter opens, but automatically closes again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Tuya wifi wall switch 2 gang
- [x ] Tasmota binary firmware version number used: 8.5.1_____
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: ___Tasmotizer__
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"2 gang","GPIO":[0,157,0,17,18,0,0,0,0,21,22,0,0],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
stat/sonoff/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:29:31 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:24","StartupUTC":"2020-10-07T18:21:07","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-10-07T19:07:59","SaveCount":66,"SaveAddress":"FA000"}}
19:29:31 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"343/699"}}
19:29:31 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BluelanHS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000000","00006000","00000000"]}}
19:29:31 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:29:31 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3114","IPAddress":"192.168.1.119","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:F4:32:AF:CC:2A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:29:31 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.190","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AFCC2A","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:29:31 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"2020-10-07T18:29:31","Local":"2020-10-07T19:29:31","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:59","Sunset":"18:15"}}
19:29:31 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2020-10-07T19:29:31","Shutter1":{"Position":0,"Direction":0,"Target":0}}}
19:29:31 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2020-10-07T19:29:31","Uptime":"0T00:08:24","UptimeSec":504,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Bluelan","BSSId":"****","Channel":1,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:04"}}}
19:29:31 MQT: stat/sonoff/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":"300:500:700:900:1000","Mode":"0"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
19:31:33 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}
19:31:33 MQT: stat/sonoff/RESULT = {"ShutterPosition1":0}
19:31:40 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
19:31:40 MQT: stat/sonoff/POWER1 = ON
19:31:41 MQT: stat/sonoff/RESULT = {"ShutterPosition1":100}
19:31:41 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":0,"Direction":1,"Target":100}}
19:31:41 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":4,"Direction":1,"Target":100}}
19:31:42 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":14,"Direction":1,"Target":100}}
19:31:43 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":24,"Direction":1,"Target":100}}
19:31:44 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":34,"Direction":1,"Target":100}}
19:31:45 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":45,"Direction":1,"Target":100}}
19:31:46 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":54,"Direction":1,"Target":100}}
19:31:47 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":64,"Direction":1,"Target":100}}
19:31:48 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":74,"Direction":1,"Target":100}}
19:31:49 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":84,"Direction":1,"Target":100}}
19:31:50 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":94,"Direction":1,"Target":100}}
19:31:51 MQT: domoticz/in = {"idx":165,"nvalue":0,"svalue":"","Battery":100,"RSSI":10}
19:31:51 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
19:31:51 MQT: stat/sonoff/POWER1 = OFF
19:31:51 MQT: stat/sonoff/SHUTTER1 = 100
19:31:51 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100}}
19:31:51 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
19:31:51 MQT: stat/sonoff/POWER2 = ON
19:31:51 MQT: stat/sonoff/RESULT = {"ShutterPosition1":0}
19:31:51 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
19:31:52 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":93,"Direction":-1,"Target":0}}
19:31:53 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":83,"Direction":-1,"Target":0}}
19:31:54 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":73,"Direction":-1,"Target":0}}
19:31:55 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":63,"Direction":-1,"Target":0}}
19:31:56 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":53,"Direction":-1,"Target":0}}
19:31:57 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":43,"Direction":-1,"Target":0}}
19:31:58 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":33,"Direction":-1,"Target":0}}
19:31:59 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":23,"Direction":-1,"Target":0}}
19:32:00 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":13,"Direction":-1,"Target":0}}
19:32:01 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":3,"Direction":-1,"Target":0}}
19:32:02 MQT: domoticz/in = {"idx":165,"nvalue":0,"svalue":"","Battery":100,"RSSI":10}
19:32:02 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
19:32:02 MQT: stat/sonoff/POWER2 = OFF
19:32:02 MQT: stat/sonoff/SHUTTER1 = 0
19:32:02 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}
19:32:02 MQT: stat/sonoff/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}
19:32:02 MQT: stat/sonoff/RESULT = {"ShutterPosition1":0}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
setoption 80 1
Interlock 1,2
Interlock ON
restart

Connect to domoticz and send open command, the shutter opens but automatically closes again,

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
When the open command is given, the shutter opens and doesn't close automatically

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-10-07:
Who can help with Domoticz? This is a special case of this HA. Please provide a log file. I assume Domoticz close the shutter

stefanbode, 2020-10-08:
The problem is that DOMOTICZ sends nvalue 0. This is the command to set the position to 0 == close. you have to configure it differently. there are some examples in the documentation for the different versions. This is nothing the driver can solve for you.

19:31:51 MQT: domoticz/in = {"idx":165,"nvalue":0,"svalue":"","Battery":100,"RSSI":10}

stefanbode, 2020-10-08:
@ahoiahoi : Can you help as a domoticz expert?

Tinus78, 2020-10-08:
@stefanbode 
I didn't find the domoticz configuration examples.
What i found so far is the following sequence:

8-10-2020 18:23:47node: df8dbc37.b85cc
domoticz/out : msg.payload : Object
object
Battery: 255
RSSI: 12
description: ""
dtype: "Light/Switch"
id: "000140F5"
idx: 165
name: "Shutter"
nvalue: 1
stype: "Switch"
svalue1: "0"
switchType: "Blinds Percentage"
unit: 1

8-10-2020 18:23:58node: c8a2260c.8580e8
domoticz/in : msg.payload : Object
object
idx: 165
nvalue: 0
svalue: ""
Battery: 100
RSSI: 10

8-10-2020 18:23:58node: df8dbc37.b85cc
domoticz/out : msg.payload : Object
object
Battery: 100
RSSI: 10
description: ""
dtype: "Light/Switch"
id: "000140F5"
idx: 165
name: "Shutter"
nvalue: 0
stype: "Switch"
switchType: "Blinds Percentage"
unit: 1

The first domoticz/out is the open or close command sent from domoticz to the switch
The domoticz/in is the status report from the switch to domoticz, when the switch turns off the relay when the shutter is completely opened or closed.
Then domoticz reacts to the change and commands the shutter opens or closes again..

ahoiahoi, 2020-10-08:
yes that is a problem that the "Telemetry period" mqtt update to domoticz might colide with the prior command domoticz sent to the switch. 
it could be solved with a "moving" state in tasmota where any further instructions are discarded until the current one is processed

stefanbode, 2020-10-09:
there is a moving state in rules "shutter#moving=1" that can be used. After movement you can use "shutter#moved=1" to cleanup everything

stefanbode, 2020-10-09:
as an alternative you can user the reported "direction". If this is <>0 then the shutter is moving.

Tinus78, 2020-10-09:
@ahoiahoi I don't think it's a collission
It's the status report from the switch that triggers the response from Domoticz
When eg relay1(for closing the shutter) is deactivated an nvalue of 0 is being published.
The nvalue 0 is being used by Domoticz to activate relay2 to open the shutter.

This way the shutter will always return to open state. This can be solved when the switch responds with nvalue 17, which means 'stopped'.

My tasmota rule skills are limited ;-) I don't know if it is possible to replace the nvalue 0, with an nvalue 17

Tinus78, 2020-10-10:
Well, I don't know what caused it, but by chance I found a solution to the problem.

In the domoticz settings you have to enter the IDX of the virtual switch within domoticz. I filled it in at IDX1. Then the switch shows the strange behavior. If I only enter the IDX for IDX 2 it seems to work without any problems.

Tinus78, 2020-10-18:
Any idea how to fix this?

ascillato2, 2020-12-02:
This issue will be auto-closed because there hasn't been any activity for quite a while. Feel free to open a new one if you still experience this problem.

closed 2020-12-02 by ascillato2