arendst/Tasmota #9493: Sonoff D1 is haunted

HNJAMeindersma, 2020-10-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sonoff D1 (dimmer) flashed with Tasmota 8.5.0 is turning on and off at random. Found events in log but cannot figure out what triggers this behaviour (see Console output).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff D1
- [X] Tasmota binary firmware version number used: 8.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:48:32 CMD: Backlog Template; Module; GPIO 255
20:48:32 SRC: WebConsole from 10.0.1.100
20:48:32 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
20:48:33 SRC: Backlog
20:48:33 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:48:33 MQT: stat/tasmota_16ED32/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:48:33 SRC: Backlog
20:48:33 CMD: Group 0, Index 1, Command "MODULE", Data ""
20:48:33 MQT: stat/tasmota_16ED32/RESULT = {"Module":{"74":"Sonoff D1"}}
20:48:33 SRC: Backlog
20:48:33 CMD: Group 0, Index 1, Command "GPIO", Data "255"
20:48:33 MQT: stat/tasmota_16ED32/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
20:49:49 CMD: Backlog Rule1; Rule2; Rule3
20:49:49 SRC: WebConsole from 10.0.1.100
20:49:49 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
20:49:49 SRC: Backlog
20:49:49 CMD: Group 0, Index 1, Command "RULE", Data ""
20:49:49 MQT: stat/tasmota_16ED32/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
20:49:49 SRC: Backlog
20:49:49 CMD: Group 0, Index 2, Command "RULE", Data ""
20:49:49 MQT: stat/tasmota_16ED32/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
20:49:49 SRC: Backlog
20:49:49 CMD: Group 0, Index 3, Command "RULE", Data ""
20:49:49 MQT: stat/tasmota_16ED32/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:50:20 SRC: WebConsole from 10.0.1.100
20:50:20 CMD: Group 0, Index 1, Command "STATUS", Data "0"
20:50:20 MQT: stat/tasmota_16ED32/STATUS = {"Status":{"Module":74,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_16ED32","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:50:20 MQT: stat/tasmota_16ED32/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T06:00:07","StartupUTC":"2020-10-07T13:50:13","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2020-10-05T13:13:29","SaveCount":528,"SaveAddress":"F4000"}}
20:50:21 MQT: stat/tasmota_16ED32/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-09T11:41:02","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"343/699"}}
20:50:21 MQT: stat/tasmota_16ED32/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["HABnet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
20:50:21 MQT: stat/tasmota_16ED32/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:50:22 MQT: stat/tasmota_16ED32/STATUS5 = {"StatusNET":{"Hostname":"tasmota_16ED32-3378","IPAddress":"10.0.0.101","Gateway":"10.0.0.1","Subnetmask":"255.0.0.0","DNSServer":"10.0.0.1","Mac":"F4:CF:A2:16:ED:32","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:50:22 MQT: stat/tasmota_16ED32/STATUS6 = {"StatusMQT":{"MqttHost":"hab","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_16ED32","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:50:22 MQT: stat/tasmota_16ED32/STATUS7 = {"StatusTIM":{"UTC":"2020-10-07T19:50:22","Local":"2020-10-07T20:50:22","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:59","Sunset":"18:15"}}
20:50:23 MQT: stat/tasmota_16ED32/STATUS10 = {"StatusSNS":{"Time":"2020-10-07T20:50:23"}}
20:50:23 MQT: stat/tasmota_16ED32/STATUS11 = {"StatusSTS":{"Time":"2020-10-07T20:50:23","Uptime":"0T06:00:10","UptimeSec":21610,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"HABnet","BSSId":"E2:91:F5:B4:13:70","Channel":9,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:01:18"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
18:40:36 DMP: AA 55 01 04 00 0A 01 01 FF FF FF FF FF FF FF FF 09
18:40:36 SRC: Switch
18:40:36 CMD: Group 0, Index 1, Command "DIMMER", Data "1"
18:40:36 MQT: stat/tasmota_16ED32/RESULT = {"POWER":"ON","Dimmer":1}
18:40:36 CFG: Saved to flash at F9, Count 523, Bytes 4096

18:58:23 DMP: AA 55 01 04 00 0A 01 64 FF FF FF FF FF FF FF FF 6C
18:58:23 SRC: Switch
18:58:23 CMD: Group 0, Index 1, Command "DIMMER", Data "100"
18:58:23 MQT: stat/tasmota_16ED32/RESULT = {"POWER":"ON","Dimmer":100}
18:58:23 CFG: Saved to flash at F7, Count 525, Bytes 4096

20:35:33 DMP: AA 55 01 04 00 0A 01 01 FF FF FF FF FF FF FF FF 09
20:35:33 SRC: Switch
20:35:33 CMD: Group 0, Index 1, Command "DIMMER", Data "1"
20:35:33 MQT: stat/tasmota_16ED32/RESULT = {"POWER":"ON","Dimmer":1}
20:35:34 CFG: Saved to flash at F4, Count 528, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Random

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Only respond to HTTP and MQTT commands

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-07:
Duplicated. Known hardware issue. See https://github.com/arendst/Tasmota/issues/7598#issuecomment-687583446

HNJAMeindersma, 2020-10-07:
Could it be RF commands that go straight to the dimmer and not through the ESP?

Jason2866, 2020-10-07:
Afaik (i dont have a D1) the Rf decoding and handling is done in the extra MCU.
This MCU is sending the data to the ESP. If wrong is sent the ESP (Tasmota) cant do anything

HNJAMeindersma, 2020-10-07:
If it goes like you say (RF > MCU > ESP > Dimmer/Relay), then Tasmota should be able to interupt the RF command (which is leaking out of the sky I guess). But that's only **IF** the ghost switching is caused by RF. As I understand from #7598 I should de-solder R11 R12 R13 R14? And if not successfull also R6 R7 R8 R9.

Jason2866, 2020-10-08:
The problem is there with the original firmware too. How should Tasmota know is it a real command or is it a ghost one?  The hardware issue has to be solved. No other solution. Ask sonoff support.

closed 2020-10-09 by arendst

HNJAMeindersma, 2020-10-09:
For my use case I don't need RF at all. Would it be possible for Tasmota to ignore all RF commands (real or ghost) through software? Or is a hardware modification the only option?

HNJAMeindersma, 2020-11-20:
Removed R6 R7 R8 R9 R11 R12 R13 R14, according to [kumy](https://github.com/arendst/Tasmota/issues/7598#issuecomment-603937986) and [kugelkopf123](https://github.com/kugelkopf123) tip. 48 hours now without any ghost switching. Seems indeed to be a bad RF design.