arendst/Tasmota #9494: Dimmer 7 and below "Hardware Watchdog" bites

stefanbode, 2020-10-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
System reboot on dimmer values < 7

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_
Define D0 = PWM1
set dimmer=3

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:50:57 MQT: hm/status/ESP_AB0565/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:50:58 MQT: hm/status/ESP_AB0565/RESULT = {"Module":{"0":"Generic"}}
21:50:58 MQT: hm/status/ESP_AB0565/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"37":"PWM1"}}
2

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-07:
It looks to happen with the Arduino PWM code we were using (PR 7022). From my experience it happens with `PWMFrequency 977` but not with `PWMFrequency 223`.

Anyways, it should be solved with #9479 using Aduino PR 7231 instead.

Can you please confirm the issue is gone?

Jason2866, 2020-10-07:
Please try latest dev version. Probably the same as #9466

stefanbode, 2020-10-08:
THX. Will try this ASAP this evening. My build is roughly 2weeks old.

stefanbode, 2020-10-08:
Yes, with the latest dev version the error is away.

closed 2020-10-08 by stefanbode