arendst/Tasmota #9496: support for WT32-ETH01: esp32 wifi + ethernet lan

megamarco833, 2020-10-07:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Describe the solution you'd like**  
i know that tasmota is at beta stage for esp32, and i fint this very cheap device that can support also the LAN connection
https://www.aliexpress.com/item/4000349581141.html?spm=a2g0o.cart.0.0.79283c00NvKUI3&mp=1
it will be great if tasmota firmware can be used on this device that cost less than 6€

the advantage could be that if wifi is not good we can use lan cable, maybe it could be PoE powering?

and maybe is it possible to use the free GPIO to connect devices such as relays?
it would be great to connect for example some relay for domotic usage and having it cable by lan instead of wifi only

**Describe alternatives you've considered**  
Olimex ESP32-PoE
i see that in tasmota this device seams to be supported, but it cost more than 3 times...
my knowledge is not so deep to understand if Olimex has more feature than the proposed cheaper device

THANKS to all for support, hope that I do not write anithing wrong...

arendst, 2020-10-08:
This device does not support PoE; it lacks the hardware to provide it. It also has no nice power connection like a USB socket so you'll need to solder power connectors yourselves.

As it is supposed to be LAN8720 compatible it might already work with Tasmota ESP32.

Just give it a go.

megamarco833, 2020-10-08:
> 
> 
> This device does not support PoE; it lacks the hardware to provide it. It also has no nice power connection like a USB socket so you'll need to solder power connectors yourselves.

thanks for kindness reply. 
Yes, there is not the USB but i see that there are pins for GPIOs and GND 5V / 3v3 so i think that soldering pins are ok to flash and then to power on :-)

> As it is supposed to be LAN8720 compatible it might already work with Tasmota ESP32.
> 
> Just give it a go.

yes, correct.
i'm searching for a device that is cheap (like 5-6€) and can be flashed with tasmota and that can be connected by LAN cable with some free GPIOs to connect relays and pushbutton.
that's why i find this one, the Olimex is too expensive :(
maybe some one has some other device to suggest? :-)

Jason2866, 2020-10-08:
As arendst already wrote the board you have linked will probably work.
We have no other boards and suggest nothing we havent tried.

sfromis, 2020-10-08:
While the board is not PoE compatible, it may be possible to do a "fake PoE" by just supplying 5v over a couple of wires not needed. If you use a full cat5e cable, you should be able to run the lower speed of 100Mbps (probably a lot more than enough) over 4 wires, leaving two pairs for power/ground. Of course, careful splitting needs to be done in parallel at both ends, and better not connect such to standard ethernet devices.

techman83, 2020-10-09:
If you really want Active PoE (803.2at) then you could spring for a [wESP32](https://wesp32.com/), buy an Active PoE splitter that breaks out the power and leaves you with data/power or do as @sfromis suggested, passive PoE. Which you can either make a cable or just buy a pair of passive PoE leads, data/power in at one end and data/power out at the other. The wESP32 is pretty nice, well designed and well made. It comes at a cost, worth it IMO though.

ascillato2, 2020-10-09:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

arendst, 2020-10-22:
I have two of them and try to get tasmota running using the eth interface. As there is almost no documentation available it's a bit of trial and error to get it going.

For now eth dosn't work.

arendst, 2020-10-23:
With latest commit the WT32-ETH01 is fully supported using below template:
```
{"NAME":"WT32-ETH01","GPIO":[1,1,1,1,1,1,0,0,1,0,1,1,3840,576,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,0,1,1,0,0,1],"FLAG":0,"BASE":1}
```

To use the ethernet connection use the following defines during compile time:
```
#define USE_ETHERNET
  // Wireless-Tag WT32-ETH01
  #define ETH_TYPE          0                    // [EthType] 0 = ETH_PHY_LAN8720, 1 = ETH_PHY_TLK110, 2 = ETH_PHY_IP101
  #define ETH_ADDR          1                    // [EthAddress] 0 = PHY0 .. 31 = PHY31
  #define ETH_CLKMODE       0                    // [EthClockMode] 0 = ETH_CLOCK_GPIO0_IN, 1 = ETH_CLOCK_GPIO0_OUT, 2 = ETH_CLOCK_GPIO16_OUT, 3 = ETH_CLOCK_GPIO17_OUT
```
or use the online command ``EthType 0``, ``EthClockMode 0`` and ``EthAddress 1``

closed 2020-10-26 by ascillato2

Blackra1n5, 2020-10-28:
How have you guys been flashing the WT32-ETH01. I'm currently trying to flash through Arduino IDE and not using Tasmota. I'm sorry if this is off topic but this seems to be one of the only results with WT32-ETH01

Jason2866, 2020-10-28:
With Arduino IDE you cant flash Tasmota32.
Do as described here https://tasmota.github.io/docs/ESP32/

arendst, 2020-10-28:
As any esp8266 or esp32 without an usb connector; ground gpio0.

Also use the correct flash mapping which is default present on platformio. Time to get rid of arduino ide and upgrade to pio!

Blackra1n5, 2020-11-01:
I've been trying to flash with  grounding GPIO 0, using TXD, RXD, and CH340. But I haven't gotten any luck flashing these WT32-ETH01 boards

efirshik, 2020-12-03:
+1, dont flash, and dont read chip_ip with esptool. only debug info if connect any ground tx0 and rxd. AT commands in standart firmware dont work in txd rxd pins, i mean send work, but dont come answer. 115200 none/8/1.

vanelm, 2020-12-22:
Same problem here. Have black one (v1.2). Tried with TX0/RX0 and TXD/RXD with grounded IO0 and IO2

1danielsan, 2020-12-23:
Hello Everybody! I am trying to flash wt32-eth1 chip!
I successfully flash it using https://www.espressif.com/.../flash_download_tool_v3.8.5.zip
[image]
(using tasmota32.bin instead of webcam.bin, of course)
configured correct template on module:
{"NAME":"WT32-ETH01","GPIO":[1,1,1,1,1,1,0,0,1,0,1,1,3840,576,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,0,1,1,0,0,1],"FLAG":0,"BASE":1}

used the online command EthType 0, EthClockMode 0 and EthAddress 1, as Theo said before.
(I get  "command" : "Unknown" for every command entered..)

but when I connect the cable there is no way to understand if it is using it... (from a network scan it is connected only via wifi)

is there something missing? is it mandatory to compile instead of using precompiled development binaries? (I am new to platformio and I don't know how to insert all files needed... is there a tutorial somewhere?)

Thank you very much!

Jason2866, 2020-12-23:
@1danielsan  you have to compile a version with ethernet support. Ethernet support is in no precompiled firmware. You can try my unofficial build https://github.com/Jason2866/Tasmota-specials/raw/firmware/firmware/tasmota32/other/tasmota32-ethernet.bin

mihai-lazar, 2020-12-24:
Sorry to ask this here, but does anyone know the pins used by the ethernet module (I can't find the info)? So I can use it with the ETH.h library for the esp32 in Arduino IDE.

efirshik, 2020-12-24:
@Jason2866 trying your tasmota-ethernet.bin with template for wt32-eth01 and commands for up ethernet. wifi work, but ethernet dont up :(

1danielsan, 2020-12-24:
as found here:
https://community.home-assistant.io/t/esphome-ethernet-connectivity-feature-request/110390/64

should it be:

Settings for WT32-ETH01

ethernet:
type: LAN8720
mdc_pin: GPIO23
mdio_pin: GPIO18
clk_mode: GPIO0_IN
phy_addr: 1
power_pin: GPIO16

efirshik, 2020-12-24:
sorry, somebody have dump original firmware for this board?

19leva89, 2021-01-07:
Good afternoon, tell me, can I use WT32-ETH01 to receive a signal via WI-FI and transmit it to the network via LAN? Make a repeater out of it.

angus51, 2021-01-09:
> Good afternoon, tell me, can I use WT32-ETH01 to receive a signal via WI-FI and transmit it to the network via LAN? Make a repeater out of it.
> [image]
> Something like that.

It can be possible, but you will need to develop a specific firmware, and will have a bad throughtput. Find an orange pi zero, a little more expensive, but ready to use.

19leva89, 2021-01-09:
> It can be possible, but you will need to develop a specific firmware, and will have a bad throughtput. Find an orange pi zero, a little more expensive, but ready to use.

The 5 Mbps bandwidth is fine for me. And the dimensions of the board are very important to me, they should be minimal, orange pi zero is large for me!

Jason2866, 2021-01-09:
No, not with Tasmota. There are no plans to implement.

madmicio, 2021-05-07:
hi guys, i tried to follow the post but i couldn't get the ethernet part to work.
I compiled the firmware as described. but the ethernet interface did not start, in the end I tried this configuration, the led started flashing, but still no connection. tips?
[image]

Mitko939, 2021-10-16:
Hi,
I have a problem to flash wt32-eth01 (v1.2) with tasmota. I tryed the tasmota web installer on Chrome, but i haven't find a tasmota32 version. I try with a simple tasmota version and for one time it works. I had rebooted the wt-32 but i can't find it like AP with wifi.
I conncet from the FT232TL to W32-eth01
GND-->GND
3.3v-->3.3v
TXD-->TX0
RXD-->TX0
and only on the power on of the device the w32-eth01 GND to IO0.
I tryed much time to flash with tasmota, but now it's impossible to put the tasmota on wt32-eth01. Some advice?
What can i do? Thank you.

szuler1979, 2021-11-09:
I have a little problem with Tasmota and Ethernet (in summary – Ethernet doesn’t work without wifi connection):
- I have successfully flashed WT32-ETH01 using FT232RL FTDU module, ESP_Flasher and tasmota32-ethernet.bin (from https://github.com/tasmota/install/raw/main/firmware/unofficial/tasmota32-ethernet.bin)  
- after flashing I have connected to WT32-ETH01 via wifi, set my wifi user and password,
- at that point I could connect to WT32-ETH01 via my network (through wifi but ethernet didn’t work)
- I have copy and pasted template {"NAME":"WT32-ETH01","GPIO":[1,1,1,1,1,1,0,0,1,0,1,1,3840,576,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,0,1,1,0,0,1],"FLAG":0,"BASE":1}
- after restart I have typed in console EthType 0, EthClockMode 0 and EthAddress 1
- from this point Ethernet started to work – I have internal ip address 192.168.0.161 for wifi connection and for Ethernet 192.168.0.162
- pinging Ethernet address showed that this is really cable connection (3ms rather than 100ms on wifi connection - also green led is flashing next to Ethernet port)
-  I wanted to have only Ethernet connection due to my wifi problems (that was the reason I chose  esp32)
- BUT when I turn wifi off (wifi 0 in console or changing wifi configuration to non existing wifi), I also loose Ethernet connection. 

Is it normal? Why I can't have Ethernet connection without wifi connection? Then what is the point having Ethernet connection?

What I’m doing wrong?

barbudor, 2021-11-09:
Turn Wifi off : https://tasmota.github.io/docs/Commands/#wifi

szuler1979, 2021-11-09:
As I wrote before - I turned wifi off (command "wifi 0" in console) but after that I lost also my ethernet connection

sfromis, 2021-11-09:
My WT32-ETH01 board with tasmota32-ethernet.bin works fine without wifi connected. In fact, I have a small Berry script `networkflipper.be` running after boot to detect if both Wifi and Ethernet are connected, and in that case turn off Wifi.
```
tasmota.set_timer(30000,
  def()
    import string
    var eth = string.find(tasmota.cmd('Status 5')['StatusNET']['Hostname'], '_eth') > 0
    if tasmota.cmd('Wifi')['Wifi'] == 'ON' == eth
      tasmota.cmd('Wifi ' .. (eth ? 0 : 1))
    end
  end)
```
I did not do anything to make it work without Wifi, and am using the same template and Ethernet settings as you posted. 

Are you aware that the host name changes with Ethernet? It is the Wifi-related host name suffixed with `_eth`

isarrider, 2022-06-12:
> As I wrote before - I turned wifi off (command "wifi 0" in console) but after that I lost also my ethernet connection

did you get it working in "airplane mode"?

szuler1979, 2022-06-12:
To be honest I gave up and switched to ESP Home - work like a charm..

paulb-opusvl, 2022-08-08:
> @1danielsan you have to compile a version with ethernet support. Ethernet support is in no precompiled firmware. You can try my unofficial build https://github.com/Jason2866/Tasmota-specials/raw/firmware/firmware/tasmota32/other/tasmota32-ethernet.bin Edit: use ONLY the additional ESP32 files from here https://github.com/arendst/Tasmota/tree/development/tools/Esptool/ESP32

Just because others may search and come across this. The links are now 404, but I found the files here: https://github.com/arendst/Tasmota-firmware/tree/main/static

Jason2866, 2022-08-08:
@paulb-opusvl  Ethernet support is default since a while in Tasmota for ESP32.
There is no need for searching and using the additional needed files for ESP32.
All needed is in factory firmware. Easiest flashing is via Tasmota WebInstaller.
Please read always the docu. Things changes fast in Tasmota.