arendst/Tasmota #9499: Missing decimal point in topic payload

Joseraj, 2020-10-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-10-09:
Hi,

Have you tried command `tempres`?

Joseraj, 2020-10-09:
Hi,
Ho I haven't, but if I understood the docs that command sets the resolution as digits after the decimal point, but not the position of the decimalpoint.
I didn't change anything in last week, just done OTA update.
Probably configuration is right, as it is showing right on the UI.

Anyway thank for evidence that command, I think resolution higher than 0.1 is useless for temperature or humidity.

ascillato, 2020-10-10:
Ok, but please, try command `tempres 1` in the Tasmota console.

Joseraj, 2020-10-10:
Just tryed, 
```06:17:03 ZIG: {"ZbEZSPReceived":"45000004010000010100010000017CBBB0F2FFFF2C187C0A01FF42250121090B0421A8430521B800062400000000006429B8056521B920662BFC8A01000A210000"}
06:17:03 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":47,"securityuse":0,"seqnumber":1,"fc":"0x18","manuf":"0x0000","transact":124,"cmdid":"0x0A","payload":"01FF42250121090B0421A8430521B800062400000000006429B8056521B920662BFC8A01000A210000"}}
06:17:03 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0000/FF01":"0121090B0421A8430521B800062400000000006429B8056521B920662BFC8A01000A210000","Endpoint":1,"LinkQuality":47}}
06:17:03 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbReceived":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","0000/FF01":"0121090B0421A8430521B800062400000000006429B8056521B920662BFC8A01000A210000","BatteryVoltage":28.25,"BatteryPercentage":62,"Temperature":1464,"Humidity":8377,"Pressure":1011,"SeaPressure":1011,"Endpoint":1,"LinkQuality":47}}}
```
Nothing changed, and dont see any change in resolution.

It seems that the sensor sends an ingegrer "1464" that should be divided by 100.
It is divided by 100 at some point because the ui shows "14.6", but not before publishing on mqtt.
[image]

Now I've seen that "BatteryVoltage" has the decimal point, but in wrong place.
```BatteryVoltage":28.25``` should be ```BatteryVoltage":2.825```

s-hadinger, 2020-10-11:
Sorry I missed this issue. Can you please paste the result of the command `ZbStatus2 0xF2B0`

I want to check ModelId and Manufacturer.

Joseraj, 2020-10-11:
Xiaomi WSDCGQ11LM
```
12:19:03` CMD: ZbStatus2 0xF2B0
12:19:03 SRC: WebConsole from 192.168.2.22
12:19:03 CMD: Group 0, Index 2, Command "ZBSTATUS", Data "0xF2B0"
12:19:03 MQT: stat/ZbBridge/RESULT = `{"ZbStatus2":[{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","IEEEAddr":"0x00158D0002786760","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1]}]}
```

s-hadinger, 2020-10-11:
Bug introduced in #9491, fixed with #9515

PR #9515 (2020-10-11): Fix multiplier not applied #9499
## Description:

With the refactoring, I forgot to re-apply the mutliplier to the MQTT message. Now solved.

**Related issue (if applicable):** fixes #9499 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.3
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#cont [... cut]
closed 2020-10-11 by s-hadinger