arendst/Tasmota #9503: Zigbee2Tasmota support for Aqara Opple Switches

Riza-Aslan, 2020-10-09:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I was able to pair Aqara Opple Switch to z2t but the batterypercentage and long-push, double-push actions are not recognized.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like to have the Aqara Opple switches to be supported like on zigbee2mqtt.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._
[https://zigbee.blakadder.com/Xiaomi_WXCJKG11LM.html]

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-09:
Can you please provide `Weblog 3` logs for each actions: long-push and double-push. Which Tasmota version are you using?

Riza-Aslan, 2020-10-09:
The first one is the left button and the second one the right button.

Single:
```
18:18:42 ZIG: {"ZbEZSPReceived":"4500000401060001010001000026FFDE4DEAFFFF0311DD0002"}
18:18:42 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xEA4D","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":139,"securityuse":0,"seqnumber":38,"fc":"0x11","manuf":"0x0000","transact":221,"cmdid":"0x00","payload":""}}
18:18:42 ZIG: ZbZCLRawReceived: {"0xEA4D":{"0006!00":"","Power":0,"Endpoint":1,"LinkQuality":139}}
18:18:42 MQT: tele/FlurEG_Switch/SENSOR = {"ZbReceived":{"FlurEG_Switch":{"Device":"0xEA4D","Name":"FlurEG_Switch","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":139}}}
18:18:48 ZIG: {"ZbEZSPReceived":"4500000401060001010001000027FFE14DEAFFFF0311DE0102"}
18:18:48 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xEA4D","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":147,"securityuse":0,"seqnumber":39,"fc":"0x11","manuf":"0x0000","transact":222,"cmdid":"0x01","payload":""}}
18:18:48 ZIG: ZbZCLRawReceived: {"0xEA4D":{"0006!01":"","Power":1,"Endpoint":1,"LinkQuality":147}}
18:18:48 MQT: tele/FlurEG_Switch/SENSOR = {"ZbReceived":{"FlurEG_Switch":{"Device":"0xEA4D","Name":"FlurEG_Switch","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":147}}}
```

Long push:
```
18:19:36 ZIG: {"ZbEZSPReceived":"4500000401000301010001000029F4D94DEAFFFF0E11E04C014500070000000000000002"}
18:19:36 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":768,"srcaddr":"0xEA4D","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":126,"securityuse":0,"seqnumber":41,"fc":"0x11","manuf":"0x0000","transact":224,"cmdid":"0x4C","payload":"0145000700000000000000"}}
18:19:36 ZIG: ZbZCLRawReceived: {"0xEA4D":{"0300!4C":"0145000700000000000000","ColorTempStepUp":69,"Endpoint":1,"LinkQuality":126}}
18:19:36 MQT: tele/FlurEG_Switch/SENSOR = {"ZbReceived":{"FlurEG_Switch":{"Device":"0xEA4D","Name":"FlurEG_Switch","0300!4C":"0145000700000000000000","ColorTempStepUp":69,"Endpoint":1,"LinkQuality":126}}}
18:19:41 ZIG: {"ZbEZSPReceived":"450000040100030101000100002AF8DA4DEAFFFF0E11E14C034500070000000000000002"}
18:19:41 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":768,"srcaddr":"0xEA4D","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":129,"securityuse":0,"seqnumber":42,"fc":"0x11","manuf":"0x0000","transact":225,"cmdid":"0x4C","payload":"0345000700000000000000"}}
18:19:41 ZIG: ZbZCLRawReceived: {"0xEA4D":{"0300!4C":"0345000700000000000000","ColorTempStepDown":69,"Endpoint":1,"LinkQuality":129}}
18:19:41 MQT: tele/FlurEG_Switch/SENSOR = {"ZbReceived":{"FlurEG_Switch":{"Device":"0xEA4D","Name":"FlurEG_Switch","0300!4C":"0345000700000000000000","ColorTempStepDown":69,"Endpoint":1,"LinkQuality":129}}}
```

Double:
```
18:20:34 ZIG: {"ZbEZSPReceived":"450000040108000101000100002DE8D64DEAFFFF0911E40201550700000002"}
18:20:34 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":8,"srcaddr":"0xEA4D","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":118,"securityuse":0,"seqnumber":45,"fc":"0x11","manuf":"0x0000","transact":228,"cmdid":"0x02","payload":"015507000000"}}
18:20:34 ZIG: ZbZCLRawReceived: {"0xEA4D":{"0008!02":"015507000000","DimmerStepDown":85,"Endpoint":1,"LinkQuality":118}}
18:20:34 MQT: tele/FlurEG_Switch/SENSOR = {"ZbReceived":{"FlurEG_Switch":{"Device":"0xEA4D","Name":"FlurEG_Switch","0008!02":"015507000000","DimmerStepDown":85,"Endpoint":1,"LinkQuality":118}}}
18:20:38 ZIG: {"ZbEZSPReceived":"450000040108000101000100002EFFE04DEAFFFF0911E50200550700000002"}
18:20:38 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":8,"srcaddr":"0xEA4D","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":145,"securityuse":0,"seqnumber":46,"fc":"0x11","manuf":"0x0000","transact":229,"cmdid":"0x02","payload":"005507000000"}}
18:20:38 ZIG: ZbZCLRawReceived: {"0xEA4D":{"0008!02":"005507000000","DimmerStepUp":85,"Endpoint":1,"LinkQuality":145}}
18:20:38 MQT: tele/FlurEG_Switch/SENSOR = {"ZbReceived":{"FlurEG_Switch":{"Device":"0xEA4D","Name":"FlurEG_Switch","0008!02":"005507000000","DimmerStepUp":85,"Endpoint":1,"LinkQuality":145}}}
```

s-hadinger, 2020-10-09:
Thanks. It actually already works:

Single: `"Power":0` and `"Power":1`
Long: `"ColorTempStepUp":69` and `"ColorTempStepDown":69`
Double: `"DimmerStepDown":85` and `"DimmerStepUp":85`

It's also described here: https://www.zigbee2mqtt.io/devices/WXCJKG11LM.html

- left click: turn off
- right click: turn on
- left double click: light dim down (by steps of 33%)
- right double click: light dim up (by steps of 33%)
- long left click: warm white
- long right click: cold white

Jason2866, 2020-10-11:
Do you any further questions? It is already implemented.

ascillato2, 2020-10-11:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-10-11 by ascillato2

Riza-Aslan, 2020-10-12:
The BatteryPercentage is still unsupported and would it be possible to get all the different "push methods" into one action? Otherwise I would have to make it more complex in my openHAB rules. z2m for example has only the "action" which shows if the button has been pushed short, long, diuble or whatever.

[zigbee2mqtt supported devices - Aqara Opple Switch](https://www.zigbee2mqtt.io/devices/WXCJKG11LM.html)

s-hadinger, 2020-10-14:
ESP8266 has limited memory resources and it is not possible to embed specific behaviors for each device. You can achieve the same with Rules, catch the events above and send whatever event you like 'short/double/long'.

For battery level, it should be bounded automatically at pairing time. Can you share the pairing time logs?