arendst/Tasmota #9506: Set USER_TEMPLATE doesn't work anymore.

kugelkopf123, 2020-10-09:
### PROBLEM DESCRIPTION
Setting a user defined template while while compiling doesn't work anymore.
Perhaps the same with the Fallback topic.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Shelly2.5
- [ ] Tasmota binary firmware version number used: 9.0.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PIO
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
22:53:36 CMD: Backlog Template; Module; GPIO 255
22:53:36 RSL: RESULT = {"NAME":"Shelly 2.5BT","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
22:53:36 RSL: RESULT = {"Module":{"0":"Shelly 2.5BT"}}
22:53:36 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:54:04 CMD: status 0
22:54:04 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Shelly25_6","FriendlyName":["Shelly25_6"],"Topic":"tasmota_05A8E9","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:54:04 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:56","StartupUTC":"2020-10-09T20:49:08","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-10-09T22:44:05","SaveCount":10,"SaveAddress":"1FA000"}}
22:54:04 RSL: STATUS2 = {"StatusFWR":{"Version":"9.0.0.1(tasmota)","BuildDateTime":"2020.10.09 22:48:26","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"391/699"}}
22:54:04 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Eiswuerstchen",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C001","2805C8000100060000005A00000000000000","00000008","00006000","00000000"]}}
22:54:04 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":474,"Free":1552,"Heap":28,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8F800586","04000001","00000087","010000C0","00000100","00000000","00000000"],"Drivers":"1,2,3,4,8,9,10,27","Sensors":"1,2,3,6"}}
22:54:04 RSL: STATUS5 = {"StatusNET":{"Hostname":"Shelly25-6-Buero","IPAddress":"192.168.1.208","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"8C:AA:B5:05:A8:E9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:54:04 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-10-09T20:54:04","Local":"2020-10-09T22:54:04","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:46","Sunset":"18:49"}}
22:54:04 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-10-09T22:54:04"}}
22:54:04 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-10-09T22:54:04","Uptime":"0T00:04:56","UptimeSec":296,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Eiswuerstchen","BSSId":"E0:28:6D:22:3F:BC","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE

Add following Lines to the User_config_override.h

```
#define USER_TEMPLATE "{\"NAME\":\"Shelly 2.5BT\",\"GPIO\":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],\"FLAG\":0,\"BASE\":18}"
#define MODULE USER_MODULE
```

Also tried the old Values:

```
#define USER_TEMPLATE "{\"NAME\":\"Shelly 2.5 (buttons)\",\"GPIO\":[56,0,19,0,21,127,0,0,6,126,5,22,156],\"FLAG\":2,\"BASE\":18}"
#define MODULE USER_MODULE
```

### EXPECTED BEHAVIOUR
After Flashing and "reset 1" the Shelly should use the right template instantly.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]
[image]

### ADDITIONAL CONTEXT
I think it has to do with the new GPIO Update.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-11:
@kugelkopf123  have you tried? Please close when the issue is solved.

kugelkopf123, 2020-10-11:
@arendst @Jason2866 
I can confirm! Now it works perfect with both. The old and the new GPIO Templates. Great Work! Thanks!

closed 2020-10-11 by kugelkopf123

BBBits, 2020-12-19:
I Get a crash on ESP32  when I configure USER_TEMPLATE.
I have traced the problem and it starts here .
#ifdef USER_TEMPLATE
  String baz = USER_TEMPLATE;
 // JsonTemplate((char *)USER_TEMPLATE);
  JsonTemplate((char *)baz.c_str());//
  /*
   there is probably another way but without this esp32 will crash @ 
   JsonParser::postProcess   _json[tok.start + tok.len] = 0;  // was perhaps writing to a flash address .
*/
#endif

arendst, 2020-12-20:
Thx. JSMN tries to use the text buffer for indexing JSON values but as there is no text buffer (direct string) it fails.

I'll implement your suggestion.

BBBits, 2020-12-20:
I ended up putting it in JsonTemplate . figured it was safer there

ramanraja, 2021-02-22:
After compiling with  #define USER_TEMPLATE , I flashed it. But the template was still not activated. 
I had to open the web UI of the device and check the box  'activate' under Template. 
Is there a way to avoid this manual step - I want the template to be activated automatically during compile time.

techman83, 2021-02-23:
@ramanraja - this is a closed issue. You would need to reset the device, as technically a template was selected already, the firmware is what is selected when the device is blank.

I think a note on this [wiki page](https://tasmota.github.io/docs/Compile-your-build/) covering templates is probably warranted, as it has come up a few times. I'll see if I can come up with something.

ramanraja, 2021-02-23:
@techman83, thank you very much.
Indeed, a simple "module 0" command from the web console solved the problem for me. 
One small comment though, about my situation:
I am sending out the devices to non-technical clients, who would like to use it out of the box.  I would like to:
 - Edit user_config_override.h and enter the JSON template for my hardware.
 - Compile and burn the binary into the device, and ship it to my client.
 - The client downloads my mobile app that controls the device. The app will configure WiFi on Tasmota, without any user input.
 - The client powers up the device,  and it just starts working.  All the GPIOs work as per my template.
 Right now the template starts life in a disabled state, so I was wondering if it can be pre-enabled by a compile-time flag.
 Regards & thanks,
RamanRaja

ramanraja, 2021-04-02:
A follow up on the above: I had the same observation about enabling timers also. I have posted the following question on the Telegram channel:
I am compiling  a custom version of Tasmota from source for my custom hardware, derived from Generic(18).  I am using the latest version  9.3.1.1
But two configuration items are not working in the compiled binary: Enabling timers and activating the template.  
After I have burned the image onto the device, I have to connect to the device's portal and issue the command
**backlog timers 1; module 0**
Is there any way to avoid this step ? I just want to enable timers and set the module at compile time itself. (Resetting the device or even power cycling does not solve the issue).

Jason2866, 2021-04-02:
@ramanraja  posting on a closed issue is not recommended.
If you think you have found a issue open a NEW issue.

horris87, 2022-05-10:
#ifdef USER_BACKLOG
#undef USER_BACKLOG
#endif
#define USER_BACKLOG "<Any command separated by a semicolon (;)>"  // Add commands executed at firmware load or when command reset is executed
runs commands at first boot