arendst/Tasmota #9509: Gosund SP112 power factor always 1.00

David91919, 2020-10-10:
### PROBLEM DESCRIPTION
Gosund SP112 reports incorrect power factor of exactly 1.00. The wattage value seems to be inversely multiplied by the real power factor (checked against SEM 600 meter from Conrad).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
 stat/tasmota/RESULT = {"NAME":"SP112new","GPIO":[57,0,56,0,132,134,0,0,131,30,21,0,0],"FLAG":4,"BASE":18}
08:45:08 RSL: stat/tasmota/RESULT = {"Module":{"0":"SP112new"}}
08:45:09 RSL: stat/tasmota/RESULT = {"GPIO0":{"57":"Led2i"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"132":"HLWBL CF1"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"30":"Relay2i"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:45:33 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"DeviceName":"PowerFridge","FriendlyName":["PowerFridge","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:45:33 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T12:27:41","StartupUTC":"2020-10-09T19:17:52","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2020-10-06T17:39:11","SaveCount":120,"SaveAddress":"FB000"}}
08:45:33 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"397/699"}}
08:45:33 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Robi438",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00006000","00000000"]}}
08:45:33 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:45:33 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-1689","IPAddress":"192.168.178.46","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"50:02:91:2A:66:99","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
08:45:33 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.31","MqttPort":1883,"MqttClientMask":"PowerFridge","MqttClient":"PowerFridge","MqttUser":"Tasmota","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:45:33 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-10-10T07:45:33","Local":"2020-10-10T08:45:33","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:03","Sunset":"18:09"}}
08:45:33 RSL: stat/tasmota/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:45:33 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-10-10T08:45:33","ENERGY":{"TotalStartTime":"2020-10-06T17:33:09","Total":8.594,"Yesterday":1.919,"Today":0.731,"Power":74,"ApparentPower":74,"ReactivePower":0,"Factor":1.00,"Voltage":233,"Current":0.317}}}
08:45:33 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-10-10T08:45:33","Uptime":"0T12:27:41","UptimeSec":44861,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Robi438","BSSId":"2C:91:AB:85:9B:B2","Channel":6,"RSSI":62,"Signal":-69,"LinkCount":2,"Downtime":"0T00:00:10"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
08:46:35 CFG: Saved to flash at FA, Count 121, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Flash Tasmota using raspberrypi. Then update to latest version using firmware upgrade function.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Correct power factor to be shown

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

It seems on very low power consumtion <10w a power factor other than 1.00 is shown

**(Please, remember to close the issue when the problem has been addressed)**

David91919, 2020-10-10:
[image]
[image]

David91919, 2020-10-10:
[image]
[image]

sfromis, 2020-10-10:
To get good data, you should first calibrate the metering using an resistive load.
https://tasmota.github.io/docs/Power-Monitoring-Calibration

David91919, 2020-10-10:
It is calibrated against 46 w resistive load as in the manual

David91919, 2020-10-10:
updated to 9.0.0.1 but no change to observed behavior

Jason2866, 2020-10-10:
46W is a too low load for a good calibration. Minimum is 60W. Better 100W. 
If you have a heat radiator with 1000-2000W it would be good.

David91919, 2020-10-10:
Calibrated it to a 1400 W resistive heater and now it works! Thanks and have a nice weekend
David

closed 2020-10-10 by David91919