arendst/Tasmota #9510: Zigbee2Tasmota Support for Moes HY386-ZB TRV

coldfire84, 2020-10-10:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I've been able to pair the [Moes HY386-ZB TRV](https://www.moeshouse.com/collections/zigbee-thermostat/products/tuya-zigbee3-0-new-mini-wifi-radiator-actuator-smart-programmable-thermostat-temperature-trv-controller) with a Sonoff Zigbee Bridge running Tasmota 9.0.0.1, using the same approach as outlined here for the 'Tuya TS0601_thermostat' device: https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

_Note I have included pairing output under additional context._

It would appear that payloads are not being translated - output is from adjusting the setpoint on the TRV itself a couple of times.

```
13:59:30 ZIG: {"ZbEZSPReceived":"450000040100EF0101000100008C5CB3BD6FFFFF0D094E01006A02020004000000E602"}
13:59:30 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":26,"securityuse":0,"seqnumber":140,"fc":"0x09","manuf":"0x0000","transact":78,"cmdid":"0x01","payload":"006A02020004000000E6"}}
13:59:30 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00<01":"006A02020004000000E6","Endpoint":1,"LinkQuality":26}}
13:59:30 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00<01":"006A02020004000000E6","Endpoint":1,"LinkQuality":26}}}
13:59:40 ZIG: {"ZbEZSPReceived":"450000040100EF0101000100008D58B2BD6FFFFF0D094F01006B03020004000000F002"}
13:59:40 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":24,"securityuse":0,"seqnumber":141,"fc":"0x09","manuf":"0x0000","transact":79,"cmdid":"0x01","payload":"006B03020004000000F0"}}
13:59:40 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00<01":"006B03020004000000F0","Endpoint":1,"LinkQuality":24}}
13:59:40 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00<01":"006B03020004000000F0","Endpoint":1,"LinkQuality":24}}}
13:59:51 ZIG: {"ZbEZSPReceived":"450000040100EF0101000100008E5CB3BD6FFFFF0D095001006C03020004000000F002"}
13:59:51 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":26,"securityuse":0,"seqnumber":142,"fc":"0x09","manuf":"0x0000","transact":80,"cmdid":"0x01","payload":"006C03020004000000F0"}}
13:59:51 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00<01":"006C03020004000000F0","Endpoint":1,"LinkQuality":26}}
13:59:51 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00<01":"006C03020004000000F0","Endpoint":1,"LinkQuality":26}}}
```

**Describe the solution you'd like**  
I'd like to be able to control the TRV thermostat setpoint, and also ensure I'm able to read TRV temperature sensor states for triggering actions such as calling for heating to be switched on etc.

**Describe alternatives you've considered**  
None.

**Additional context**  
Pairing console output:
```
13:52:59 CMD: zbpermitjoin 1
13:52:59 SRC: WebConsole from _ip address_
13:52:59 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
13:52:59 ZIG: ZbEZSPSend 22003C
13:52:59 ZIG: ZbEZSPSend 3600FCFF00003600000000000000021E0103023C01
13:52:59 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
13:52:59 MQT: stat/zbbridge-1/RESULT = {"ZbPermitJoin":"Done"}
13:52:59 ZIG: {"ZbEZSPReceived":"220000"}
13:52:59 ZIG: NAK, resending packet 0
13:52:59 ZIG: {"ZbEZSPReceived":"3600009E"}
13:52:59 ZIG: {"ZbEZSPReceived":"450005000036000000000000009EFF000000FFFF03023C01"}
13:52:59 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
13:53:00 ZIG: {"ZbEZSPReceived":"3F0006FCFF000036000000000000009E010000"}
13:53:08 ZIG: {"ZbEZSPReceived":"23000101BD6FC120ABFEFF23A46004"}
13:53:08 ZIG: {"ZbEZSPReceived":"2400BD6FC120ABFEFF23A46001000000"}
13:53:08 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x60A423FFFEAB20C1","ShortAddr":"0x6FBD","ParentNetwork":"0x0000","Status":1,"Decision":0}}
13:53:09 ZIG: {"ZbEZSPReceived":"6200C120ABFEFF23A460"}
13:53:09 ZIG: {"ZbEZSPReceived":"450004000013000000000400006264B5BD6FFFFF0C83BD6FC120ABFEFF23A4608002"}
13:53:09 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x60A423FFFEAB20C1","ShortAddr":"0x6FBD","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
13:53:09 ZIG: ZbEZSPSend 340000BD6F0000050000004001000003010303BD6F
13:53:09 ZIG: {"ZbEZSPReceived":"340000A1"}
13:53:09 ZIG: {"ZbEZSPReceived":"450000000005800000400100006364B5BD6FFFFF060300BD6F010102"}
13:53:09 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
13:53:09 ZIG: ZbEZSPSend 340000BD6F0401000001014001000001010700010004000500
13:53:09 ZIG: {"ZbEZSPReceived":"3F0000BD6F00000500000040010000A1010000"}
13:53:09 ZIG: {"ZbEZSPReceived":"340000A2"}
13:53:09 ZIG: {"ZbEZSPReceived":"450000040100000101000100006464B5BD6FFFFF2318010104000042105F545A453230305F636B75643775326C050000420654533036303102"}
13:53:09 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":32,"securityuse":0,"seqnumber":100,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"04000042105F545A453230305F636B75643775326C0500004206545330363031"}}
13:53:09 ZIG: ZbZCLRawReceived: {"0x6FBD":{"0000/0004":"_TZE200_ckud7u2l","0000/0005":"TS0601","Endpoint":1,"LinkQuality":32}}
13:53:09 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","Manufacturer":"_TZE200_ckud7u2l","ModelId":"TS0601","Endpoint":1,"LinkQuality":32}}}
13:53:09 ZIG: {"ZbEZSPReceived":"3F0000BD6F04010000010140010000A2010000"}
13:53:09 ZIG: {"ZbEZSPReceived":"3F0006FDFF0000130000000004000062000000"}
13:53:10 ZIG: ZbEZSPSend 340000BD6F0000040000004001000004010404BD6F01
13:53:10 ZIG: {"ZbEZSPReceived":"340000A3"}
13:53:10 ZIG: {"ZbEZSPReceived":"450000000004800000400100006568B6BD6FFFFF190400BD6F140104015100010400000400050000EF0219000A0002"}
13:53:10 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":33,"Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0051","DeviceVersion":1"InClusters":["0x0000","0x0004","0x0005","0xEF00"],"OutClusters":["0x0019","0x000A"]}}
[... 95 lines cut ...]
13:53:35 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00<01":"005E6C0200040000000F","Endpoint":1,"LinkQuality":0}}}
13:53:36 ZIG: {"ZbEZSPReceived":"450000040100EF0101000100008144ADBD6FFFFF0D094301005F6D0200040000000002"}
13:53:36 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":11,"securityuse":0,"seqnumber":129,"fc":"0x09","manuf":"0x0000","transact":67,"cmdid":"0x01","payload":"005F6D02000400000000"}}
13:53:36 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00<01":"005F6D02000400000000","Endpoint":1,"LinkQuality":11}}
13:53:36 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00<01":"005F6D02000400000000","Endpoint":1,"LinkQuality":11}}}
13:53:37 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000822CA7BD6FFFFF0A09440100606E0100010002"}
13:53:37 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":130,"fc":"0x09","manuf":"0x0000","transact":68,"cmdid":"0x01","payload":"00606E01000100"}}
13:53:37 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00<01":"00606E01000100","Endpoint":1,"LinkQuality":0}}
13:53:37 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00<01":"00606E01000100","Endpoint":1,"LinkQuality":0}}}
13:53:38 ZIG: {"ZbEZSPReceived":"450000040100EF0101000100008324A5BD6FFFFF0A09450100616F0400010002"}
13:53:38 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":131,"fc":"0x09","manuf":"0x0000","transact":69,"cmdid":"0x01","payload":"00616F04000100"}}
13:53:38 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00<01":"00616F04000100","Endpoint":1,"LinkQuality":0}}
13:53:38 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00<01":"00616F04000100","Endpoint":1,"LinkQuality":0}}}
13:53:39 ZIG: {"ZbEZSPReceived":"450000040100EF0101000100008428A6BD6FFFFF1B09460100627000001206001408000F0B1E0F0C1E0F111E1416000F02"}
[... cut ...] 0001286001408000F0B1E0F0C1E0F111E1416000F"}}
13:53:50 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00<01":"00637100001286001408000F0B1E0F0C1E0F111E1416000F","Endpoint":1,"LinkQuality":0}}
13:53:50 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00<01":"00637100001286001408000F0B1E0F0C1E0F111E1416000F","Endpoint":1,"LinkQuality":0}}}
```
Happy to provide any further information. console outputs etc.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-10:
Since ESP8266 has limited resources not every device which is supported in zigbee2mqtt can be added.
There is already one zigbee TRV supported. Eurotronic Spirit Zigbee SPZB0001
@s-hadinger can you check if this one can be easily added? Thx.

s-hadinger, 2020-10-10:
Gosh. Why did Tuya invent a new proprietary format and didn't use ZCL? Since Tuya is a major brand, it's worth trying to convert attributes.

The best documentation I've found about their format is this one: https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d

s-hadinger, 2020-10-11:
Please try the latest version. I have implemented attribute reporting and tested with the inputs above. The weekly schedule is not decoded yet.

Sending commands and changing values to the TRV is not yet implemented and will come in phase 2.

Reopening

coldfire84, 2020-10-11:
> Please try the latest version. I have implemented attribute reporting and tested with the inputs above. 

Awesome - thank you! I can confirm I'm now getting decoded sensor/ configuration information from the TRV.

[image]

Example console output showing the various information the TRV 'spews' out on reconnecting.

```
20:42:55 ZIG: {"ZbEZSPReceived":"3F0006FCFF00001F0000000000000000000000"}
20:43:00 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000C038AABD6FFFFF0A0912010AED6F04000100"}
20:43:00 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":3,"securityuse":0,"seqnumber":192,"fc":"0x09","manuf":"0x0000","transact":18,"cmdid":"0x01","payload":"0AED6F04000100"}}
20:43:00 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/046F":0,"Endpoint":1,"LinkQuality":3}}
20:43:00 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaWeekSelect":0,"Endpoint":1,"LinkQuality":3}}}
20:43:10 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000C168B6BD6FFFFF1B0913010AEE7000001206001408000F0B1E0F0C1E0F111E1416000F"}
20:43:10 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":34,"securityuse":0,"seqnumber":193,"fc":"0x09","manuf":"0x0000","transact":19,"cmdid":"0x01","payload":"0AEE7000001206001408000F0B1E0F0C1E0F111E1416000F"}}
20:43:10 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0070":"06001408000F0B1E0F0C1E0F111E1416000F","Endpoint":1,"LinkQuality":34}}
20:43:10 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaScheduleWorkdays":"06001408000F0B1E0F0C1E0F111E1416000F","Endpoint":1,"LinkQuality":34}}}
20:43:21 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000C274B9BD6FFFFF1B0914010AEF7100001286001408000F0B1E0F0C1E0F111E1416000F"}
20:43:21 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":42,"securityuse":0,"seqnumber":194,"fc":"0x09","manuf":"0x0000","transact":20,"cmdid":"0x01","payload":"0AEF7100001286001408000F0B1E0F0C1E0F111E1416000F"}}
20:43:21 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0071":"86001408000F0B1E0F0C1E0F111E1416000F","Endpoint":1,"LinkQuality":42}}
20:43:21 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaScheduleHolidays":"86001408000F0B1E0F0C1E0F111E1416000F","Endpoint":1,"LinkQuality":42}}}
20:43:31 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000C374B9BD6FFFFF0D0915010AF0720200040000000F"}
20:43:31 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":42,"securityuse":0,"seqnumber":195,"fc":"0x09","manuf":"0x0000","transact":21,"cmdid":"0x01","payload":"0AF0720200040000000F"}}
20:43:31 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0272":15,"Endpoint":1,"LinkQuality":42}}
20:43:31 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaAwayTemp":15,"Endpoint":1,"LinkQuality":42}}}
20:43:42 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000C470B8BD6FFFFF0A0916010AF17301000100"}
20:43:42 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":39,"securityuse":0,"seqnumber":196,"fc":"0x09","manuf":"0x0000","transact":22,"cmdid":"0x01","payload":"0AF17301000100"}}
20:43:42 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0173":0,"Endpoint":1,"LinkQuality":39}}
20:43:42 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","EF00/0173":0,"Endpoint":1,"LinkQuality":39}}}
20:43:52 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000C578BABD6FFFFF0A0917010AF27401000100"}
20:43:52 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":45,"securityuse":0,"seqnumber":197,"fc":"0x09","manuf":"0x0000","transact":23,"cmdid":"0x01","payload":"0AF27401000100"}}
20:43:52 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0174":0,"Endpoint":1,"LinkQuality":45}}
20:43:52 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaAutoLock":0,"Endpoint":1,"LinkQuality":45}}}
20:44:03 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000C664B5BD6FFFFF0D0918010AF37502000400000001"}
20:44:03 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":32,"securityuse":0,"seqnumber":198,"fc":"0x09","manuf":"0x0000","transact":24,"cmdid":"0x01","payload":"0AF37502000400000001"}}
20:44:03 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0275":1,"Endpoint":1,"LinkQuality":32}}
20:44:03 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaAwayDays":1,"Endpoint":1,"LinkQuality":32}}}
20:45:04 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000C77CBBBD6FFFFF0D0919010AF402020004000000D7"}
20:45:04 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":47,"securityuse":0,"seqnumber":199,"fc":"0x09","manuf":"0x0000","transact":25,"cmdid":"0x01","payload":"0AF402020004000000D7"}}
20:45:04 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0202":215,"0201/0012":2150,"Endpoint":1,"LinkQuality":47}}
20:45:05 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaTempTarget":21.5,"OccupiedHeatingSetpoint":21.5,"Endpoint":1,"LinkQuality":47}}}
[... cut ...]
20:45:36 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000CA40ACBD6FFFFF0A091C010AF70404000102"}
20:45:36 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":8,"securityuse":0,"seqnumber":202,"fc":"0x09","manuf":"0x0000","transact":28,"cmdid":"0x01","payload":"0AF70404000102"}}
20:45:36 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0404":2,"Endpoint":1,"LinkQuality":8}}
20:45:36 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaPreset":2,"Endpoint":1,"LinkQuality":8}}}
```

s-hadinger, 2020-10-11:
Thanks, this is useful. I couldn't find what `"EF00/0173":0` means, probably something related to auto-lock. 

I'm also interested by the value of TuyaBattery when you see it. I'm not sure if it's a voltage or a percentage.

s-hadinger, 2020-10-13:
@coldfire84 Can you please try:
```
ZbSend {"Device":"0x6FBD","Endpoint":1,"Write":{"TuyaTempTarget":20.5}}
```

closed 2020-10-13 by s-hadinger

coldfire84, 2020-10-14:
@s-hadinger I have tried the command:

```
ZbSend {"Device":"0x6FBD","Endpoint":1,"Write":{"TuyaTempTarget":19.0}}
```

Console output:

```
13:11:01 CMD: ZbSend {"Device":"0x6FBD","Endpoint":1,"Write":{"TuyaTempTarget":19.0}}
13:11:01 SRC: WebConsole from _ip address_
13:11:01 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x6FBD","Endpoint":1,"Write":{"TuyaTempTarget":19.0}}"
13:11:01 ZIG: ZbEZSPSend 340000BD6F040100EF0101400100004C0108104C02020229BE00
13:11:01 MQT: stat/zbbridge-1/RESULT = {"ZbSend":"Done"}
13:11:01 ZIG: {"ZbEZSPReceived":"34000012"}
13:11:04 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000CC6CB7BD6FFFFF06184C0486020202"}
13:11:04 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":37,"securityuse":0,"seqnumber":204,"fc":"0x18","manuf":"0x0000","transact":76,"cmdid":"0x04","payload":"860202"}}
13:11:04 ZIG: ZbZCLRawReceived: {"0x6FBD":{"Endpoint":1,"LinkQuality":37}}
13:11:04 ZIG: {"ZbEZSPReceived":"3F0000BD6F040100EF01014001000012010000"}
```

The TRV itself is showing the wrong target temperature (value was 21.5, still 21.5, Tasmota GUI also still showing 21.5).

s-hadinger, 2020-10-14:
The TRV returned an error. I must have missed something. Let me get a closer look.

s-hadinger, 2020-10-14:
@coldfire84 Did you use the latest version, after #9534 ?
The packet sent looks wrong.

coldfire84, 2020-10-14:
May have updated too soon after your phase 2 commit. 

I will re-run the update and re-test/ paste results.

coldfire84, 2020-10-14:
Updated firmware looks _much_ better, apologies - more haste less speed.

```
16:00:56 CMD: ZbSend {"Device":"0x6FBD","Endpoint":1,"Write":{"TuyaTempTarget":19.0}}
16:00:56 SRC: WebConsole from _ip address_
16:00:56 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x6FBD","Endpoint":1,"Write":{"TuyaTempTarget":19.0}}"
16:00:56 ZIG: ZbEZSPSend 340000BD6F040100EF01014001000002010D110200000102020004000000BE
16:00:56 MQT: stat/zbbridge-1/RESULT = {"ZbSend":"Done"}
16:00:56 ZIG: {"ZbEZSPReceived":"34000095"}
16:00:57 ZIG: {"ZbEZSPReceived":"3F0000BD6F040100EF01014001000095010000"}
16:00:57 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000D760B4BD6FFFFF0D09530226C202020004000000BE"}
16:00:57 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x6FBD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":29,"securityuse":0,"seqnumber":215,"fc":"0x09","manuf":"0x0000","transact":83,"cmdid":"0x02","payload":"26C202020004000000BE"}}
16:00:57 ZIG: ZbZCLRawReceived: {"0x6FBD":{"EF00/0202":190,"Endpoint":1,"LinkQuality":29}}
16:00:57 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"0x6FBD":{"Device":"0x6FBD","TuyaTempTarget":19,"Endpoint":1,"LinkQuality":29}}}
```

TRV is showing the correct value, although I've lost the current, target and position info in Tasmota GUI for some reason.

[image]

galsom, 2020-11-05:
I have setup the same TRV using 9.1.0 and am going to integrate it with Home Assistant. I see different MQTT messages indicating the local temperature, the child lock etc, so it seems to be working well.

Does somebody have an example of how to do this using the Home Assistant HVAC MQTT integration? I had a look at the [Z2M example](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html), but do not know what the T2M equivalents are for e.g. `system_mode`, `away_mode`,`running_state`etc.

Draggon, 2020-11-21:
> I have setup the same TRV using 9.1.0 and am going to integrate it with Home Assistant. I see different MQTT messages indicating the local temperature, the child lock etc, so it seems to be working [...]

Hi

I'm using this one - but it just basic implementation - to read params and set temperature.
One sensor + automation to 'route' mqtt message to proper format.

```javascript
climate:
  - platform: mqtt
    name: "Thermostat M"
    availability_topic: "sonoffzb/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true
    qos: 0
    modes:
      - "auto"
      - "heat"
    temp_step: 0.5
    initial: 22
    min_temp: 15
    max_temp: 25
    json_attributes_topic: "sonoffzb/tele/5E68/SENSOR"
    json_attributes_template: >
      {% set properites = ["LocalTemperature", "TuyaTempTarget", "TuyaValvePosition", "LinkQuality", "TuyaAwayDays", "TuyaAutoLock", "TuyaAwayTemp", "TuyaScheduleHolidays", "TuyaScheduleWorkdays", "TuyaWeekSelect", "TuyaEcoTemp", "TuyaComfortTemp", "TuyaForceMode", "TuyaBoostTime", "TuyaChildLock", "TuyaPreset", "TuyaMinTemp", "TuyaMinTemp"] %}      
      {%- set json_template -%}
      {
      {%- for key in properites -%}
        "{{key}}": "{{ value_json.ZbReceived['0x5E68'][key] | default(state_attr('sensor.trv',key))}}",
      {%- endfor -%}
         "Node" : ""}
      {%- endset -%}
      {{ json_template  }}   
    current_temperature_topic: "sonoffzb/tele/5E68/SENSOR"
    current_temperature_template: >
      {% if value_json.ZbReceived['0x5E68'].LocalTemperature is defined %}
        {{ value_json.ZbReceived['0x5E68'].LocalTemperature }}
      {% else %}
        {{ state_attr('climate.thermostat_m','LocalTemperature') }}
      {% endif %}      
    temperature_state_topic: "sonoffzb/tele/5E68/SENSOR"
    temperature_state_template: >
      {% if value_json.ZbReceived['0x5E68'].TuyaTempTarget is defined %}
        {{ value_json.ZbReceived['0x5E68'].TuyaTempTarget }}
      {% else %}
        {{ state_attr('climate.thermostat_m','TuyaTempTarget') }}
      {% endif %}    
    mode_command_topic: /routing/thermostat/0x5E68/mode 
    temperature_command_topic: "routing/thermostat/0x5E68/currentsetpoint"

automation:
  - alias: tuya_temp_helper
    description: ''
    trigger:
    - platform: mqtt
      topic: routing/thermostat/+/currentsetpoint
    condition: []
    action:
      service: mqtt.publish
      data_template:
        payload: >          
          {"Device":"{{trigger.topic.split('/')[2]}}","Endpoint":1,"Write":{"TuyaTempTarget": {{trigger.payload}}}}
        topic: sonoffzb/cmnd/5E68/ZbSend`
```

Changing mode would be cool - help appreciated (ZbSend message) ;)

danielk117, 2021-02-06:
Only `TuyaPreset` und `TuyaTempTarget` are writable with Z2T. 
I’am not able to turn it off, like the `system_mode` in Z2M. `TuyaPreset` only support something like heat (2), auto (6) and boost (5). 
Setting `TuyaChildLock` would be nice. Or receiving the `battery_low` state. 
Any chance to get this things working with Z2T?

kilnageer2, 2021-07-30:
Wow! Thank-you so much @s-hadinger   for adding Tuya TRV support to ZbBridge code. I've been having grief with these TRVs for a year both in zigbee2mqtt and ZbBridge. But upgrading from 8.4.0.3(zbbridge) to 9.5.0(zbbridge) has povided full support :) Thanks again to everybody for driving, testing and coding.

PR #9516 (2020-10-11): Zigbee support for Tuya Moes devices
## Description:

Support for proprietary format of Zigbee attribute of Tuya/Moes TRV devices.

Phase 1: attribute reported by TRV. This does not allow to send commands yet.

**Related issue (if applicable):** fixes #9510

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.3
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](http [... cut]
PR #9534 (2020-10-13): Zigbee tuya phase 2
## Description:

Support for Tuya Zigbee devices: writing attributes.

Ex:
```
ZbSend {"Device":"0x6FBD","Endpoint":1,"Write":{"TuyaTempTarget":20.5}}
```

Note: due to limitations in the Tuya Zigbee protocol, you can send only one attribute per command.

**Related issue (if applicable):** fixes #9510

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.3
  - [x] The code change is tested [... cut]
PR #10074 (2020-12-06): Zigbee better support for Tuya Protocol
## Description:

Better support for  Tuya Zigbee protocol, including both receiving and writing attributes.

The Tuya Zigbee protocol is different from standard Zigbee, in that you don't send command, but instead you write to an attribute. So instead of `ZbSend {"Device":"my_device","Send":{"Power":1}}` you need instead to `ZbSend {"Device":"my_device","Write":{"Power":1}}`. I.e. use `Write` instead of `Send`, whereas in standard Zigbee the `Power` attribute in normally not writeable.

Tuya Zigbee uses cluster `0xEF00`. Since some attributes exist both i [... cut]