arendst/Tasmota #9512: Shutter compile fails if scripting is enabled

Jason2866, 2020-10-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Compilation fails when scripter and shutter is enabled
```
/home/runner/work/Merge/Merge/tasmota/xdrv_27_shutter.ino: In function 'void ShutterAllowPreStartProcedure(uint8_t)':
/home/runner/work/Merge/Merge/tasmota/xdrv_27_shutter.ino:521:91: error: 'rules_vars' was not declared in this scope
   AddLog_P2(LOG_LEVEL_DEBUG_MORE, PSTR("SHT: Delay Start. var%d <99>=<%s>, max10s?"),i+i, rules_vars[i]);
                                                                                           ^
*** [.pio/build/tasmota-scripting/src/tasmota.ino.cpp.o] Error 1
``` 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO latest
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Enable scripting and Shutter function

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
successfull compile

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-10-11:
this is not related to scripting.
this modification to shutter was meant for rules and has to be ifdefed

Jason2866, 2020-10-11:
Confirmed

closed 2020-10-11 by Jason2866