arendst/Tasmota #9518: The 1K resistor mod for PZEM-004T v3 is wrong

radudorurus, 2020-10-11:
### PROBLEM DESCRIPTION
The resistor mod
https://tasmota.github.io/docs/PZEM-0XX/#pzem-004t-version-v3_1
is wrong.
For the PZEM004T v3 the 1k resistor is soldered between 5V and GND. (mistakenly following the visual from v1)
https://user-images.githubusercontent.com/34340210/63591794-cf3d5d80-c57d-11e9-9945-eb062bebf71b.jpeg
This is because in v3 the optocouplers are swapped compared to v1.
http://samopal.pro/wp13_samopal/wp-content/uploads/2681/100.jpg

Basically, the mod on PZEM-004T v1 was to connect a 1k resistor in parallel with the existing one for the optocoupler to increase the current since it is provided with 3.3V instead of 5V.
This can be done in v3 by putting the 1K resistor in parallel with R8.
In the image it is placed between pin1 (RX opto) and VCC which is the same.

No issues was reported by now probably because The 3.3V and 1k existing R8 is enough to open the Rx opto LED.(2.06mA)
The 1k resistor in parallel will increase the current for the opto to 4.12mA. (actually being designed for VCC 5V/ ~3.76mA)

Datasheet of the opto
http://www.ct-micro.com/uploads_zip/CT817.pdf

### REQUESTED INFORMATION
Please correct the guide

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): not related
- [x] Tasmota binary firmware version number used: not related
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: not related
- [x] Flashing tools used: not related
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`: not related
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`: not related
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`: not related
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
-

### EXPECTED BEHAVIOUR
-

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-11:
Sharp eyes!  You can edit the docs and provide a PR. 
Tasmota docu is Open Source too. We would apreciate it!

radudorurus, 2020-10-12:
Thanks.
I can do the edit.
But the PR it seems difficult to create. I did not found any docs section.

Jason2866, 2020-10-12:
You find a now to here https://tasmota.github.io/docs/Contributing/

Jason2866, 2020-10-13:
Please close. Since you provided the solution :-) Thx!!

closed 2020-10-13 by radudorurus

ascillato, 2021-02-03:
The resistor is just for communications. Will not influence on the readings.

rrelande, 2023-03-16:
HI, 
appreciate such valuable post. 
Still referring to original PZEM-004T v3 with CT817C optocoupler WITHOUT 1k modifications to R8

according to calculations, the 2.06 mA forward current 
with a CTR of .5, it still delivers 1mA collector current
should indeed be enough to open the optocoupler, 
even with a power source of 3V3

This could be barely enough to open the optocoupler , but could work. 

Is there any feedback from running an unmodified PZEM-004 v3 on 3V3 ?

dg1sek, 2024-01-01:
For me adding the 1K resistor shown in the drawing in pink AND REMOVING the R8 SMD resistor worked. Without removing the SMD resistor I did not get RS232 with a NodeMCU board up and running. I used the Vin connector of the NodeMCU board as 5V power supply to the PZEM module, this seems to work fine, the USB PSU module does bring stable 5V, so I don't really see why to add an additional buck converter.

ipa64, 2025-05-04:
> For me adding the 1K resistor shown in the drawing in pink AND REMOVING the R8 SMD resistor worked. 

That means you did nothing, replacing R8 SMD 1K by another resistor 1K 🤔

Dinesh6252, 2025-05-06:
Why no modification on r4 resistor? Can someone explain please, I finally got reading on my pzem-004t v3 module but still get nan valu occasionally. I think there should be 1k resistor in parallel with r4 also.

Dinesh6252, 2025-05-07:
Then why my readings gets blank after some time of showing up properly. I have done modification of paralleling 1k resistor with r8 and using 3.3v VCC.

ipa64, 2025-05-07:
> Then why my readings gets blank after some time of showing up properly…

Sorry, I'm just passing through here, I don't know your project, what nodemcu do you use, what firmware do you have, do you have a schematic of your assembly?