arendst/Tasmota #9522: RDM6300 fake readings

dbvcode, 2020-10-12:
### PROBLEM DESCRIPTION
I have created 8 devices based on the [AZ Delivery Amica NodeMCU](https://www.az-delivery.de/en/products/nodemcu-lua-amica-v2-modul-mit-esp8266-12e-unverloetet?_pos=2&_sid=11852c248&_ss=r).
All have an RDM6300 card reader, a buzzer and a WS2812 LED. They are supposed to be card readers as their main function.
They work as expected, but **I also get fake card numbers in the readings**. 
Examples are:
000000078F88, 00F00000F000, F000060007F1, B59A99E5A9FA, 4FFB388FFFFC, DA9386B5EA90, DAEF6FEE2B9F
These readings are not actual cards, but 'fake' numbers. Strange part is they are VALID from the validation perspective and if I do an XOR taking the HEX numbers two by two([I play with this tool](https://toolslick.com/math/bitwise/xor-calculator)) i get something like:

_000000078F88_:  00 ^ 00 ^ 00 ^ 07 ^ 8F = 88

This makes the 'card' valid, but it's not real and the reader has never seen this card!

What could be the issues:

- I have let the RX pin of the RDM6300 not connected(the driver does not need it). Could this be an issue?
- The fact that I've connected RDM6300's TX to GPIO3 might be an issue? Would changing the pin fix the issue?
- Could the serial interface for the RDM6300 driver be reading some fake data over time? Not clearing some buffers or something?
- Could all the RDM6300 readers be faulty? I have put one of them on a digital oscilloscope and does not send anything unless a card is there.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: **9.0.0.1(tasmota)**
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VS Code with PlatformIO
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
11:59:50 MQT: stat/RDR_F4CFA2D14DC8/RESULT = {"NAME":"Amica Deko","GPIO":[1,1824,1,1792,1,1,1,1,1,480,1,1376,1,1],"FLAG":0,"BASE":18}
11:59:50 MQT: stat/RDR_F4CFA2D14DC8/RESULT = {"Module":{"0":"Amica Deko"}}
11:59:50 MQT: stat/RDR_F4CFA2D14DC8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1824":"SerBr Rx"},"GPIO2":{"0":"None"},"GPIO3":{"1792":"SerBr Tx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"480":"Buzzer"},"GPIO14":{"0":"None"},"GPIO15":{"1376":"WS2812"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
12:02:01 MQT: stat/RDR_F4CFA2D14DC8/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":274,"Free":237,"Rules":"ON system#boot DO backlog Var1=0; Var2=0 ENDON ON wifi#disconnected DO backlog Var1=Var1+1; if (Var1>5) Restart 99 endif ENDON ON wifi#connected DO Var1=0 ENDON ON mqtt#disconnected DO backlog Var2=Var2+1; if (Var2>2) Restart 99 endif ENDON ON mqtt#connected DO Var2=0 ENDON"}
12:02:01 MQT: stat/RDR_F4CFA2D14DC8/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:02:01 MQT: stat/RDR_F4CFA2D14DC8/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"RDR_F4CFA2D14DC8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"3T14:43:32","StartupUTC":"2020-10-08T19:19:11","Sleep":40,"CfgHolder":4617,"BootCount":126,"BCResetTime":"2020-07-02T12:06:44","SaveCount":98257,"SaveAddress":"F7000"}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS2 = {"StatusFWR":{"Version":"9.0.0.1(tasmota)","BuildDateTime":"2020-10-03T10:39:06","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RFID","RFID1"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008008","28050A000000060000005A00000000000000","000002C0","00006000","00000000"]}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F983","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,51"}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS5 = {"StatusNET":{"Hostname":"RDR_F4CFA2D14DC8-3528","IPAddress":"10.155.203.6","Gateway":"10.155.200.1","Subnetmask":"255.255.240.0","DNSServer":"10.155.200.1","Mac":"F4:CF:A2:D1:4D:C8","Webserver":2,"WifiConfig":4,"WifiPower":20.5}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS6 = {"StatusMQT":{"MqttHost":"10.155.200.11","MqttPort":1883,"MqttClientMask":"RDR_%12X","MqttClient":"RDR_F4CFA2D14DC8","MqttUser":"mqttdeko","MqttCount":110,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS7 = {"StatusTIM":{"UTC":"2020-10-12T10:02:43","Local":"2020-10-12T12:02:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+02:00","Sunrise":"08:06","Sunset":"19:05"}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS10 = {"StatusSNS":{"Time":"2020-10-12T12:02:43"}}
12:02:43 MQT: stat/RDR_F4CFA2D14DC8/STATUS11 = {"StatusSTS":{"Time":"2020-10-12T12:02:43","Uptime":"3T14:43:32","UptimeSec":312212,"Heap":22,"SleepMode":"Dynamic","Sleep":40,"LoadAvg":24,"MqttCount":110,"POWER":"OFF","Dimmer":99,"Color":"4E79FC","HSBColor":"225,69,99","Channel":[31,47,99],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"A2:F3:C1:C5:95:58","Channel":1,"RSSI":48,"Signal":-76,"LinkCount":9,"Downtime":"0T00:11:46"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Can't exactly say. It happens randomly on all devices.

### EXPECTED BEHAVIOUR
For the 'fake' readings not to appear at all.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
All the 8 devices output this fake readings and all 8 are almost identical(some have an inverted buzzer) but the RDM6300 side is the same.
These fake readings appear randomly and on some of the devices more often than the others.
The system has 181 registered cards and to give you an idea, since the start of the month 819 fake cards/readings have popped up.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-12:
You cant use Gpio1 and Gpio3 since the USB-Serial converter of the NodeMCU is already connected to.

dbvcode, 2020-10-12:
> You cant use Gpio1 and Gpio3 since the USB-Serial converter of the NodeMCU is already connected to.

So I'm getting a bunch of serial data that the driver reads? And sometimes matches a valid 'card'?

Jason2866, 2020-10-12:
Just saying there are two devices sending data on the same GPIOs. Not a good idea.
To rule out problems, you have to change this first.

Jason2866, 2020-10-12:
Btw, why are you not using the Tasmota RDM6300 driver?

dbvcode, 2020-10-12:
> Btw, why are you not using the Tasmota RDM6300 driver?

That is what I am using, like documented here, [https://tasmota.github.io/docs/RDM6300/](https://tasmota.github.io/docs/RDM6300/)

dbvcode, 2020-10-12:
> Just saying there are two devices sending data on the same GPIOs. Not a good idea.
> To rule out problems, you have to change this first.

I will do this and report back tomorrow.

dbvcode, 2020-10-12:
> > Btw, why are you not using the Tasmota RDM6300 driver?
> 
> That is what I am using, like documented here, https://tasmota.github.io/docs/RDM6300/

Now I see there is a RDM6300RX option when configuring the Template!
When did that get here? :+1:

Jason2866, 2020-10-12:
Ahh, the docu is outdated. You could update :-)  It is Open Source too. A PR is welcome!

dbvcode, 2020-10-12:
> Ahh, the docu is outdated. You could update :-) It is Open Source too. A PR is welcome!

I've proposed the changes in the doc.

dbvcode, 2020-10-13:
Unfortunately I've changed the pin for the RDM6300 TX to GPIO5 and also used the 'native' driver.
The strange thing you will see here is that I use one card, **0085CCC7** that is valid and you will also see a **00000000** showing up  that is NOT a card!
My MQTT server failed while I was playing with the readers.
The fake number show up very shortly after _'Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec'_
So I assume it's related to the MQTT disconnecting/reconnecting.
I don't have the log for it, but while I was doing some configs in the console I also got **0010000** as a fake card out of the blue once.

`17:58:12 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
17:58:13 RSL: tele/RDR_50029179A1D5/SENSOR = {"Time":"2020-10-13T17:58:13","RDM6300":{"UID":"0085CCC7"}}
17:58:23 MQT: Attempting connection...
17:58:28 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
17:58:39 MQT: Attempting connection...
17:58:44 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
17:58:51 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T17:58:51","Uptime":"0T00:00:50","UptimeSec":50,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:AA:AB:28","Channel":5,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:09"}}
17:58:54 MQT: Attempting connection...
17:59:00 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
17:59:10 MQT: Attempting connection...
17:59:16 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
17:59:17 RSL: tele/RDR_50029179A1D5/SENSOR = {"Time":"2020-10-13T17:59:17","RDM6300":{"UID":"00000000"}}
17:59:26 MQT: Attempting connection...
17:59:32 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
17:59:36 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T17:59:36","Uptime":"0T00:01:35","UptimeSec":95,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:AA:AB:28","Channel":5,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:09"}}
17:59:42 MQT: Attempting connection...
17:59:48 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
17:59:58 MQT: Attempting connection...
18:00:03 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
18:00:03 RSL: tele/RDR_50029179A1D5/SENSOR = {"Time":"2020-10-13T18:00:03","RDM6300":{"UID":"00000000"}}
18:00:14 MQT: Attempting connection...
18:00:19 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
18:00:20 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T18:00:20","Uptime":"0T00:02:19","UptimeSec":139,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":199,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:AA:AB:28","Channel":5,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:09"}}
18:00:30 MQT: Attempting connection...
18:00:35 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
18:00:46 MQT: Attempting connection...
18:00:52 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
18:01:01 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T18:01:01","Uptime":"0T00:03:00","UptimeSec":180,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:AA:AB:28","Channel":5,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:09"}}
18:01:02 MQT: Attempting connection...
18:01:08 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
18:01:18 MQT: Attempting connection...
18:01:24 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
18:01:24 RSL: tele/RDR_50029179A1D5/SENSOR = {"Time":"2020-10-13T18:01:24","RDM6300":{"UID":"00000000"}}
18:01:34 MQT: Attempting **connection...`**

dbvcode, 2020-10-13:
@Jason2866 @ascillato2 Could someone please reopen this issue?

arendst, 2020-10-13:
I ordered some rdm6300 128khz today and will have a look what I find.

In the meantime to rule out power supply issues you might want to try one device with only the rdm6300 connected and see what happens. In the end responses with all 0 can easily be removed from the list.

dbvcode, 2020-10-13:
> I ordered some rdm6300 128khz today and will have a look what I find.
> 
> In the meantime to rule out power supply issues you might want to try one device with only the rdm6300 connected and see what happens. In the end responses with all 0 can easily be removed from the list.

Ok, I'll do this and report back.

Jason2866, 2020-10-13:
It is not a good idea for bug hunting to connect other stuff.
`{"Time":"2020-10-13T18:01:01","Uptime":"0T00:03:00","UptimeSec":180,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:AA:AB:28","Channel":5,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:09"}}
`
Connect only the RDM!

dbvcode, 2020-10-13:
Ok, so I've let the config the same and connected just the RDM.
It's still showing up. 
The power right now is USB from the PC. When it happens in 'production' I have an 1.4Ah stabilized source.
What next? Should I remove all the other devices from the config and leave only the RDM?

```
18:48:23 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:48:34 MQT: Attempting connection...

18:48:39 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:48:47 RSL: tele/RDR_50029179A1D5/SENSOR = {"Time":"2020-10-13T18:48:47","RDM6300":{"UID":"000000000000"}}

18:48:50 MQT: Attempting connection...

18:48:55 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:48:57 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T18:48:57","Uptime":"0T00:07:22","UptimeSec":442,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":81,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:BD:AB:28","Channel":5,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:07"}}

18:49:06 MQT: Attempting connection...

18:49:11 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:49:16 RSL: tele/RDR_50029179A1D5/SENSOR = {"Time":"2020-10-13T18:49:16","RDM6300":{"UID":"000000000000"}}

18:49:22 MQT: Attempting connection...

18:49:27 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:49:37 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T18:49:37","Uptime":"0T00:08:02","UptimeSec":482,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:BD:AB:28","Channel":5,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:07"}}

18:49:38 MQT: Attempting connection...

18:49:43 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:49:54 MQT: Attempting connection...

18:49:59 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:50:10 MQT: Attempting connection...

18:50:15 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:50:22 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T18:50:22","Uptime":"0T00:08:47","UptimeSec":527,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:BD:AB:28","Channel":5,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:07"}}

18:50:26 MQT: Attempting connection...

18:50:31 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:50:42 MQT: Attempting connection...

18:50:47 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:50:58 MQT: Attempting connection...

18:51:03 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:51:07 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T18:51:07","Uptime":"0T00:09:32","UptimeSec":572,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:BD:AB:28","Channel":5,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:07"}}

18:51:14 MQT: Attempting connection...

18:51:19 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:51:30 MQT: Attempting connection...

18:51:35 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:51:46 MQT: Attempting connection...

18:51:51 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:51:52 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T18:51:52","Uptime":"0T00:10:17","UptimeSec":617,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":215,"MqttCount":0,"POWER":"OFF","Dimmer":95,"Color":"4BF20E","HSBColor":"104,94,95","Channel":[29,95,5],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:BD:AB:28","Channel":5,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:07"}}

18:52:02 MQT: Attempting connection...

18:52:07 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:52:18 MQT: Attempting connection...

18:52:23 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec

18:52:23 RSL: tele/RDR_50029179A1D5/SENSOR = {"Time":"2020-10-13T18:52:23","RDM6300":{"UID":"000000000000"}}
```

Jason2866, 2020-10-13:
Yes, only the RDM configured and connected. Every other Gpio to none and stuff disconnected

dbvcode, 2020-10-13:
> Yes, only the RDM configured and connected. Every other Gpio to none and stuff disconnected

Ok, so I've set the template to `{"NAME":"Test","GPIO":[0,0,0,0,0,3616,0,0,0,0,0,0,0,1],"FLAG":0,"BASE":18}` and discovered issue #9532 
So GPIO5 is RDM, GPIO17 is User and all other are None and everything else is disconnected. 
The fake number shows up again:

```
19:28:12 MQT: Attempting connection...
19:28:17 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
19:28:28 MQT: Attempting connection...
19:28:33 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
19:28:40 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T19:28:40","Uptime":"0T00:00:50","UptimeSec":50,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:BD:AB:28","Channel":5,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:09"}}
19:28:44 MQT: Attempting connection...
19:28:49 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
19:29:00 MQT: Attempting connection...
19:29:05 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
19:29:16 MQT: Attempting connection...
19:29:21 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
19:29:25 RSL: tele/RDR_50029179A1D5/STATE = {"Time":"2020-10-13T19:29:25","Uptime":"0T00:01:35","UptimeSec":95,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:BD:AB:28","Channel":5,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:09"}}
19:29:31 MQT: Attempting connection...
19:29:37 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
19:29:37 RSL: tele/RDR_50029179A1D5/SENSOR = {"Time":"2020-10-13T19:29:37","RDM6300":{"UID":"00000000"}}
19:29:47 MQT: Attempting connection...
19:29:53 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
19:30:03 MQT: Attempting connection...
19:30:09 MQT: Connect failed to 10.155.200.11:1883, rc -2. Retry in 10 sec
```

dbvcode, 2020-10-13:
> I ordered some rdm6300 128khz today and will have a look what I find.
> 
> In the meantime to rule out power supply issues you might want to try one device with only the rdm6300 connected and see what happens. In the end responses with all 0 can easily be removed from the list.

Regarding the readings, the zeroes seem to be the most frequent errors, but just now 018C6000 popped up.

arendst, 2020-10-14:
Just a thought but I would think you use the received reading against a known database and act upon this. That way false responses are no issue as the chances are small they are one of your database records.

I still have to receive my devices but considering the low low price I'm not expecting top quality of these devices.

Another point is the possible data voltage conversion needed between the device 5V and the esp8266 3V3 power.

dbvcode, 2020-10-14:
@arendst indeed I read the cards against a database. Unfortunately I only send the cards and device ids to some other guy's ERP and let's just say he's not happy I'm the one implementing the readers solutions so he's giving me a hard time about any issue, especially fake readings.

On the devices themselves, your expectations will be met :) They are indeed low quality and by the time I started the project could not find any better that were working on 125KHz. This week I've found some Seeed Studio readers that should be the same in terms of serial output but another beast in terms of hardware. I ordered a few of them so I'm also expecting them to see if the fake readings disappear.

As for the 5V to 3.3 that might be something. Unfortunately I have no expertise when it gets to this low level stuff.

Jason2866, 2020-10-14:
Has it to be 125khz? The PN532 Rfid reader works very well and is supported from Tasmota. The Tags are cheap too.

dbvcode, 2020-10-14:
Unfortunately they already have 2 125Khz readers at the gates and already have the cards.

gemu2015, 2020-10-15:
@dbvcode 
did you supply the rdm6300 with 5 Volts ?
and did you connect the rdm6300 via a current limit resistor with the ESP ?
resistor is mandatory 1 to 5k, otherwise excessive current may flow into the ESP receive pin
this also may tear down the supply voltage of the reader.

since the RDM driver does a checksum test and rejects false telegrams it not possible that noise is responsible for your faulty  readings.

the reader itself must generate these errors.

mine is working without any error.

arendst, 2020-10-15:
Thx Gemu.

I now have the reader but had to order additional rfid tags (125kHz)... :-(

dbvcode, 2020-10-16:
> @dbvcode
> did you supply the rdm6300 with 5 Volts ?

Hi @gemu2015 I've supplied 5V to the RDM yes.
> and did you connect the rdm6300 via a current limit resistor with the ESP ?
> resistor is mandatory 1 to 5k, otherwise excessive current may flow into the ESP receive pin
> this also may tear down the supply voltage of the reader.

I did NOT use a resistor for the data line. By 1 to 5k you mean 1k to 5k or 1 Ohm?
> 
> since the RDM driver does a checksum test and rejects false telegrams it not possible that noise is responsible for your faulty readings.
> 
> the reader itself must generate these errors.

So you're saying the readers are have are faulty.
>
> mine is working without any error.

Is it the Chinese one? (Not SeeedStudio RDM630)

gemu2015, 2020-10-16:
the readers a probably not faulty in principle but may get faulty if you do not apply the current limit resistor. it must be in the range of 1k to 5k

yes i have a Chinese reader. (not seedStudio)

dbvcode, 2020-10-16:
> the readers a probably not faulty in principle but may get faulty if you do not apply the current limit resistor. it must be in the range of 1k to 5k
> 
> yes i have a Chinese reader. (not seedStudio)

Just put a 2.7K resistor there. I'll see what happens next.
Btw, what's the deal with the resistor? Is it needed only for the RDM? Or other devices that send serial data need it for the ESP8266?

gemu2015, 2020-10-16:
the RDM outputs 5V while the ESP works with 3.3V.  when you apply 5 Volts to a 3.3V device the input protection diodes become active and current will flow into VCC of the ESP.
this may have adverse effects depending on the current the RDM can deliver up to killing the input.

dbvcode, 2020-10-16:
> the RDM outputs 5V while the ESP works with 3.3V. when you apply 5 Volts to a 3.3V device the input protection diodes become active and current will flow into VCC of the ESP.
> this may have adverse effects depending on the current the RDM can deliver up to killing the input.

Thanks for the explanation.

dbvcode, 2020-10-16:
Regarding the fake readings, even with the 2.7K resistor they still show up so I guess the readers are faulty. Must have had been a bad batch.
In the meantime I've got a SeeedStudio reader that outputs the same serial way as the RDM6300 but is way snappier, picks cards from a bigger distance and for now does not seem to add fake readings.

Btw @gemu2015 what GPIO pin are you using for your RDM?

gemu2015, 2020-10-16:
you may use serial receive pin or any other suitable pin (depending on module)
the driver uses hardware serial when possible, else software serial.
(i use hardware serial, pin 3, in a bare module) 

with your devices the hardware USB to serial converter already uses the serial pins.

i would choose a software serial input e.g. GPIO 4

dbvcode, 2020-10-16:
Thanks for the feedback and help everybody!
@arendst I guess we can conclude the readers are faulty!

arendst, 2020-10-16:
I have to agree these readers are crap. I have the same issue as you have where valid ghost id are being reported:
```
16:12:47 DMP: 02 42 43 37 44 44 46 46 44 44 46 33 43 03
16:12:47 MQT: tele/wemos9/SENSOR = {"Time":"2020-10-16T16:12:47","RDM6300":{"UID":"7DDFFDDF"}}
16:13:48 DMP: 02 43 30 30 43 30 30 30 30 30 30 43 43 03
16:13:48 MQT: tele/wemos9/SENSOR = {"Time":"2020-10-16T16:13:48","RDM6300":{"UID":"0C000000"}}
16:16:42 DMP: 02 30 30 30 30 30 30 30 30 30 30 30 30 03
16:16:42 MQT: tele/wemos9/SENSOR = {"Time":"2020-10-16T16:16:42","RDM6300":{"UID":"00000000"}}
```
So let's safely decide to use RDM630 instead.

At least I now have the crappy RDM6300 too ;-)

dbvcode, 2020-10-16:
@arendst I am so glad I'm not the only one! :))
This does make me feel better even if I had to purchase 8 SeeedStudio ones out of my pocket to make the project work.
Cheers

dbvcode, 2020-10-28:
@arendst the issue seems to be somewhere else.
I've bought 8 SeeedStudio readers, changed the GPIO(as of my initial setup) and the fake readings pop up again.
What can we do next?

arendst, 2020-10-28:
As you cannot easily switch to differnet card types allowing the use of a PN532 based reader I'm out of ideas too.

I bought a seeedstudio grove reader and it has the same fault; it kept on reporting 00000000 so the latest driver filters out all zeros as fake.

As these all zeros are being read as valid data (it has a valid header and a valid tail) I'm afraid it's the reader that's at fault.

dbvcode, 2020-10-29:
Yes, the 00000000 is the most frequent fake number. Others pop up as well.
For now I filter for cards that are registered in the system, but it seems quite strange the readers behave this way.

vergoossen, 2020-10-29:
I had fake reading with an wiegand reader.
I tried another model wiegand reader wich I had also, and no false readings :).

dbvcode, 2020-10-29:
I've put the reader on the serial interface on the pc and I'm watching what it does...

dbvcode, 2020-10-29:
@arendst I've tried these tests:

1. Tried to trick Tasmota. Setup is as before. After boot I scanned one card then I disconnected the TX pin from the ESP8266. Nothing pops up for 2 hours.
2. Watch the output of the reader on the PC with a serial adapter. Did it for 2.5 hours, nothing shows out of the blue. Only when the tag is close to the reader it spits out the data!

What I'm trying right now:
Cleared flash, installed and configured [https://github.com/esprfid/esp-rfid]( https://github.com/esprfid/esp-rfid).
This one gets the RDM6300 TX signal on the NodeMCU's RX pin by default.
Will let it sit like this for a while and see if fake numbers show up.

If any of the tests make sense to you I'd be very happy if you tried them yourself as something is fishy!

Will post updates,
dbv

dbvcode, 2020-10-29:
So, with esp-rfid and the cheap RDM6300 reader the fake readings also pop up.
5 fake ones including the 0000000000(once) in in 4.5 hours.
Not good!

Jason2866, 2020-10-29:
Please close this issue since the problem is not Tasmota related.
Sorry.

closed 2020-10-29 by dbvcode

dbvcode, 2020-10-29:
@arendst the SeeedStudio reader you have[(if it's this one)](https://www.seeedstudio.com/Grove-125KHz-RFID-Reader.html) also has a pin to change from serial to Wiegand protocol.
Do you think it's hard to implement that into Tasmota?
SeeedStudio offers some details [here](https://wiki.seeedstudio.com/Grove-125KHz_RFID_Reader/) and they also include some code to do it, [details here(zip)](https://files.seeedstudio.com/wiki/Grove-125KHz_RFID_Reader/res/RFID_Wiegand_INT.zip).
My C skills to tasmotize the code are 0 so I ask pretty please if someone would do it?