arendst/Tasmota #9525: Add an option to make the ESP32 BLE-Driver listening to teleperiod

dikt1338, 2020-10-12:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
The ESP32 bluetooth driver is created in a way, that it works for BLE beacons: if a BLE-message is received, it will be transmitted to MQTT. This is perfect for BLE-beacons but not for sensors like MI-Flora which is also supported by the driver.
At the moment, the global option TELEPERIOD is totally ignored by the driver xsns_62_MI_ESP32.ino

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
There should be an option to only send mqtt messages from BLE-sensors like MI-Flora every x seconds as specified in TELEPERIOD

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I tried to understand the coding of xsns_62_MI_ESP32.ino to add the "listen to teleperiod"-feature in order to add a pull request. But I would need some help. Maybe the contributor of the driver @Staars could help me here? 

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-10-12:
I can add this option, but I am really curious what problem occurs with the direct message translation. It is simply the way, that the vendor designed these sensors. They just send their data at a certain frequency.

I could see no problems in my data flow with many incoming, but as a Node-Red-user it is usually hard to understand the glitches, that  dedicated home automation software might have.

My rough idea is to add a command, that can set the options as needed at boot time in the form of a bitfield, so something like option 0+2+3 could translate to 1+4+8-> MI32Mode 13. No fear, with the right documentation it should be easy to follow and it would not add much overhead (more important for both ESP8266-BLE-drivers).

dikt1338, 2020-10-12:
Thank you very much for your kind consideration of my feature request. I do agree that one could argue that the problem is rather “cosmetic”.
Of course I can drop messages in the backend.

As stated in my initial post already, I think that the driver should listen to teleperiod (as an option) as I don’t need 10sec updates on my plant’s soil humidity :)

Another problem with the “push a mqtt message as soon as you get a ble event” is that every now and then some values are transmitted empty causing an ugly “NaN” error in my home automation dashboard (Home Assistant) until new values arrive 10sek later.

Additionally if I do not filter the messages in the backend, my Bluetooth plant sensor is responsible for as many “sensor state changes” in my backend as the next 10 sensors causing unnecessary database growth.

I like to filter data as early as possible which would be tasmota 

Again, I understand that this is me being „Mr Monk“ 😊

Staars, 2020-10-12:
Ah, HASS ... I should have known it 😊.

Okay, I was already told about some potential ways to better interact with it by another developer. My plan is to add this in the next few days, probably after doing the same to the NRF24L01 and followed by the HM-10.
I still have to think about some details.

Staars, 2020-10-20:
https://github.com/arendst/Tasmota/pull/9588

This should revert to the old style. Other options can be selected in the code, but shall end in a new command in the future.

Next thing will be changes for HASS

pmknowles, 2020-10-22:
I second this. MIFlora floods MQTT every 9 seconds with the readings. Sensors like DS18B20 and consumption sensors update on a regular basis but only transmit on teleperiod. We can then decide if we want more frequ6updates by changing teleperiod.
If you do make the change it would also help to use a separate topic for the battery status. openHAB grumbles when JSONPATH doesn't find the battery status in the JSON.

dikt1338, 2020-10-22:
Thank you very much for the changes in the recent PR. It works like a charm here. The Flora-MQTT-Flood in my broker is gone now. One little thing about the MI Flora sensor: It seems that every now and then the values in tasmota driver change to zero. I guess its the MI Flora sensor itself which is responsible for that. Values are back in Tasmota with the next update from MI Flora (~10sek later). But if this "zero value sending" appears directly before the mqtt message, a mqtt message with empty sensor values is send which is not that good.
I am not sure whether this is something the tasmota MI driver should take care of. I leave this additional information just for discussion.

Staars, 2020-10-22:
Glad to hear that.

Regarding the zero-value issue I have no idea at the moment. To narrow down the problem, you could eventually use another project for testing in comparison to see if this is a problem of the driver or of the sensor. Feel free to open an issue, maybe more users have this problem too. I did not notice such behavior yet.
In one of the next updates I plan to integrate an automatic fallback for Floras with older firmware, that do not advertise their sensors readings.

closed 2020-10-22 by dikt1338