arendst/Tasmota #9529: Feature request: ability to add a "description" to Tasmota

hazymat, 2020-10-13:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
No

**Describe the solution you'd like**  
Ability to add a "description" to the Tasmota so that it shows on the front page. When we have multiple Tasmostas in the home with a naming convention, e.g. "sonoff-mini-009", it would be useful to quickly see the purpose or description of that Tasmota. When you open the web interface you see only its TYPE and NAME, e.g. "Sonoff Basic Module" and "Sonoff-Mini-009". Wouldn't it be good to see a subtitle like "Living Room Light" or whatever, this would help to remember what device has what function.

**Describe alternatives you've considered**  

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-13:
Already there. Use Template function. Described in the docs.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

sfromis, 2020-10-13:
Also, the device name is already shown on the front page.

closed 2020-10-13 by hazymat