arendst/Tasmota #9532: Template not setting correct values

dbvcode, 2020-10-13:
### PROBLEM DESCRIPTION
Seems if GPIO17 is set to None(0 in console), both from the console and the web interface the template does not set correctly:
```
19:21:56 CMD: Template {"NAME":"Test","GPIO":[0,0,0,0,0,3616,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
19:21:56 RSL: stat/RDR_50029179A1D5/RESULT = {"NAME":"Test","GPIO":[0,0,0,0,0,259,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
```
If I set it to User(1 in console):
```
19:23:56 CMD: Template {"NAME":"Test","GPIO":[0,0,0,0,0,3616,0,0,0,0,0,0,0,1],"FLAG":0,"BASE":18}
19:23:56 RSL: stat/RDR_50029179A1D5/RESULT = {"NAME":"Test","GPIO":[0,0,0,0,0,3616,0,0,0,0,0,0,0,1],"FLAG":0,"BASE":18}
```

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS = {"Status":{"Module":2,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"RDR_50029179A1D5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:39","StartupUTC":"2020-10-13T18:20:15","Sleep":50,"CfgHolder":4617,"BootCount":52,"BCResetTime":"2020-10-06T12:45:28","SaveCount":282,"SaveAddress":"FA000"}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS2 = {"StatusFWR":{"Version":"9.0.0.1(tasmota)","BuildDateTime":"2020-10-13T20:19:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"367/699"}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RFID","RFID1"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008008","28050A000000060000005A00000000000000","00000040","00006000","00000000"]}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F983","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,51"}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS5 = {"StatusNET":{"Hostname":"RDR_50029179A1D5-0469","IPAddress":"10.155.192.1","Gateway":"10.155.200.1","Subnetmask":"255.255.240.0","DNSServer":"10.155.200.1","Mac":"50:02:91:79:A1:D5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS6 = {"StatusMQT":{"MqttHost":"10.155.200.11","MqttPort":1883,"MqttClientMask":"RDR_%12X","MqttClient":"RDR_50029179A1D5","MqttUser":"mqttdeko","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS7 = {"StatusTIM":{"UTC":"2020-10-13T18:24:54","Local":"2020-10-13T19:24:54","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:08","Sunset":"18:03"}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS10 = {"StatusSNS":{"Time":"2020-10-13T19:24:54"}}
19:24:54 RSL: stat/RDR_50029179A1D5/STATUS11 = {"StatusSTS":{"Time":"2020-10-13T19:24:54","Uptime":"0T00:04:39","UptimeSec":279,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"RFID","BSSId":"CE:2D:E0:BD:AB:28","Channel":5,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Try `Template {"NAME":"Test","GPIO":[0,0,0,0,0,3616,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}` and ` Template {"NAME":"Test","GPIO":[0,0,0,0,0,3616,0,0,0,0,0,0,0,1],"FLAG":0,"BASE":18}`

### EXPECTED BEHAVIOUR
Expect the values from the output to equal the command.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-13:
The Tasmota setup for ESP8266 has no GPIO17. If you mean the analog pin, that is in the FLAG key.

arendst, 2020-10-14:
@sfromis since 9.x the adc state is set as gpio17 and flag is currently not used. See changelog.

@dbvcode Why this won't work I need to chk. It might be solved in the latest dev v9.0.0.2 so pls try that first. Also see changelog.

closed 2020-10-14 by ascillato2