arendst/Tasmota #9535: ModuleAddress to support 4 addresses instead of 3

adizag, 2020-10-13:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I wish to measure the NEUTRAL LINE with additional PZEM, but ModuleAddress let only define 3.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Adding additional address (0.0.0.4) to be defined for a 4th PZEM

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

TrDA-hab, 2020-10-22:
Today I read it again and did not find any restrictions from @arendst, I would also like to get 6 or 9 addresses for PZEM-004v3:
https://github.com/arendst/Tasmota/issues/2315

TrDA-hab, 2020-10-22:
@adizag
> I wish to measure the NEUTRAL LINE with additional PZEM, but ModuleAddress let only define 3.

You won't see anything interesting if you put PZEM # 4 on the neutral line.

You will see the sum of the asymmetry of the voltage vectors in your three-phase load (I don’t know how this can be explained in simple words) ... in other words: you will see information that will seem absolutely not interesting to you, professionals don’t do that (yes, they can get it , but use this information 0.01% of the time).

adizag, 2020-10-22:
It is not really matter. I want to be able to connect as many pzems for measuring specific deviced.

TrDA-hab, 2020-10-22:
I would also like this, but it seems to me that this is not possible without a global reconstruction of Tasmota. Alas, my opinion is not possible.

adizag, 2020-11-01:
I understood there could be up to 9 devices per one controller.
I saw that in one of the forums, realised with another firmware than Tasmota (which is the best 😍)

TrDA-hab, 2020-11-01:
It is very interesting! Can you give us this information?
Thanks.

closed 2020-12-11 by github-actions[bot]