arendst/Tasmota #9537: Scripting - button state (bt[x]) returns inconsistent values

jsgotham, 2020-10-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
First evalution of bt[x] after a state change yields 0 or 1.  Subsequent evaluations yield -127 or -128 ( state |= 0x80).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____ Sonoff Pow
- [x] Tasmota binary firmware version number used: _____ v9.0.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: _____ Platformio
- [x] Flashing tools used: _____ OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

05:49:53 RSL: stat/tasmota_075C8F/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2592":"HLWBL SEL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"2656":"HLWBL CF1"},"GPIO14":{"2688":"HLW8012 CF"},"GPIO15":{"288":"Led1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
05:50:23 RSL: stat/tasmota_075C8F/STATUS = {"Status":{"Module":6,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_075C8F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:57:22","StartupUTC":"2020-10-14T02:53:01","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-10-13T23:43:52","SaveCount":59,"SaveAddress":"3F7000"}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS2 = {"StatusFWR":{"Version":"9.0.0.1(tasmota)","BuildDateTime":"2020-10-14T13:24:43","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"345/699"}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gnomenclature2G4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS4 = {"StatusMEM":{"ProgramSize":637,"Free":1408,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAE797","043E8001","000000CF","010013C0","C000F881","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS5 = {"StatusNET":{"Hostname":"tasmota_075C8F-7311","IPAddress":"10.1.1.23","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"2C:3A:E8:07:5C:8F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_075C8F","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS7 = {"StatusTIM":{"UTC":"2020-10-14T04:50:23","Local":"2020-10-14T05:50:23","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:09","Sunset":"18:01"}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS10 = {"StatusSNS":{"Time":"2020-10-14T05:50:23","ENERGY":{"TotalStartTime":"2020-10-13T23:43:52","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
05:50:23 RSL: stat/tasmota_075C8F/STATUS11 = {"StatusSTS":{"Time":"2020-10-14T05:50:23","Uptime":"0T01:57:22","UptimeSec":7042,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Gnomenclature2G4","BSSId":"54:25:EA:CF:0C:60","Channel":4,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Compile with USE_SCRIPT & USE_BUTTON_EVENT
2. in section >b (or anywhere), evaluate bt[x] more than once.  First time = 0/1 state of button, subsequent times = previous value | 0x80  (as per line 1824 in x_drv10_scripter.ino).

So, for example the script on the wiki ...
if bt[1]==0  
then  
print falling edge of button1  
endif  
if bt[1]==1  
then  
print rising edge of button1  
endif 

Should never print 'rising edge of button1' because by the second evaluation bt[1] is evaluating as -127 or -128. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

bt[x] should evaluate to the same value until the actual state of the button changes

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

The wiki states that bt[x] is only valid in section b.
This is not true .... it evaluates anywhere, as long as compiled with USE_BUTTON_EVENT ... althougjh, as it stands, the value reported by bt[x] is inconsistent.

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-10-14:
the wiki example is outdated.
the bt[x] variable is updated by a button event and if read once is marked with a flag (ored 128)
this flag is cleared by the next event. this prevents events to be evaluated more than once

so the example should read
\>b
tmp=bt[1]
if tmp==0
then
print falling edge of button1
endif
if tmp==1
then
print rising edge of button1
endif

you may also read bt[x] in any other section and if you evaluate the flag can determine if there was an event though the fastest reaction would be in the >b section
\>S
tmp=bt[1]
if tmp<=1
then
// must be a new event
endif

jsgotham, 2020-10-14:
I can get what I want by using bt[1]&1 to eliminate the change flag, but I guess I'm wondering what purpose the change flag has ...
My thinking;
- If you're going to use button state change for any purpose, it's going to be used in the >b section (and in my case, a single button device, there's no need for a flag since it's always processed in >b).
- using the button state elsewhere has value (at least in my head - without having to worry about mapping between button index and gpio pin index)
- the only use for the state change flag would be in >b with multiple buttons (to figure out which button raised the event)
- knowing the index of the button that raised the event would seem to be more logically (and simpler) done by a >b scope variable "raising_index"  ...

The same would possibly apply to the >P section (having not played with a device with multiple power relays, I'm having trouble getting my head around that).  

BTW ... wanted to express my appreciation for your efforts (and everyone on the project). I'm really loving the scripting functionality which has got my project up and running much easier.  I'm also hoping that these kind of issues (and needing to respond to them) aren't a nuisance for you.

gemu2015, 2020-10-14:
this mechanism is needed to support more then one button events
you have several options to get button state.
the fastest is in the >p section like above
since buttons mostly are bound to a power output you may also monitor the power state in the >P section
or you may even monitor the raw button pin state with pin(x) 

Unfortunately the documentation is not always up to date and sometimes not very clear.

you are encouraged to commit changes to the Tasmota docs to explain things better

closed 2020-10-14 by ascillato2