arendst/Tasmota #9539: sunset/sunrise in permanent daylight trigger at 20h19

cybermaus, 2020-10-14:
### PROBLEM DESCRIPTION
Sunset and sunrise are evaluated as 1219 (20h19) in the timer, which is an actual value and causes the timer to trigger at the wrong time (at 20h19 in the evening)

I have 12 of these devices running on a camping in Sweden for 2 years now, and only now am I getting round to figure out why we need to manually turn off the lights during the core of the summer.

Additionally, it is evaluated as "15555" for rules variables %sunset% %sunrise%, and (255:2) in the timer webpage display. I suspect these are also accidental as actual internal values are 255:255 for hour:minute and I am sure it is just truncated to 5 positions, but I did not check

Ultimately, I am planning to build full intuitive, rule-less, daylight behavior to the timers, I already have such prototype code running. But that may be a bit outside the scope of this report.

I propose that if the DawnTillDusk() provides (255:255), then ApplyTimerOffsets() supplies 0x7FF or 0x7FE, depending on whether this is due to permanent day (summer) or permanent night (winter)
Both values are never reached in actual timer code, not even when random window offsets are added, and the timer never makes a false trigger. I already have this fix active and tested, and can provide it in a PR.

BTW: having separate values for winter and summer is useful for other changes I have in the works for myself, on a fully intuitive non-rule timer for less advanced users. I realize that may never make it to the official code, but having separate values does not bother this fix. 

Whether the rule variables and web display should also be adjusted, I am not sure.
Both do not cause false events now. But I could imagine having separate values on %sunset% and %sunrise% to distinguish between permanent day and permanent night may be useful to some people

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [X] Tasmota binary firmware version number used: 9.0.0.1
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Visual Studio/PlatformIO
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
na
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
na
```
- [ ] Provide the output of this command: `Status 0`:
```
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
```

### TO REPRODUCE
Go to the north of Sweden in summer.
Or alternatively, change your altitude to 67 or so and date to roughly June 21
Check Timer Webpage shows (255:2) for sunset/sunrise
Set 1 timer for OFF 30 minutes before sunrise
Set 2 timer to ON 30 minutes after sunset
Watch the lights go on at 20h49 (20h19+30 minutes)

### EXPECTED BEHAVIOUR
Timers should not activate, since it is still broad daylight  

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-14:
@s-hadinger  can you look into? Maybe a regression from #8384 ?

s-hadinger, 2020-10-14:
Oh that's an interesting one.

arendst, 2020-10-14:
I think it has always been "wrong" this way. I mean, who wants to camp that high up in Sweden????

arendst, 2020-10-14:
Yes. Just submit your PR please and I see if it's mergable ;-)

cybermaus, 2020-10-14:
Ok, this evening. I have to work with a customer this afternoon (day job to pay the bills)

PR #9543 (2020-10-15): Timer support for no-sunset permanent day
## Description:

**Related issue (if applicable):** fixes #9539 

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.3
  - [ ] The code change is tested and works on core ESP32 V.1.12.2
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tests. **Your  [... cut]
closed 2020-10-15 by arendst